arjenhiemstra/ithowifi #1: Only working in standard or medium setting

ServusNL, 2020-11-03:
I think it's a good idea to track potential progress on this via a issue:

The add-on is able to control the itho box in standard or medium mode setting only. This means you can use the original remote but if you leave the itho box in low or high setting the itho won't accept commands from the add-on. This is itho designed behaviour.

arjenhiemstra, 2020-11-12:
Hi ServusNL,

Unfortunately this seems to be an issues that can't be fixed, at least not with the current hardware from itho. 
Itho states in its manual that add-ons have this behaviour, doesn't matter if it is this one or the CO2, PWM of other module from itho itself.
There is only one solution I can think of that might mitigate part of the problem, integrate a RF module on the add-on that receives the remote commands instead of the itho box, basically taking over all the RF functionality from the itho box itself. Hardware wise this would mean a different and completely redesigned add-on (which I'm interested in but this is quite beyond the scope of the current project)

arjenhiemstra, 2021-01-18:
Hardware revision 2 adds the possibility include a CC1101 RF module to receive and learn itho remotes onto the add-on. 
This way it is possible to circumvent itho's limitation that add-ons are only able to control the itho box in standard / medium / auto mode setting only. You unlearn the remote from the itho box and learn the remote onto the add-on. The remote will then control the itho box through the add-on and the remote status is visible in your domotica system.

closed 2021-01-18 by arjenhiemstra