arjenhiemstra/ithowifi #2: Request: MQTT Home Assistant integration (automatic discovery)

remkolems, 2020-11-10:
Could you explore the possibility of implementing the auto discovery via mqtt of the ithowifi device within Home Assistant?

Example given
https://www.zigbee2mqtt.io/integration/home_assistant.html

Reference
https://developers.home-assistant.io/docs/device_registry_index/

arjenhiemstra, 2020-11-12:
I will look into it. I have no experience with building an auto discovery feature for HA but I like the idea and it might be handy with other projects :)

wimb0, 2020-11-24:
This link to the Home Assistant docs is also heplful: https://www.home-assistant.io/docs/mqtt/discovery/
Basically it's sending the configuration via MQTT messages.

arjenhiemstra, 2020-11-24:
> This link to the Home Assistant docs is also heplful: https://www.home-assistant.io/docs/mqtt/discovery/
> Basically it's sending the configuration via MQTT messages.

Thanks Wim, very useful info!

PR #15 (2021-03-28): Home Assistant Discovery
PR is still a WIP but wanted to share my progress. 
Warning; still in development, contains ugly hook into `execMQTTTasks` to keep triggering `mqttHADiscovery` for debugging purposes. Also has debug print/log statements.

This will eventually close #2.

- Implements Home Assistant's 'Auto Discovery' over MQTT functionality.
- adds config switch enable/disable (default: `disabled`)
- adds config input for Home Assistant MQTT prefix topic (default: `homeassistant`)

The core functionality is functional. struggling with adding 'config settings', hoping to get some feedbac [... cut]
closed 2021-03-28 by arjenhiemstra