arjenhiemstra/ithowifi #3: Integration of Itho Daalderop CO2 + Temperature/Relative Humidity sensor?

remkolems, 2020-11-10:
I have multiple Itho CVE-S mechanical ventilation units. One of them is an Optima variant - [CVE-S Optima IN type 03-00421](CVE-S Optima IN type 03-00421), which has the CO2 and RV (temperature + relative humidity) sensor integrated. 

The solution provided by Itho Daalderop is almost exactly like your IthoWifi hardware module.

Is there a way/could we think of a way to implement/integrate/circumvent the existing Itho Daalderop module with your module? 

See also the supplied images.

[image]
[image]
[image]
[image]

arjenhiemstra, 2020-11-12:
That's an interesting one! Unfortunately I don't have a CO2 add-on at my disposal but there might be two ways to approach this:
1- integrate the CO2 sensor onto the wifi add-on.
2- stack the boards

I don't have a CO2 module at my disposal and a bit of the design must be change probably but option 2 might be the easiest one, multiple i2c devices can reside on the bus and the wifi add-on is then just an extra device. (the wifi add-on has formally already two devices on the bus, the wemos and the attiny) The attiny can the left out if the CO2 module uses the same protocol.

Option 1 probably needs a bit more engineering depending on the way the CO2 sensor communicates with the board (analog, digital and in the latter case what kind of protocol).

muppet99, 2021-02-10:
Got my Logic analyzer yesterday. Hopefully next weekend i have some time to try it out on my co2 sensor. I have the same setup and same question.

sjorsjuhmaniac, 2021-03-28:
Is there any progress here? I'm also interested in an CO2 sensor. 

@arjenhiemstra, would it be possible to install a aftermarket i2c CO2 sensor like i'm running the tem/hum now? (directly connected to the ithowifi pins)

arjenhiemstra, 2021-03-29:
>would it be possible to install a aftermarket i2c CO2 sensor like I'm running the temp/hum now? (directly connected to the ithowifi pins

That always a possibility, at least with a little soldering to the I2C pins. Although I think the most CO2 sensors I know of communicate through UART instead of I2C

TD-er, 2021-03-29:
The SenseAir S11 does have an option for I2C.
You have to pull some pin (high or low, have to look in the datasheet) to switch to I2C mode.
I haven't used it myself yet, so I don't have any demo code.

arjenhiemstra, 2021-03-29:
>The SenseAir S11 ("Sunrise") does have an option for I2C.

Thanks! I see on their website that the SenseAir K30 and K33 also support I2C, interesting stuff!

TD-er, 2021-03-29:
Well those are rather expensive and quite large.
Just placed 3 of them together to show the scale
[image]
N.B. the LP8 is the same scale as the S8, but absolutely not exchangable, so just for scale here as I didn't have the S8 in reach.

arjenhiemstra, 2021-03-29:
That's quite a difference! Thanks!
It seems the S11 is the only viable option then.

rplevka, 2021-06-23:
I noticed this thread just now.
I am a happy user of this little guy:
https://www.sensirion.com/en/environmental-sensors/carbon-dioxide-sensors/carbon-dioxide-sensors-scd30/

Also, i just got my hands on their new and possibly revolutionary
https://www.sensirion.com/en/environmental-sensors/evaluation-kit-sek-environmental-sensing/evaluation-kit-sek-scd41/
which uses Photoacoustic principle and they were able to break the size barrier (the sensor itself is 1cm^3 in size).

they both run on i2c.

TD-er, 2021-06-24:
Yep, Senserion does make quite nice sensors.
Just keep in mind that applying Automatic Baseline Calibration (ABC) is something that requires some knowledge.
"Outside air" isn't always 400 ppm. For example in a windstill situation, when the sun is down and you have a lot of plants around, the CO2 concentration can easily reach 700 - 800 ppm outside.

Also in a room with drying paint the CO2 levels may be lower than 400 ppm. (and also near plants when the sun shines)

It also matters how long the ABC period is set. For example the MH-Z19 does have a very impractical 24h period.
So after one day of poor ventilation, the CO2 values will report too low for at least a day.
Even worse, in an environment which never reaches 400 ppm, there will be an offset so the sensor will eventually report 400 ppm on the least bad air conditions. For example city area, near industry or busy roads.

rplevka, 2021-07-02:
@TD-er you are absolutely right.
That's why I use the sensor with ABC disabled and perform controlled, manual recalibrations (sensirion allows that).

Anyway, `400 ppm`  is gone for good (it's more like 420 these days) :)
https://www.co2.earth/

I usually set the base to 450-500

jasperschwencke, 2021-10-29:
> That's an interesting one! Unfortunately I don't have a CO2 add-on at my disposal but there might be two ways to approach this: 1- integrate the CO2 sensor onto the wifi add-on. 2- stack the boards [...]

@arjenhiemstra Are you still in need of an original Itho CO2 sensor and PWM-board? I have on available and am happy to lend you one for research purposes. Let me know...

arjenhiemstra, 2021-10-29:
There is a tweaker trying it out, I'm waiting for the results :)
https://gathering.tweakers.net/forum/view_message/69183262

arjenhiemstra, 2021-10-29:
But thanks for the offer! I was given a demandflow device that contains a CO2 sensor, it look the same as the module in the CVE, not sure yet if it really is the same, still have to test is. I will keep you offer in my mind!

arjenhiemstra, 2022-01-20:
A more detailed post: https://gathering.tweakers.net/forum/list_message/69403538#69403538
It works with limitations. The CO2 sensor seems to have preference over other inputs (apart from remote commands of a real remote or virtual remote).

robert-blip, 2022-01-26:
Using the latest hardware with extended pins (request Arjen before buying): remove CO2 sensor, plug-in CVE add-on and on top of that the CO2 sensor. Values are shown on the add-on website and also provided via MQTT in the status.

Would be very nice if more entities are created as that would enable creating a history like: Fan speed (as it differs from the set value, or maybe one of the other Speed setting or Ventilation setpoint) and CO2Level. You can always disable sensors you don't want. Just a thought.

arjenhiemstra, 2022-01-26:
The information is send to HA just not parsed as a sensor out of the box. 
You might be interested to follow these examples on how to parse information relevant to you:
https://github.com/arjenhiemstra/ithowifi/wiki/Home-Assistant#sensors-in-home-assistant

robert-blip, 2022-01-27:
Thanks, I'll have a look at that!
New to MQTT while using this add-on.

Btw, I did not do this: https://github.com/arjenhiemstra/ithowifi/wiki/Home-Assistant#mqtt-enabling-in-home-assistant. MQTT integration can be configured via the GUI (now?).
Home Assistant seems to create 2 sensors 'out of the box' when it finds the Itho device. I'll check the Wiki on adding more.

robert-blip, 2022-01-27:
Working great.
Got this for the internal CO2 sensor:
  - platform: mqtt
    name: "ITHO CO2"
    unique_id: "itho_co2"
    device_class: "carbon_dioxide"
    state_topic: "itho/ithostatus"
    unit_of_measurement: "ppm"
    value_template: "{{ value_json['CO2level (ppm)'] }}"

Not sure what the device is doing. Might be because auto-discovery is also enabled. However, working nicely showing values of the internal CO2 sensor.

closed 2022-03-02 by arjenhiemstra