arjenhiemstra/ithowifi #4: Allow changing low/med/high via MQTT

wimb0, 2020-11-24:
Pleass add MQTT topic to allow setting the speed to low/med/high as set in the web interface setting.
For example something like "itho/speed" and then (off,) low, med or high as values.
This also gives better integration with the Home Assistant Fan component.

arjenhiemstra, 2020-11-24:
Hi Wim, this has already been implemented in the last firmware version, see this piece of code:
`  else {
    if (strcmp(s_payload, "low") == 0) {
      val = systemConfig.itho_low;
    }
    else if (strcmp(s_payload, "medium") == 0) {
      val = systemConfig.itho_medium;
    }
    else if (strcmp(s_payload, "high") == 0) {
      val = systemConfig.itho_high;
    }
    else {
      val = strtoul (s_payload, NULL, 10);
    }`

So if the MQTT command is not low, medium or high then it will try to translate the input to a number. if that also fails nothing gets processed.

wimb0, 2020-11-24:
Nice! Didn't see that, sorry.
I'll update my HA code to work with those values, thanks!
This issue can be closed then.

arjenhiemstra, 2020-11-24:
No problem! Documentation also needs to be updated on this point. Furthermore I'm working on implementing timer commands and the possibility to queue commands with a time2live on the queue.

closed 2020-11-24 by arjenhiemstra