arjenhiemstra/ithowifi #5: Way of working

bwired-nl, 2020-12-31:
Installed the d1 with print and working as described
Its to bad that we are not able to put the Itho in AUTO mode via the ESP8266 as well.
This reduces the functionality....
This is only possible via the remote as designed :-(
I will build a 433 remote on top of a Wemos D1
Regards !

arjenhiemstra, 2020-12-31:
This is unfortunately indeed itho designed behaviour. 
For the RF remotes to work you need a 866mhz remote btw not a 433 one. I'm working on a version that can include an optional CC1101 module to be able to let the add-on receive RF commands. I have a pre-prod version here that could use a tester, if you like to test it, please send me an email.

bwired-nl, 2020-12-31:
Great👍 Love to test knuvers@bwired.nl

bwired-nl, 2020-12-31:
Great
Graag test ik👍
Hoor graag
Groet Pieter

arjenhiemstra, 2020-12-31:
Same as #1

closed 2020-12-31 by arjenhiemstra