arjenhiemstra/ithowifi #7: Request: support for MQTT last-will topic

onderweg, 2021-01-24:
For monitoring if the add-on is online, it would me nice if a "last-will topic" can be set by the MQTT client in the firmware.
If then the wifi add-on disconnects unexpectedly from the MQTT broker for whatever reason, the MQTT broker will automatically notify all connected clients.

I'm not into ESP8266 programming, but I think it's a matter of only adding a setting for the topic name, and adding an argument in the `connect` function?
https://pubsubclient.knolleary.net/api.html#connect

arjenhiemstra, 2021-02-26:
Added as of Version 2.2-beta5

onderweg, 2021-02-26:
Thanks! Seems to work well :)

onderweg, 2021-02-26:
Small issue I see now: the 'online' message published is not retained. I guess that is solved is last argument in

```
 mqttClient.publish(systemConfig.mqtt_lwt_topic, "online", false);
```

is `true`.
(if the 'online' message is not retained, newly connected clients will see 'offline' as last state)

arjenhiemstra, 2021-02-27:
Fixed with beta6, please let me know if it now works as expected

onderweg, 2021-02-27:
Works as expected now! Thanks again.

arjenhiemstra, 2021-02-27:
Thanks for you feedback!

closed 2021-02-27 by arjenhiemstra