arjenhiemstra/ithowifi #11: Addon cannot to Wifi reliably with v2.0.9 and higer.

wimb0, 2021-02-22:
As discussed on Tweakers.net:

My Itho wifi addon v1.3 cannot connect (reliably) to my Unifi Wifi with firmware v2.0.9 and higher.

**What goes right:** 
Flash with 2.0.7 via web interface or USB. Enter Wifi details. Reset, done.
_Unifi AP Logging with v2.0.7:_
`Mon Feb 22 16:47:23 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb IEEE 802.11: associated
Mon Feb 22 16:47:23 2021 user.info : wevent[2227]: wevent.ubnt_custom_event(): EVENT_STA_JOIN ra1: 48:3f:da:95:9e:eb / 14
Mon Feb 22 16:47:23 2021 kern.warn kernel: ra1: Send EAPOL-Key M1, DA=48:3f:da:95:9e:eb, SA=f6:92:bf:1b:9b:53, len=113
Mon Feb 22 16:47:23 2021 kern.warn kernel: ra1: Recv EAPOL-Key M2, DA=f6:92:bf:1b:9b:53, SA=48:3f:da:95:9e:eb, len=129
Mon Feb 22 16:47:23 2021 kern.warn kernel: ra1: Send EAPOL-Key M3, DA=48:3f:da:95:9e:eb, SA=f6:92:bf:1b:9b:53, len=169
Mon Feb 22 16:47:23 2021 kern.warn kernel: ra1: Recv EAPOL-Key M4, DA=f6:92:bf:1b:9b:53, SA=48:3f:da:95:9e:eb, len=107
Mon Feb 22 16:47:23 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb WPA: pairwise key handshake completed (RSN)
Mon Feb 22 16:47:25 2021 user.info : wevent[2227]: wevent.ubnt_custom_event(): EVENT_STA_IP ra1: 48:3f:da:95:9e:eb / 192.168.3.43
Mon Feb 22 16:47:33 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb RADIUS: starting accounting session 3B01848EC557C770
Mon Feb 22 16:47:33 2021 user.info : stahtd[2228]: [STA-TRACKER].stahtd_dump_event(): {"message_type":"STA_ASSOC_TRACKER","ip_assign_type":"dhcp","wpa_auth_delta":"30000","mac":"48:3f:da:95:9e:eb","auth_ts":"346289.862188","vap":"ra1","assoc_status":"0","event_type":"success","ip_delta":"2320000","arp_reply_gw_seen":"yes","assoc_delta":"0","auth_delta":"0","event_id":"1"}
`
**What does not work correctly:**
With v2.0.9 and higher the addon does not always connect to Wifi:
_Unifi logging with firmware **hw1-v2.2-beta1-extra-wifi-log.bin**_
`Mon Feb 22 16:41:53 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb IEEE 802.11: associated
Mon Feb 22 16:41:53 2021 user.info : wevent[2227]: wevent.ubnt_custom_event(): EVENT_STA_JOIN ra1: 48:3f:da:95:9e:eb / 14
Mon Feb 22 16:41:53 2021 kern.warn kernel: ra1: Send EAPOL-Key M1, DA=48:3f:da:95:9e:eb, SA=f6:92:bf:1b:9b:53, len=113
Mon Feb 22 16:41:54 2021 kern.warn kernel: ra1: Send EAPOL-Key M1, DA=48:3f:da:95:9e:eb, SA=f6:92:bf:1b:9b:53, len=113
Mon Feb 22 16:41:54 2021 kern.warn kernel: ra1: Recv EAPOL-Key M2, DA=f6:92:bf:1b:9b:53, SA=48:3f:da:95:9e:eb, len=129
Mon Feb 22 16:41:54 2021 kern.warn kernel: ra1: Send EAPOL-Key M3, DA=48:3f:da:95:9e:eb, SA=f6:92:bf:1b:9b:53, len=169
Mon Feb 22 16:41:54 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb WPA: pairwise key handshake completed (RSN)
Mon Feb 22 16:41:54 2021 kern.warn kernel: ra1: Recv EAPOL-Key M4, DA=f6:92:bf:1b:9b:53, SA=48:3f:da:95:9e:eb, len=107
Mon Feb 22 16:42:03 2021 user.info : stahtd[2228]: [STA-TRACKER].stahtd_dump_event(): {"message_type":"STA_ASSOC_TRACKER","ip_assign_type":"roamed","wpa_auth_delta":"1050000","mac":"48:3f:da:95:9e:eb","auth_ts":"345959.964137","vap":"ra1","assoc_status":"0","event_type":"soft failure","assoc_delta":"0","auth_delta":"0","event_id":"1"}
Mon Feb 22 16:42:04 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb RADIUS: starting accounting session 059672CB94A6E3CB
Mon Feb 22 16:42:23 2021 daemon.info hostapd: ra1: STA 48:3f:da:95:9e:eb IEEE 802.11: disassociated
Mon Feb 22 16:42:23 2021 kern.warn kernel: Can't find pEntry(48:3f:da:95:9e:eb) in ApStaDel
Mon Feb 22 16:42:23 2021 user.info : wevent[2227]: wevent.ubnt_custom_event(): EVENT_STA_LEAVE ra1: 48:3f:da:95:9e:eb / 14
Mon Feb 22 16:42:23 2021 user.info : stahtd[2228]: [STA-TRACKER].stahtd_dump_event(): {"message_type":"STA_ASSOC_TRACKER","mac":"48:3f:da:95:9e:eb","assoc_status":"0","vap":"ra1","event_id":"1","event_type":"sta_leave"}

Logfiles:
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/6023206/logfile0.current.log)
[logfile1.current.log](https://github.com/arjenhiemstra/ithowifi/files/6023207/logfile1.current.log)

`

arjenhiemstra, 2021-02-22:
Could you try a recompiled version of 2.0.7? I just want to check it is not my build env. that is causing this.

[2.0.7-recompile.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6024235/2.0.7-recompile.bin.zip)

wimb0, 2021-02-22:
The recompiled version works fine, after a second reboot after updating.
See logfiles.
[logfile1.log](https://github.com/arjenhiemstra/ithowifi/files/6025295/logfile1.log)
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/6025296/logfile0.current.log)

arjenhiemstra, 2021-02-22:
Great thanks, there was a few bytes difference in this compile and the original one so just to be sure.
I'll build from this point and look again what changes were made.

arjenhiemstra, 2021-02-22:
So because 2.0.10 was not working either and also to establish a baseline (cause from 2.1 and onwards there are changes to the wifi section) could you also try a recompile of 2.0.10. The changes compared to 2.0.7 are:

`void logInput(char * inputString)` changed to: `void logInput(const char * inputString)`
and: `void jsonMessageBox(char* message1, char* message2)`
changed to: `void jsonMessageBox(const char* message1, const char* message2)`

The other changes are either only for HW rev 2 and not in code for HW rev 1 build or only active in access point mode (15 minute timeout).
[2.0.7-recompile.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6025812/2.0.7-recompile.bin.zip)

wimb0, 2021-02-23:
I've tested with the 2.0.10 recompile. Works fine!
Upload, wait, it reboots, and comes right back online.
So far no issues with that.

arjenhiemstra, 2021-02-23:
Thanks! Next is version 2.1 but with the wifi code of version 2.0.10. That will narrow it down to 1 function to look at after that.
Much appreciated if you could try this.
[2.1-wifi.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6029287/2.1-wifi.bin.zip)

wimb0, 2021-02-23:
Tested it: New 2.1 is also working correctly:

```
2021-02-23 14:12:51 N: Reboot requested
2021-02-23 14:13:06 N: System boot, last reset reason: Software/System restart
2021-02-23 14:13:06 N: WiFi: connection successful
2021-02-23 14:13:06 N: WiFi info:
2021-02-23 14:13:07 N: Mode:STA
2021-02-23 14:13:07 N: PHY mode:N
2021-02-23 14:13:07 N: Channel:11
2021-02-23 14:13:07 N: AP id:0
2021-02-23 14:13:07 N: Status:5
2021-02-23 14:13:07 N: Auto connect:1
2021-02-23 14:13:07 N: SSID (6):WB_IoT
2021-02-23 14:13:07 N: IP:192.168.3.43
2021-02-23 14:13:07 N: MQTT: connected, System config: on
2021-02-23 14:13:08 N: Webserver: started
2021-02-23 14:13:08 N: mDNS: started
2021-02-23 14:13:08 N: Hostname: nrg-itho-9eeb
2021-02-23 14:13:08 N: Setup: done
```

arjenhiemstra, 2021-02-23:
I'm now pulling in the wifi changes of the regular 2.1 firmware one by one, starting with the most important one: 
WiFi.persistent(false); // Do not use SDK storage of SSID/WPA parameters

[2.1-wifi-1.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6029627/2.1-wifi-1.bin.zip)

wimb0, 2021-02-23:
This version is also updating OK and working fine:
```
2021-02-23 14:53:05 N: Reboot requested
1970-01-01 00:00:05 N: System boot, last reset reason: Software/System restart
1970-01-01 00:00:05 N: WiFi: connection successful
1970-01-01 00:00:05 N: WiFi info:
1970-01-01 00:00:05 N: Mode:STA
1970-01-01 00:00:05 N: PHY mode:N
1970-01-01 00:00:05 N: Channel:11
1970-01-01 00:00:05 N: AP id:0
1970-01-01 00:00:05 N: Status:5
1970-01-01 00:00:06 N: Auto connect:1
1970-01-01 00:00:06 N: SSID (6):WB_IoT
1970-01-01 00:00:06 N: IP:192.168.3.43
1970-01-01 00:00:06 N: MQTT: connected, System config: on
1970-01-01 00:00:06 N: Webserver: started
1970-01-01 00:00:06 N: mDNS: started
1970-01-01 00:00:06 N: Hostname: nrg-itho-9eeb
1970-01-01 00:00:06 N: Setup: done
```
[image]

arjenhiemstra, 2021-02-23:
Nice! Now with: WiFi.setAutoReconnect(false); added
[2.1-wifi-2.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6029709/2.1-wifi-2.bin.zip)

wimb0, 2021-02-23:
Seems to work fine!
Instant Wifi after reboot.
[image]

arjenhiemstra, 2021-02-23:
Great, now almost the last one, added force wifi wake mode: WiFi.forceSleepWake();
[2.1-wifi-3.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6029934/2.1-wifi-3.bin.zip)

wimb0, 2021-02-23:
Same result, runs fine:
[image]

arjenhiemstra, 2021-02-23:
Ok, this is getting interesting! Now the last change and we are on the wifi code of the last beta1 release, the only change here is how quick the checks for a wifi connection come after each other (2 seconds instead of every second).
[2.1-wifi-4.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6030000/2.1-wifi-4.bin.zip)

wimb0, 2021-02-23:
Nope. This one doesn't come back online at my Wifi.

Also starts up its own AP:
[image]

arjenhiemstra, 2021-02-23:
Ok, did some research, it might be very well the case that the extra delay is causing issues / blocking the execution of wifi related processes on a low level. I reversed that piece of code and made the delay even a factor of 10 shorter than the original one.
[nrgitho-hw1-v2.2-beta3.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6030451/nrgitho-hw1-v2.2-beta3.bin.zip)

wimb0, 2021-02-23:
Works fine! :)
[image]

arjenhiemstra, 2021-02-23:
Nice!! Thanks a lot for al the testing! I will include the changed code to the mainline

wimb0, 2021-02-23:
Thank you for replying quickly and finding + fixing the issue in a few hours!
Looking forward to new stable builds with this code.
Also looking forward for the SHT31 implementation.

arjenhiemstra, 2021-02-23:
I will include (part) of the SHT31 implementation probably in the next version. The first indications are that running the sensor on a different address works fine without crashing the itho.

closed 2021-02-23 by arjenhiemstra