arjenhiemstra/ithowifi #12: MQTT stalls and makes addon reboot

GH2user, 2021-02-24:
Taken from tweakers.net to an issue here.

MQTT details filled out but stays at 'disabled'.
Every 20-30 seconds a reboot of the device.
Debug page sometimes is completely blank when device is restarting obviously, shown is:
memory 187212 / free 179248

log:
`

Config version: 003

Itho I2C connection status: connected

File system: 17319 bytes used / 173441 bytes total

CC1101 task memory: 1956 bytes free
MQTT task memory: 0 bytes free
Web task memory: 5892 bytes free
Config and Log task memory: 5876 bytes free
SysControl task memory: 5828 bytes free

2021-02-24 16:34:05 N: System boot, last reset reason: SDIO_RESET
2021-02-24 16:34:08 N: WiFi: connection successful
2021-02-24 16:34:08 N: WiFi info:
2021-02-24 16:34:08 N: Mode:STA
2021-02-24 16:34:08 N: Status:3
2021-02-24 16:34:08 N: IP:192.168.178.41
2021-02-24 16:34:08 N: Setup: init of CC1101 RF module successful
2021-02-24 16:34:26 N: MQTT: connection failed, System config: on
2021-02-24 16:34:26 N: Webserver: started
2021-02-24 16:34:26 N: mDNS: started
2021-02-24 16:34:27 N: Hostname: nrg-itho-21a4
2021-02-24 16:34:27 N: Setup: done
2021-02-24 16:34:27 N: Error: Task MQTT timed out!
2021-02-24 16:34:34 N: System boot, last reset reason: SDIO_RESET
2021-02-24 16:34:37 N: WiFi: connection successful
2021-02-24 16:34:37 N: WiFi info:
2021-02-24 16:34:37 N: Mode:STA
2021-02-24 16:34:37 N: Status:3
2021-02-24 16:34:37 N: IP:192.168.178.41
2021-02-24 16:34:37 N: Setup: init of CC1101 RF module successful
      2154 N: System boot, last reset reason: POWERON_RESET
      5005 N: WiFi: connection successful
      5093 N: WiFi info:
      5112 N: Mode:STA
      5130 N: Status:3
      5149 N: IP:192.168.178.41
      5250 N: Setup: init of CC1101 RF module successful
2021-02-24 16:35:02 N: MQTT: connection failed, System config: on
2021-02-24 16:35:02 N: Webserver: started
2021-02-24 16:35:02 N: mDNS: started
2021-02-24 16:35:02 N: Hostname: nrg-itho-21a4
2021-02-24 16:35:02 N: Setup: done
2021-02-24 16:35:03 N: Error: Task MQTT timed out!
`
MQTT details which are working on other device;
Status:  disabled
MQTT active; ON
server: 192.168.178.037
user: abcd
pwd: abcd
port: 1883
state topic: itho/state
cmd topic: itho/cmd
Domotiqz: OFF

User and Password are replaced by abcd but are 4 char each in real life
the server address has the 0 automatically added in front of 37 in the last byte of the address.

arjenhiemstra, 2021-02-24:
Ok, there are probably two issues here. If the MQTT server is not reachable the add-on reboots. Need to look into it further but most probably this is a stupid bug somewhere.
Second, the zero in front of the 37. This was added automatically?? That weird and probably the reason you cannot connect to the MQTT server.

For now there are two possible ways the solve the immediate issue.
Downgrade to v2.1 or manually edit the config file to remove the zero. First switch of MQTT to prevent the reboots. Then go to http://[IP or DNS]/edit  login/pw admin and edit system.json to remove the zero.

GH2user, 2021-02-24:
Regarding the 0 in front of 37.
I added the IP as 192.168.178.37 using Edge chromium and after clicking save it came back with the leading zero in the display.

GH2user, 2021-02-24:
Edit the config.json as there is no system.json
It then indeed connected correctly !
Now to find out where this bloody zero is coming from ...
thanks for the quick solution!

arjenhiemstra, 2021-02-24:
Sorry config.json is what I meant indeed!
Are you using Edge on windows? Do you have for example chrome to test wether you observe the same behaviour?

GH2user, 2021-02-25:
Indeed using Edge on win10.
Tried now using firefox and edge one after the other.
both switch off mqtt, save
both switch on and retype the IP, save
both work as expected, that is no extra 0 on edge.
do I need to change the config.json inbetween stop / start ?

arjenhiemstra, 2021-02-25:
the config.json file is not removed when you switch MQTT off. Best is to delete the config.json file (menu reset -> reset system config) and try it again.
Your wifi settings will be unaffected by this.

GH2user, 2021-02-25:
hi arjen,
did that and all mqtt settings are gone, filled them out again and it worked on both firefox and win10 edge.
No clue what went wrong yesterday.
the whole system seems more responsive now than it was yesterday even prior to setting the mqtt settings, maybe it was still doing something after starting up.
Anyhow shouldn't put to much thoughts into it for now and for me this issue can be closed.
I learned a lot more about the system now, going into the edit-section and doing all this.
Thanks for the quick responses and helping me out.

arjenhiemstra, 2021-02-26:
Strange, I cannot reproduce it either. But still two interesting glitches found, thanks for that!
The MQTT task timeout is shorter than the MQTT connection timeout, this caused the reboot. I will fix that in the next version.
I will also look at input validation for the IP address fields.

closed 2021-02-26 by arjenhiemstra