arjenhiemstra/ithowifi #13: Module cannot control ITHO after RF usage

zuidwijk, 2021-02-25:
I installed your latest hardware revision. And it works, until I use my remote. Than you cannot control the ITHO box. If I choose to restart the device via web gui, no effect. I have to physically power off and on the ITHO to make it work again.

Software 2.2-beta1 and later 2.1
ITHO: CVE ECO RFT SP
Control board: h60s40 **E.CVERF-B-S_U2**
````
      2210 N: System boot, last reset reason: POWERON_RESET
      3962 N: Setup: AP mode active
      3987 N: Webserver: started
      4008 N: mDNS: started
      4028 N: Hostname: nrg-itho-688c
      4048 N: Setup: done
      2154 N: System boot, last reset reason: TG0WDT_SYS_RESET
      3816 N: Setup: AP mode active
      3842 N: Webserver: started
      3864 N: mDNS: started
      3884 N: Hostname: nrg-itho-688c
      3904 N: Setup: done
    152728 N: Reboot requested
      2154 N: System boot, last reset reason: SDIO_RESET
      6612 N: WiFi: connection successful
      6631 N: WiFi info:
      6650 N: Mode:STA
      6669 N: Status:3
      6687 N: IP:10.0.0.154
      6713 N: Webserver: started
      6734 N: mDNS: started
      6755 N: Hostname: nrg-itho-688c
2021-02-24 17:48:42 N: Setup: done
2021-02-24 17:52:23 N: Reboot requested
2021-02-24 17:52:26 N: System boot, last reset reason: SDIO_RESET
2021-02-24 17:52:28 N: WiFi: connection successful
2021-02-24 17:52:28 N: WiFi info:
2021-02-24 17:52:28 N: Mode:STA
2021-02-24 17:52:28 N: Status:3
2021-02-24 17:52:28 N: IP:10.0.0.154
2021-02-24 17:52:28 N: Webserver: started
2021-02-24 17:52:28 N: Setup: init of CC1101 RF module successful
2021-02-24 17:52:28 N: mDNS: started
2021-02-24 17:52:28 N: Failed to deserialize remotes config json
2021-02-24 17:52:28 N: Hostname: nrg-itho-688c
2021-02-24 17:52:28 N: Setup: done
      2156 N: System boot, last reset reason: TG0WDT_SYS_RESET
      4612 N: WiFi: connection successful
      4631 N: WiFi info:
      4651 N: Mode:STA
      4669 N: Status:3
      4687 N: IP:10.0.0.154
      4715 N: Webserver: started
      4735 N: Setup: init of CC1101 RF module successful
2021-02-24 18:14:34 N: mDNS: started
2021-02-24 18:14:34 N: Hostname: nrg-itho-688c
2021-02-24 18:14:34 N: Setup: done
2021-02-25 00:14:29 N: Mem free: 182172, Mem low: 128828, Mem block: 97392
2021-02-25 06:14:29 N: Mem free: 182176, Mem low: 128828, Mem block: 97392
2021-02-25 07:17:10 N: Reboot requested
2021-02-25 06:52:35 N: System boot, last reset reason: SDIO_RESET
2021-02-25 06:52:37 N: WiFi: connection successful
2021-02-25 06:52:37 N: WiFi info:
2021-02-25 06:52:37 N: Mode:STA
2021-02-25 06:52:37 N: Status:3
2021-02-25 06:52:37 N: IP:10.0.0.154
2021-02-25 07:17:16 N: Setup: init of CC1101 RF module successful
2021-02-25 07:17:17 N: MQTT: connected, System config: on
2021-02-25 07:17:17 N: Webserver: started
2021-02-25 07:17:17 N: mDNS: started
2021-02-25 07:17:17 N: Hostname: nrg-itho-688c
2021-02-25 07:17:17 N: Setup: done
2021-02-25 07:29:33 N: Firmware update: NRG_itho_wifi_HW2x_FW2.1.bin
2021-02-25 07:29:50 N: Reboot requested
2021-02-25 07:29:55 N: System boot, last reset reason: 6
2021-02-25 07:29:55 N: WiFi: connection successful
2021-02-25 07:29:55 N: WiFi info:
2021-02-25 07:29:55 N: Mode:STA
2021-02-25 07:29:55 N: Status:3
2021-02-25 07:29:55 N: IP:10.0.0.154
2021-02-25 07:29:57 N: MQTT: connected, System config: on
2021-02-25 07:29:57 N: Webserver: started
2021-02-25 07:29:57 N: mDNS: started
2021-02-25 07:29:57 N: Hostname: nrg-itho-688c
2021-02-25 07:29:57 N: Setup: done
2021-02-25 07:29:57 N: Setup: init of CC1101 RF module successful
      4313 N: System boot, last reset reason: 7
      4332 N: WiFi: connection successful
      4351 N: WiFi info:
      4370 N: Mode:STA
      4389 N: Status:3
      4408 N: IP:10.0.0.154
      4443 N: MQTT: connected, System config: on
      4468 N: Webserver: started
      4488 N: mDNS: started
      4507 N: Hostname: nrg-itho-688c
      4528 N: Setup: done
````

The logs aren't telling me much, I hope you see something more 👍

zuidwijk, 2021-02-25:
**Something worth mentioning:**
When I disconnected the power, the blue Itho_status led blinks a few times slow, than fast, than dies.
Reconnecting the power after this moment and the module doesn't work. Not connection to WiFi, not going into AP-mode (standalone).
Disconnecting the power, wait till Itho_status led blinks fast and than reconnect the power, and it works again.

arjenhiemstra, 2021-02-25:
Could you send a picture of the itho main pcb? I somehow sounds like the itho firmware is locking up.

>When I disconnected the power, the blue Itho_status led blinks a few times slow, than fast, than dies.
That's the itho power supply trickling the last bits of power to the add-on (you might also see the led's on the itho main pcb blinking a few times) After the led is completely off you should wait at least 15 sec or so according to the itho manual.

zuidwijk, 2021-02-25:
> Could you send a picture of the itho main pcb? I somehow sounds like the itho firmware is locking up.
[image]

> That's the itho power supply trickling the last bits of power to the add-on (you might also see the led's on the itho main pcb blinking a few times) After the led is completely off you should wait at least 15 sec or so according to the itho manual before powering on again.

Ah ok, I'm not sure how much time I've waited. I'll keep that in mind next time ;-)

arjenhiemstra, 2021-02-25:
>And it works, until I use my remote

Is there a difference using the slider on the webinterface or the remote?
Does the itho status led turn on when turning the itho on and does it go off after a few seconds?

zuidwijk, 2021-02-25:
> > And it works, until I use my remote
> 
> Is there a difference using the slider on the webinterface or the remote?
> Does the itho status led turn on when turning the itho on and does it go off after a few seconds?

Yes there is... the rf remote is connected directly to the box (not your print). Once the box is triggered via RF (either the original remote, or the "old" esp remote I already have) than the box doesn't respond to your pcb.

**Tried several things:**
- Made the box powerless for 30 seconds... reconnect the power, the module does not go active (I'll make a video of it)
- When changing speed via your module, the two leds on the mainboard won't lid
- When changing speed via RF, the left led (red) blinks 1x and the right led (amber) blinks several times

arjenhiemstra, 2021-02-25:
Ok, if you switch on the itho box and don't send any commands using RF or otherwise yet, does moving the slider on the webinterface change the speed of the itho?

zuidwijk, 2021-02-25:
> Ok, if you switch on the itho box and don't send any commands using RF or otherwise yet, does moving the slider on the webinterface change the speed of the itho?

Yes, it works perfectly. MQTT connected and shows the speed changes too (either via the buttons, or slider).

Awkward ... while making a video, powerless for about 10 seconds, now it reconnects to the wifi ... That's a minor ...

arjenhiemstra, 2021-02-25:
And if you used your remote and the last command given by remote is medium or auto (depending on your remote), what is then the result?

zuidwijk, 2021-02-25:
> And if you used your remote and the last command given by remote is medium or auto (depending on your remote), what is then the result?

I've got a low/med/high/timer remote. When last command is medium, than the module can change the speed.
When pressing high or low on the remote, I can't use the module.

arjenhiemstra, 2021-02-25:
https://github.com/arjenhiemstra/ithowifi/raw/master/handleiding.pdf

Please read the second page: "belangrijk om te vermelden"

zuidwijk, 2021-02-25:
> https://github.com/arjenhiemstra/ithowifi/raw/master/handleiding.pdf
> 
> Please read the second page: "belangrijk om te vermelden"
> And since you have a HW rev 2 with CC1101 module, please also read page three to make use of the CC1101 module to circumvent this behaviour.

Check, read it ... that's a bummer

So best way is to connect the remote to the add-on? As mentioned in 
**Extra informatie hardware revisie 2 > RF remote support (alleen hardware revisie 2 – Experimenteel!):**

arjenhiemstra, 2021-02-25:
That is the whole purpose of the added RF module. 

Unlearn the remote from the itho and add up to 10 remotes to the add-on. Functionality of the remote is at least exactly the same as normal but the result of the remote is now also visible in your domotica system. (in theory you could assign your own functions to the remote but for now you can only change the speed settings for every button)

closed 2021-02-25 by arjenhiemstra