arjenhiemstra/ithowifi #14: Doesn't work with mesh networks.

t3hk0d3, 2021-03-11:
Looks like i've found an issue when an add-on module is used with a wifi mesh network.

When the module first starts with its own AP network everything works fine, even after a soft restart from WebUI.

However, after restarting it never connects to the configured WiFi network.
I've tried it with a regular (non-Mesh) wifi network, and it works fine.

I think it gets confused when several different APs respond to the same SSID.

TD-er, 2021-03-11:
What I do in ESPEasy is to perform a scan and pick out all known SSIDs and keep records of the best RSSI + BSSID + channel.
Then I connect to the specific AP with the best RSSI by handing over the BSSID + channel to the WiFi.begin call.

Maybe that will help here too?

N.B. do you use hidden SSIDs?

t3hk0d3, 2021-03-11:
@TD-er No, SSID is visible.

TD-er, 2021-03-11:
Some mesh APs support a range of acceptable RSSI values for the initial connection and some threshold for kicking it so it will reconnect to another AP in the mesh.
Maybe you can check your AP settings / logs to see why it can't reconnect?

t3hk0d3, 2021-03-11:
@TD-er MV is quite close to Wi-Fi AP. Also it connects and works fine after soft-reset. Problem is hard-reset.

t3hk0d3, 2021-03-11:
I'll try to get my hands on the FTDI serial port connector, maybe logs can provide more information.

t3hk0d3, 2021-03-11:
Also it should create WiFi hotspot if it can't connect to Wi-Fi network, but it doesn't happen:

https://github.com/arjenhiemstra/ithowifi/blob/master/software/NRG_itho_wifi/09_init_code.ino#L279

arjenhiemstra, 2021-03-11:
If @TD-er is out of ideas and starts asking for a real developer than I'm scared ;-)

@t3hk0d3, with a hard-restart you mean a power cycle of the itho I guess? In that case, does the itho status led light up briefly (depending on you itho model somewhere between 1 and 7 seconds) and thereafter turns off?

TD-er, 2021-03-11:
Could it be the node is waiting for a DHCP indefinately?

t3hk0d3, 2021-03-11:
@arjenhiemstra 
exactly. Itho status flashes briefly and nothing flashes afterwards.

@TD-er 
I'd seen it connected to wifi ap in peer list. Also with soft-reset it works flawlessly.

arjenhiemstra, 2021-03-11:
Ok thanks, that confirms that you waited long enough before powering on again (I have seen otherwise so please forgive me for checking)

If the wifi status led doesn't show any activity after 'cold' start this indicates a hardware issue.
Before anything else happens the add-on checks the failsave pin, the wifi status led then flashes rapidly about 5x a second. Only after this part of the code anything Wifi related starts.
Your description of the issue indicates the code never reaches the part where even the pin status check happens. My conclusion is then that this cannot have anything to do with the wifi settings or code but must be something in hardware or closely related.

t3hk0d3, 2021-03-11:
@arjenhiemstra well, a fail-safe pin works for me, and I can connect to the 192.168.4.1/update page. Power-cycling MV box (with waiting until capacitors discharge), connecting to AP hotspot, configuring wifi and doing soft-reset make it works....until next power cycle.

arjenhiemstra, 2021-03-11:
I know and agree it is not completely logical and I don't not exactly what the reason could be. But I know for sure that there is one piece of code that gets executed before anything else, flash the led. In all circumstances, before anything gets loaded, whatever the config is.
If the wifi led doesn't flash on power on, the code doesn't start is my conclusion. There is normally nothing you as a user can do to influence that proces (at least not with the firmware versions I publish).
I'm happy to supply you with a replacement module, I will include a return label so that you can return the current module.

t3hk0d3, 2021-03-11:
@arjenhiemstra Strange thing when I short fail-safe pin WiFi led flashes normally. I made a short video to illustrate issues:
https://www.youtube.com/watch?v=wVbBjLmOQwU&feature=youtu.be

My Tindie order Id is #251001.

arjenhiemstra, 2021-03-11:
>Strange thing when I short fail-safe pin WiFi led flashes normally.
That contradicts my line of thinking. Very strange.

If you could test the following (only use DHCP please cause there is a static IP bug in this version)
Please install firmware version 2.1 instead of 2.1.1 or the 2.2 beta version and test if the issues are still there? There has been a small change in 2.1.1 to the failsafe code, basically the first since version 2.0. You never know.... maybe I'm overlooking something stupid here

t3hk0d3, 2021-03-12:
@arjenhiemstra I've tried 2.1, unfortunately it has same behavior.

arjenhiemstra, 2021-03-12:
Ok, thanks a lot for trying! I will send you a new module later today.

t3hk0d3, 2021-03-12:
> Ok, thanks a lot for trying! I will send you a new module later today.

Dankjewel! I am looking forward for it.

arjenhiemstra, 2021-03-14:
Have you had a change to try the replacement module?

t3hk0d3, 2021-03-15:
@arjenhiemstra Yes, replacement module works perfectly. Thanks a lot!

arjenhiemstra, 2021-03-15:
Great! Then I'll close the issue.

closed 2021-03-15 by arjenhiemstra