arjenhiemstra/ithowifi #18: 'Waiting' appears while upgrading OTA

sjorsjuhmaniac, 2021-03-18:
During the upload of the firmware file the 'loading'/'waiting' thingy appears before the upload/upgrade is completed.

By not doing anyting the it disappears again after 5 seconds but the progressbars for the OTA are gone. 
Doning nothing again and just waiting eventually shows the 'Done...' twice + the 'text ... reboot in 20 seconds'. The numbers are counting down.

The upgrade is successfull so the OTA process is continued and it seems to be a website/visual thing only. 
I'm unable to reproduce this consistently. It looks like it happens in about 1/3 of every OTA.
Currently experience this with 2.2-beta8 builds but also had this with the 2.2-beta7 builds.

arjenhiemstra, 2021-03-18:
I've seen it as well. Seems that the websocket hangs. Not sure why yet.

arjenhiemstra, 2021-03-24:
I've made some changes to beta9 that I think could solve this. I haven't seen the websocket error and reconnect anymore since. I'd like to hear your experience.

sjorsjuhmaniac, 2021-05-13:
i've been uploading on firmware > 2.2 for the last couple of days and this no longer happens. :+1:

closed 2021-05-13 by sjorsjuhmaniac