arjenhiemstra/ithowifi #24: CC1101 RF module failed

rkuijer, 2021-04-08:
Today I received the temp/hum sensor and was planning to move my remote to the addon when I saw that the CC1101 wasn't working.
The steps I took:
* Installed beta 10, OK
* Turned on the HA discovery, OK
* Installed the temp/hum sensor, OK (thanks !)
* Turned on the activate the CC1101 RF module, NOK
The logging shows `Setup: init of CC1101 RF module failed`

When I received the addon in the beginning I'm sure I checked this and at the time everything was OK.
[logfile1.current.log](https://github.com/arjenhiemstra/ithowifi/files/6280708/logfile1.current.log)

arjenhiemstra, 2021-04-08:
This is the case since the upgrade to beta 10? Or are you not sure?
Otherwise maybe a good idea to revert back to the previous version just as a test. If that doesn't work the only thing that I can come up with is that somehow the CC1101 module failed or the connection to the module failed. In that case I will send you a replacement module.

rkuijer, 2021-04-08:
I'm not sure if it's since version beta 10.
Let me revert back to an older version to rule out if it's just beta 10.
Is there a way to retrieve logging of the init process of the CC1101 module?

arjenhiemstra, 2021-04-08:
You just gave the logging of the init process ;-)
What happens is that if the module is not working or one of the Power and/or SPI pins are not connected the module won't initialise. The firmware detects this and log the message you copied. There is one other pin that cannot be checked this way but that will result in a good init but remote signals won't be processed.

rkuijer, 2021-04-08:
I mean a more detailed logging 😛
Ok, so if dust of my scope I could try to see if the it's a connection issue. I'll look for the data sheets of the CC1101 module to guide me. 
Does the Attiny only handle the init of the Itho or also the init of the CC1101 module?

arjenhiemstra, 2021-04-08:
I can reproduce it, beta9 is working just fine.... somehow, don't know why I implemented a bug.....

arjenhiemstra, 2021-04-08:
A recompile of the same source seems to 'fix' the issue. I'm afraid I made a mistake somewhere and selected the wrong build environment for the final build of beta10, doh! 
2.2-rc1 is the version you can try to see if this fixes your issue.

TD-er, 2021-04-09:
> edit: btw 'tomorrow' has become today :P

That happens every day ;)

rkuijer, 2021-04-09:
Sorry to tell you..........
`Setup: init of CC1101 RF module failed`
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/6283642/logfile0.current.log)

arjenhiemstra, 2021-04-09:
grmbl.... i'll investigate further. Could you please downgrade to beta9 and see if the init works again with that version. That at least rules out a hardware issue then.

rkuijer, 2021-04-09:
Let's look at the code changes......beta 9 is OK
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/6284234/logfile0.current.log)

arjenhiemstra, 2021-04-09:
I believe I found it. There was a compiler directive not clean-up completely which resulted in an SPI pin not being set as output. This only manifests itself after a cold start.... so upgrade would go fine and the CC1101 module would work just fine but after a powercycle the CC1101 module is not detectable anymore. pfff, this was a nasty one.
Could you try this firmware file?
[nrgitho-hw2-v2.2-rc2.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6286907/nrgitho-hw2-v2.2-rc2.bin.zip)

rkuijer, 2021-04-09:
This afternoon I looked at the delta between beta 9 and 10 and could find a lead.
How did you find it?

The new bin is OK
```
2163 N: System boot, last reset reason: POWERON_RESET
2182 N: HW rev: 2, FW ver.: 2.2-rc2
4240 N: WiFi: connection successful
4307 N: WiFi info:
4326 N: Mode:STA
4345 N: Status:3
4364 N: IP:192.168.41.149
4433 N: Setup: Virtual remote ID: 18,142,112
4471 N: Setup: Alternative SHT30 sensor found
4507 N: Setup: init of CC1101 RF module successful
5991 N: MQTT: connected, System config: 1
6069 N: Webserver: started
6089 N: mDNS: started
6109 N: Hostname: nrg-itho-8e70
6131 N: Setup: done
```

arjenhiemstra, 2021-04-09:
> This afternoon I looked at the delta between beta 9 and 10 and could find a lead.
> How did you find it?

I removed '#include "hardware.h"' with commit 92fdcb0 which is actually not really used anyway but forgot to remove https://github.com/arjenhiemstra/ithowifi/blob/8ac0b6fb13f52bbba5f6ca59eebafa623c568bb8/software/NRG_itho_wifi/CC1101.cpp#L11
which results in 'pinMode(SS, OUTPUT);' not being compiled in.

Apparently only a cold boot resets the pin regs of the ESP32, so an upgrade from beta9 to beat10 would first just work fine.
That set me on the wrong foot a few times but after a lot of trying I could narrow it down to the mentioned commit.
> 
> The new bin is OK
> 

great!

rkuijer, 2021-04-09:
Great catch!
I saw the changes but didn't link them to this issue.

arjenhiemstra, 2021-04-09:
> I saw the changes but didn't link them to this issue.

My neither but luckily time and effort solves all these kind of things :)

arjenhiemstra, 2021-04-09:
solved with 2a65bd6

closed 2021-04-09 by arjenhiemstra