arjenhiemstra/ithowifi #25: low on remote doesn't work

dworkin74, 2021-04-09:
After adding new remote the fan doesn't switch to low. In the GUI it shows as low but the fan is still at the previous setting

arjenhiemstra, 2021-04-09:
Ok, that is kinda strange... does changing the speed through the webinterface slider or API effectively change the speed to low?

dworkin74, 2021-04-09:
Yes, everything else is working fine.

dworkin74, 2021-04-09:
2021-04-09T18:00:31.813Z: RemoteID=116,216,207 / Command=timer3
Device type?: 22 - CMD: 142,34,243,3,99,0,30,168,127,255,118,115,205,167,132,239,51,175,245,201
2021-04-09T18:00:29.047Z: RemoteID=116,216,207 / Command=timer2
Device type?: 22 - CMD: 139,34,243,3,99,0,20,181,127,255,247,126,161,45,169,29,249,147,214,171
2021-04-09T18:00:26.154Z: RemoteID=116,216,207 / Command=timer1
Device type?: 22 - CMD: 137,34,243,3,99,0,10,193,127,255,186,142,97,51,104,132,236,19,238,12
2021-04-09T18:00:24.797Z: RemoteID=116,216,207 / Command=medium
Device type?: 22 - CMD: 136,34,241,3,99,3,4,199,127,255,187,142,252,235,183,6,206,51,44,200
2021-04-09T18:00:21.675Z: RemoteID=116,216,207 / Command=high
Device type?: 22 - CMD: 135,34,241,3,99,4,4,199,127,255,115,159,187,53,79,28,173,56,115,143
2021-04-09T17:58:42.754Z: RemoteID=116,216,207 / Command=low
Device type?: 22 - CMD: 134,34,241,3,99,2,4,202,127,255,57,248,187,253,15,8,250,72,68,164

2021-04-09T17:57:25.850Z: RemoteID=116,216,162 / Command=timer3
Device type?: 22 - CMD: 127,34,243,3,99,0,30,228,127,255,52,72,121,141,191,241,5,56,59,108
2021-04-09T17:57:25.114Z: RemoteID=116,216,162 / Command=timer2
Device type?: 22 - CMD: 126,34,243,3,99,0,20,239,127,255,191,152,199,14,23,90,21,214,0,191
2021-04-09T17:57:15.508Z: RemoteID=116,216,162 / Command=timer1
Device type?: 22 - CMD: 122,34,243,3,99,0,10,253,127,255,63,171,141,63,135,142,136,66,138,88
2021-04-09T17:56:56.769Z: RemoteID=116,216,162 / Command=medium
Device type?: 22 - CMD: 121,34,241,3,99,3,4,3,127,255,114,158,162,207,120,106,17,71,220,219
2021-04-09T17:56:52.914Z: RemoteID=116,216,162 / Command=high
Device type?: 22 - CMD: 120,34,241,3,99,4,4,3,127,255,49,253,173,92,238,225,121,114,247,17
2021-04-09T17:56:45.089Z: RemoteID=116,216,162 / Command=low
Device type?: 22 - CMD: 119,34,241,3,99,2,4,6,127,255,63,115,125,175,34,55,81,64,248,60

arjenhiemstra, 2021-04-09:
I'm really scratching my head here on this issue. The code that translates a remote command (low, med, high) to a speed setting on the itho unit is one and the same function.
Just to start of clean: 
1. could you please install the just released rc2 firmware
2. on the debug page trigger a file system format
3. after completion reboot the add-on
4. do not setup anything yet but activate RF support, reboot and pair the remotes to the add-on
5. test if everything works or if there are still issues
6. if everything works setup wifi, mqtt etc. and test if it then still works

This way I can rule out config issues, something on the network influencing the result etc.

dworkin74, 2021-04-10:
Hi Arjen,
That fixed the problem. Thanks for the help

arjenhiemstra, 2021-04-10:
Your welcome!

closed 2021-04-10 by arjenhiemstra