arjenhiemstra/ithowifi #26: CVD-S Eco connect, but MQTT not working

jwmaas, 2021-04-12:
I have put the Itho module into a CVD-S Eco roofventilator that has automatic sampling of CO2 and humidity. I put the ventilator on 2, which is automatic mode.
Through the module the API is accessible without any issues. Home Assistant reports the speedvalue without issue. However, the MQTT connection does not appear to work - the sensor does not exist in Home Assistant even though the API reports MQTT as "connected". I have four other MQTT devices and a broker running smoothly. So it is not a setup issue. Has anybody else run into this issue? How to solve it?

Many thanks.

arjenhiemstra, 2021-04-12:
Which firmware version are you using?

arjenhiemstra, 2021-04-12:
Ok cool!
Interesting to see that the module works with the CVD-S Eco as well! Thanks for that info!

closed 2021-04-12 by arjenhiemstra