arjenhiemstra/ithowifi #27: Board plugged CVD-S Eco does not update speed in API nor in Matt

jwmaas, 2021-04-13:
The board plugged into the Itho CVD-S Eco appears to work well. When plugged in I can log into its standard wifi SSID and set the SSID of my local network. The API by web is accessible on the local network. Switching on mqtt works, the current speed is reported as 20, both by mqtt and by rest call to the api.

However, the Itho is set on automatic mode (switch set to position 2). When detecting high humidity in the bathroom it automatically switches to a higher speed. However, this new speed is not picked up by the rest call, nor is it reported in the mqtt sensor (topic: itho/state). What could be the issue? Is it the humidity sensor which is blocking the updated info somehow?

Fanman76, 2021-04-13:
Same kind of issue here.

CVE eco. When I set the speed to high with the RF remotes (with direct link to the cve) the board seems to miss the status change. It still reports 20% both on the webinterface and the api.

Setting the speed with the interface board, via http or api, then reporting of the speed seems correct.

arjenhiemstra, 2021-04-13:
RV sensors and remotes have privilege over installed add-ons, this is itho designed behaviour. Furthermore the itho firmware does not report status back to add-on about these inputs.

The way around this:
for remotes - use the optional CC1101 module to handle the remote signals via the add-on instead of the itho main pcb 
humidity sensor - use the alternative humidity sensor I recently created (can be found on the tindie store incl. a use case description)

arjenhiemstra, 2021-08-30:
Closing stale issue

closed 2021-08-30 by arjenhiemstra