arjenhiemstra/ithowifi #29: MQTT Status: Connection Timeout

bkleef, 2021-07-05:
First of all, thank you for this awesome controller!

I can't seem to get MQTT working with Mosquitto 2.0.11 in Docker Compose on ARM64.

Mosquitto logs:

```
1625501614: mosquitto version 2.0.11 starting
1625501614: Config loaded from /mosquitto/config/mosquitto.conf.
1625501614: Opening ipv4 listen socket on port 1883.
1625501614: Opening ipv6 listen socket on port 1883.
1625501614: mosquitto version 2.0.11 running
1625501616: New connection from 192.168.1.111:63386 on port 1883.
1625501616: Client <unknown> disconnected due to protocol error.
1625501618: New connection from 192.168.1.41:6575 on port 1883.
1625501618: New client connected from 192.168.1.41:6575 as shellyswitch25-483FDA827188 (p2, c1, k60).
1625501620: New connection from 172.19.0.1:35916 on port 1883.
1625501620: New client connected from 172.19.0.1:35916 as 7FNNeLndNuWcyFsztiXyFE (p2, c1, k60).
1625501631: New connection from 192.168.1.111:63387 on port 1883.
1625501631: Client <unknown> disconnected due to protocol error.
```

Mosquitto config:

```
allow_anonymous true

listener 1883
protocol mqtt
```

Docker Compose file:

```yaml
version: "3.4"
services:
  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto:2.0.11
    ports:
      - "1883:1883"
    volumes:
      - ./configs/mosquitto/mosquitto.conf:/mosquitto/config/mosquitto.conf:ro
    restart: always
```

Itho WiFi controller firmware version is 2.2 and config:

```json
{
    "sys_username": "admin",
    "sys_password": "admin",
    "syssec_web": 0,
    "syssec_api": 0,
    "syssec_edit": 0,
    "syssht30": 1,
    "itho_rf_support": 1,
    "rfInitOK": true,
    "mqtt_active": 1,
    "mqtt_serverName": "192.168.1.180",
    "mqtt_username": "",
    "mqtt_password": "",
    "mqtt_port": 1883,
    "mqtt_version": 1,
    "mqtt_state_topic": "",
    "mqtt_sensor_topic": "",
    "mqtt_state_retain": "yes",
    "mqtt_cmd_topic": "",
    "mqtt_lwt_topic": "",
    "mqtt_domoticz_active": 0,
    "mqtt_ha_active": 0,
    "mqtt_ha_topic": "homeassistant",
    "mqtt_idx": 0,
    "sensor_idx": 0,
    "itho_fallback": 0,
    "itho_low": 0,
    "itho_medium": 120,
    "itho_high": 220,
    "itho_timer1": 10,
    "itho_timer2": 20,
    "itho_timer3": 30,
    "itho_sendjoin": 0,
    "itho_forcemedium": 0,
    "itho_vremapi": 0,
    "itho_vremswap": 0,
    "nonQ_cmd_clearsQ": 1,
    "version_of_program": "004"
}
```

System log shows:

```
MQTT: connection failed, System config: 1
```

Home Assistant 4.6 and Shelly devices are able to connect.

The controller is also unable to connect with test.mosquitto.org at port 1883 or encrypted port 8883. Then also shows the error: _MQTT Status_ _Connection Timeout_.

Any idea what's wrong?

arjenhiemstra, 2021-07-05:
No idea really, had it once myself. Reset of the system config did the trick.
Could you please try that (wifi settings will be left unaffected)

If that doesn't work, a format of the file system might something to try but you will have to setup you wifi network after that.

bkleef, 2021-07-05:
Reset of system config + reboot fixed it! 🍻

Current config:

```json
{
    "sys_username": "admin",
    "sys_password": "admin",
    "syssec_web": 0,
    "syssec_api": 0,
    "syssec_edit": 0,
    "syssht30": 1,
    "itho_rf_support": 1,
    "rfInitOK": false,
    "mqtt_active": 1,
    "mqtt_serverName": "192.168.1.180",
    "mqtt_username": "",
    "mqtt_password": "",
    "mqtt_port": 1883,
    "mqtt_version": 1,
    "mqtt_state_topic": "itho/state",
    "mqtt_sensor_topic": "itho/sensor",
    "mqtt_state_retain": "yes",
    "mqtt_cmd_topic": "itho/cmd",
    "mqtt_lwt_topic": "itho/lwt",
    "mqtt_domoticz_active": 0,
    "mqtt_ha_active": 0,
    "mqtt_ha_topic": "homeassistant",
    "mqtt_idx": 1,
    "sensor_idx": 1,
    "itho_fallback": 20,
    "itho_low": 20,
    "itho_medium": 120,
    "itho_high": 220,
    "itho_timer1": 10,
    "itho_timer2": 20,
    "itho_timer3": 30,
    "itho_sendjoin": 0,
    "itho_forcemedium": 0,
    "itho_vremapi": 0,
    "itho_vremswap": 0,
    "nonQ_cmd_clearsQ": 1,
    "version_of_program": "004"
}
```

closed 2021-07-05 by bkleef