arjenhiemstra/ithowifi #30: IthoWifi won't connect to WiFi automatically after power cut

ezwep, 2021-07-08:
When I interrupt the power from the fuse box, the device does not automatically connect to my wifi. I first have to connect to the Ithowifi module and then perform a restart from the web interface. Then it will work.

TD-er, 2021-07-08:
Is something connected to GPIO-0 ?

arjenhiemstra, 2021-07-08:
> When I interrupt the power from the fuse box, the device does not automatically connect to my wifi. I first have to connect to the Ithowifi module and then perform a restart from the web interface. Then it will work.

What version of the PCB do you have? (picture will also work cause the version number is on the back of the pcb)

arjenhiemstra, 2021-07-08:
> Is something connected to GPIO-0 ?

Sure is! It is connected to a pull up and the DTR pin through a zener diode / cap combo with the EN pin for auto reset during upload.
[image]

TD-er, 2021-07-08:
Maybe the value of C2 is a bit low to delay the ESP from booting before power stabilized?
Or maybe the DTR line is low during boot?

arjenhiemstra, 2021-07-08:
Good point but then it would not start an AP but will stay in "waiting for download" mode

TD-er, 2021-07-09:
Well there is a way this may also cause issues to connect to WiFi.
My experience for these kinds of issues is mainly based on ESP8266, but the ESP32 does behave similar.

What happens at boot is that the ESP does perform some WiFi calibration, which is partly based on the 3v3 supplied.
This to make sure it can operate on WiFi even with partly depleted batteries or a power supply which does have some offset to the default 3v3.
If this calibration is based on the wrong (not yet stable) 3v3, then the ESP may not have reliable/stable WiFi.
This means it can not create a connection to an AP.

This calibration can be done in various steps, or even be disabled at boot.
If memory serves me well, the default is the full RF calibration, which does also take quite a bit of current, thus putting extra strain on the not yet stable Vcc.

Psyveira, 2021-07-16:
I have exactly the same problem. I contacted @arjenhiemstra about this trough e-mail and I said I would create an issue on GitHub. I did not have any spare time creating the issue yet until today. But then I noticed the issue was already created by @ezwep.

@arjenhiemstra my Tindie order number was #271204, so I guess you can lookup what version I have. The web interface say the following:
` Current firmware version: 2.2
Hardware revision: 2 `

My log file says the following. The first boot shows it's unable to connect to WiFi. Then I connected to the AP Wifi and requested a reboot using the web interface.  As you can see the second boot connects to WiFi successfully.

`2021-06-18 13:07:10 N: System boot, last reset reason: SDIO_RESET
2021-06-18 13:07:10 N: HW rev: 2, FW ver.: 2.2
2021-06-18 13:07:11 N: Static IP config OK
2021-06-18 13:07:41 N: Setup: Wifi connect STA failed
2021-06-18 13:07:43 N: wifi AP mode started
2021-06-18 13:07:43 N: Setup: AP mode active
2021-06-18 13:07:43 N: Webserver: started
2021-06-18 13:07:43 N: Setup: init of CC1101 RF module successful
2021-06-18 13:07:43 N: mDNS: started
2021-06-18 13:07:43 N: Hostname: nrg-itho-e9a0
2021-06-18 13:07:43 N: Setup: done
2021-06-18 13:22:43 N: Reboot requested
2021-06-18 13:22:44 N: System boot, last reset reason: SDIO_RESET
2021-06-18 13:22:44 N: HW rev: 2, FW ver.: 2.2
2021-06-18 13:22:44 N: Static IP config OK
2021-06-18 13:22:48 N: WiFi: connection successful
2021-06-18 13:22:48 N: WiFi info:
2021-06-18 13:22:48 N: Mode:STA
2021-06-18 13:22:48 N: Status:3
2021-06-18 13:22:48 N: IP:10.2.0.15`

arjenhiemstra, 2021-07-16:
I'll conduct some test to see if I can reproduce it. In the meanwhile I'll work on a software workaround.

arjenhiemstra, 2021-07-18:
I've created a simple workaround whilst investigation potential hardware based glitches. 
Could you try [Version 2.2.1](https://github.com/arjenhiemstra/ithowifi/blob/2.2/compiled_firmware_files/hardware_rev_2/nrgitho-hw2-v2.2.1.bin)

Psyveira, 2021-07-18:
I just upgraded to 2.2.1, but it seems the problem still exists.
Is there anything else I need to do after the upgrade?

```
2021-07-18 19:57:16 N: Firmware update: nrgitho-hw2-v2.2.1.bin
2021-07-18 19:57:30 N: Reboot requested
2021-07-19 01:02:00 N: System boot, last reset reason: SDIO_RESET
2021-07-19 01:02:00 N: HW rev: 2, FW ver.: 2.2.1
2021-07-19 01:02:01 N: Static IP config OK
2021-07-19 01:02:31 N: Setup: Wifi connect STA failed
2021-07-19 01:02:33 N: wifi AP mode started
2021-07-19 01:02:33 N: Setup: AP mode active
2021-07-19 01:02:33 N: Webserver: started
2021-07-19 01:02:33 N: Setup: init of CC1101 RF module successful
2021-07-19 01:02:33 N: mDNS: started
2021-07-19 01:02:33 N: Hostname: nrg-itho-e9a0
2021-07-19 01:02:33 N: Setup: done
2021-07-19 01:04:59 N: Reboot requested
2021-07-19 01:05:02 N: System boot, last reset reason: SDIO_RESET
2021-07-19 01:05:02 N: HW rev: 2, FW ver.: 2.2.1
2021-07-19 01:05:02 N: Static IP config OK
2021-07-19 01:05:05 N: WiFi: connection successful
2021-07-19 01:05:05 N: WiFi info:
2021-07-19 01:05:05 N: Mode:STA
2021-07-19 01:05:05 N: Status:3
2021-07-19 01:05:05 N: IP:10.2.0.15
2021-07-19 01:05:05 N: Setup: Virtual remote ID: 183,233,160
2021-07-19 01:05:06 N: Webserver: started
2021-07-18 20:00:42 N: Setup: init of CC1101 RF module successful
2021-07-18 20:00:42 N: mDNS: started
2021-07-18 20:00:42 N: Hostname: nrg-itho-e9a0
2021-07-18 20:00:42 N: Setup: done
```

arjenhiemstra, 2021-07-19:
Ok, used reset method might have an influence... (WDT reset, documentation is not clear about this)  Here with a different method, basically the usual and same method used through the user interface:
[nrgitho-hw2-v2.2.1.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/6843297/nrgitho-hw2-v2.2.1.bin.zip)

Basically what this does is boot, wait 2 seconds (to stabilise power if that's an issue) and trigger an extra reboot. This happens only on power-on, not on a reset through the user interface. If this 'fixes' the issue I would very much like to swap the add-on so that I can examine the one you have.

If this doesn't work another workaround might be to reboot if no connection can be made on a cold boot whilst having a wifi network configured but that would be a later in the boot process.

Psyveira, 2021-07-19:
No difference with that firmware after the upgrade I had to manually restart it to make the WiFi reconnect.

I did try powering my ventilation unit off and on, and that fix does seem to work. However it is pretty slow. At first I gave up thinking it would not work. But since I was busy doing something else I did not manually reboot the board that time. And a few minutes later after refreshing again the board was up and running. So the fix does work after a cold boot.

Here is the log file for what it's worth:
```
     34648 N: wifi AP mode started
     34669 N: Setup: AP mode active
     34745 N: Webserver: started
     34765 N: Setup: init of CC1101 RF module successful
     34805 N: mDNS: started
     34824 N: Hostname: nrg-itho-e9a0
     34897 N: Setup: done
    934655 N: Reboot requested
      2159 N: System boot, last reset reason: SDIO_RESET
      2177 N: HW rev: 2, FW ver.: 2.2.1
      2542 N: Static IP config OK
      5715 N: WiFi: connection successful
      5734 N: WiFi info:
      5752 N: Mode:STA
      5771 N: Status:3
      5839 N: IP:10.2.0.15
2021-07-19 20:19:42 N: Setup: Virtual remote ID: 183,233,160
2021-07-19 20:19:42 N: Webserver: started
2021-07-19 20:19:42 N: Setup: init of CC1101 RF module successful
2021-07-19 20:19:42 N: mDNS: started
2021-07-19 20:19:42 N: Hostname: nrg-itho-e9a0
2021-07-19 20:19:42 N: Setup: done

```

As for fixing the issue when you are doing a manual (soft) reboot:

- After boot: if `AP mode` is active check if there are any WiFi settings configured.
  - Yes: Then reboot 1 extra time to make the "correct" boot happen.
  - No: Then do nothing and stay in `AP mode`
Please be sure to only do 1 extra boot. Else you might end up with an endless loop in case the configured WiFi settings are incorrect.

arjenhiemstra, 2021-07-19:
Thanks for trying!

>So the fix does work after a cold boot.

The fix has been designed to work only on a cold boot (that should be the only moment that the power supply of the itho could be unstable) so that seems like the correct behaviour.

>And a few minutes later after refreshing again the board was up and running.

Strange but not uncommon in some situations due to various possible reasons. 
For a better test you could try to let a ping running and look how long it takes to get it reply again.

>As for fixing the issue when you are doing a manual (soft) reboot

So if I understand correctly, you also experience this issue after of soft reboot through the user interface? Did I incorrectly understood that this was the way you solved the issue in the past?

Psyveira, 2021-07-27:
> So if I understand correctly, you also experience this issue after of soft reboot through the user interface? Did I incorrectly understood that this was the way you solved the issue in the past?

That's correct. I have this problem every time I reboot the addon using the user interface. So not only during a cold-boot.

I tried it again today after changing the IP config from static to DHCP (added a static binding on my router). I pressed reboot in the user interface and boom: unreachable. I had to connect to the AP using my phone to reboot it a second time to get it back up.

Could it just be a software related problem? Since there is no possible power spike of the Itho starting up when doing a soft-reboot of the addon.