arjenhiemstra/ithowifi #34: Incompatibility with iOS 15.0 and Safari on MacOS Monterey Beta B7

arjenhiemstra, 2021-09-22:
the websocket is not functioning properly on these versions, on iOS it seems totally unusable, on MacOS chrome seems to be working fine and after a few minutes Safari seems to get restored and functions properly until the next reload of the browser window.

Error message:
`[Error] WebSocket connection to 'ws://nrg-itho-1234.local/ws' failed: The operation couldn’t be completed. (kNWErrorDomainPOSIX error 54 - Connection reset by peer)`

arjenhiemstra, 2021-09-29:
On iOS:
System setting=>safari =>Advanced=>Experimental Features=>"NSURLSession WebSocket" is default On. 
Switching it to off significantly improved the experience. Most seems to work fine after.

arjenhiemstra, 2021-11-01:
I looks like it has been fixed with the latest beta4 firmware version. Some testing would be nice.

jasperslits, 2021-11-02:
Confirmed it works on IOS 15 with iPad with workaround reverted. On MacOS Monterey it works with Safari now too

jasperslits, 2021-11-25:
@arjenhiemstra : this should be closed. It works fine on all apple devices without workarounds

closed 2021-11-25 by arjenhiemstra