arjenhiemstra/ithowifi #35: security and other features

mr4hughz, 2021-09-23:
Please make the hostname/domain name editable, 
mqtt hostname "generic" instead of itho/<item> itho<xxxx>/
and tls/username password option available (if not already) 

Username/passwords for website (access), admin, and such changeable both, 
it would be better to either upload a "config" file with all settings or .. a 1 time "wizard"  that sets all options  
- wifi SSID/AP 
- username (custom) and custom pw (all access) 
- checked "options"  like checked boxes for "common" setup 
> - enable rf-join, temp/hum sensor etc 

later-on config "details" 
- max of 10 users each with all options to read / write certain functions 

The reason to raise this issue is that defaults can lead to security issues, setting a custom username/password mitigates this issue  a bit.  uploading a custom config file makes it easy for testing (wiki on how to automate on detecting this default 1st setup is nice for home automation. 

ps expose also via mqtt the firmware / and a checkbox for update is available check (x.x and 2.x )

closed 2024-01-17 by arjenhiemstra

retorquere, 2025-11-15:
Does the dongle support tls-secured mqtt?

arjenhiemstra, 2025-11-15:
You should be able to connect to any TLS server. But be aware that the CA cert chain is not part of the firmware due to size restrictions. The cert check is therefore skipped.