arjenhiemstra/ithowifi #43: retrieving unit's settings while multiple browser sessions opened leads to a crash

rplevka, 2021-10-29:
reproducer:
1. navigate to the addon's url in 2 separate browser sessions (opening browser console in both of them helps)
2. n one of the sessions, navigate to `Itho settings` and hit `Retrieve settings`
3. observe the websocket events never stop flowing in
4. get off your bed and go to reset your board manually

it looks like a forkbomb, where new ws connections are being spawned indefinitely, probably consuming all the memory so the sketch crashes.
Eventually, the board fails to reboot properly but that might be another bug.

I think as a new row is spawned, in 1 session, it replicates to other session as well together with a new ws connection, which is in turn being replicated in the first session, etc.,

arjenhiemstra, 2021-10-29:
I've seen this as well, the loop is not handled correctly if multiple WS sessions start to pick it up... need to find a solution for this indeed

arjenhiemstra, 2022-03-02:
should be fixed with 
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha2

closed 2022-03-12 by arjenhiemstra