arjenhiemstra/ithowifi #44: Firmware update - Version check not correct due to firmware.json outdated

jasperslits, 2021-10-30:
Small problem. I am having the non-CVE version. 

The update page shows: 

Current firmware version: 2.3-beta3
Hardware revision: NON-CVE 1
Latest firmware version: 2.3-beta2

<Download firmware file> button points to https://github.com/arjenhiemstra/ithowifi/raw/master/compiled_firmware_files/non-cve_rev_1/nrgitho-noncve-v2.3-beta3.bin

Upon checking the code, the data comes from https://raw.githubusercontent.com/arjenhiemstra/ithowifi/master/compiled_firmware_files/firmware.json has this:

 "NON-CVE 1": {
      "latest_fw":"2.3-beta2",
      "link":"https://github.com/arjenhiemstra/ithowifi/raw/master/compiled_firmware_files/non-cve_rev_1/nrgitho-noncve-v2.3-beta3.bin"
    }

So perhaps it's cleaner to have latest_fw referred to that in the link?  E.g. https://github.com/arjenhiemstra/ithowifi/raw/master/compiled_firmware_files/non-cve_rev_1/nrgitho-noncve-{{latest_fw}}.bin so it avoids double maintenance for new releases.

arjenhiemstra, 2021-10-30:
thanks for the suggestion, will change that

jasperslits, 2021-11-02:
Same problem with the new non-CVE release with fw_version and bin link.

jasperslits, 2021-11-02:
Confirmed it's OK now

closed 2021-11-02 by jasperslits