arjenhiemstra/ithowifi #45: /edit option does not show existing files (ESP Editor)

jasperslits, 2021-11-01:
I was doing some checks against Wiki and documentation and wanted to see if there was anything in remotes.json 

https://github.com/arjenhiemstra/ithowifi/tree/master/remotes (also available in wiki as https://github.com/arjenhiemstra/ithowifi/wiki/Remotes) refers to an /edit link to update remotes.json

Step to reproduce on latest beta (2.3-beta3) as per Wiki:  

`
Go to: http://[nrg-itho-A1B2]/edit (username/pwd both admin)
`

Observed issues:
1) There is no prompt for credentials
So I guess it's sufficient to change the Wiki as the prompt for File Editor only shows up if the "File Editor authentication" is set to "On" in "System Settings". 
But for the non-CVE the prompt still doesn't show but I couldnt tell from 09_init_code.ino where this is coming from and if systemConfig.syssec_edit is empty somehow..

2) No files are available in the left menu

Upon debugging, I noticed that it upon loading the page it calls /edit?edit=%2F/index.htm which returns a 404

3) "Refresh List" does not do anything (probably connected to the previous item)
This calls edit?list=%F which returns [] with status code 200. 

This shows up with MacOS 12 / Chrome and Windows / Chrome. I cannot test it with another browser.

This is obviously a low priority item.

arjenhiemstra, 2021-11-01:
1. There is no prompt for credentials
I cannot reproduce this (I notice though that you have to reboot before the credentials are asked)

2.
I'm not sure what this is. I did some testing and I have now 1 device that does not show files and 1 that does. Same FW version.... no idea why

3.
index.htm is a file that doesn't exist, so that is correct.

arjenhiemstra, 2021-11-01:
Interesting. I did a file system format and after that the file system editor shows files again. There is definitely some bug or error here. What that is.... no idea at this point

jasperslits, 2021-11-01:
“There is no prompt for credentials
I cannot reproduce this (I notice though that you have to reboot before the credentials are asked)“.

I enabled it without rebooting because I expected an reboot message when changing it if reboot was needed. So it could either be mentioned on the web interface itself as a comment or do you want to change it? 

I think it is more a nice to have as I expect those with the add-on to use it within a domestic setting on a trusted WiFi network only. The number of users to use this password protection to be very low anyway. 

Is there today any other reason to use the /edit than to modify the remotes.json file?

rplevka, 2021-11-02:
the 404 on `/edit?edit=%2F/index.htm` is reproducible and is caused by the implicit behavior of the upstream lib:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/2f784268f0a358741ee6384480d48656e159d726/src/edit.htm#L605
Also, it's most probably not related to other issues you're seeing.

the easiest solution/workaround would be to update `src` [here](https://github.com/arjenhiemstra/ithowifi/blob/a985ad0ddba9d1800f84376f21286daa95eb09e3/software/NRG_itho_wifi/02_JS_UI.ino#L1469) to e.g. `/edit?edit=%2F/config.json`

rplevka, 2021-11-02:
i also agree that the password-protected file editor makes little to no sense in the current implementation - since it uses the same creds as the web ui, everybody who knows web ui credentials ultimately knows creds to file editor, so I can't think of any scenario where this would be useful other than having multiple user support (regular user and admin).

arjenhiemstra, 2021-11-02:
Don't know, it was a feature request and imho there is no real issue with having these password options. Implementation was a 5 min. effort and it never failed afterwards. I don't use it but I know some users do.
As far as I can reproduce a formate and reinstall of the add-on solves the no files showing up issue. If that can be confirmed it is probably due to some flash issues (maybe arising due to crashes while writing flash?) difficult to say, even more difficult to investigate i'm afraid.

jasperslits, 2021-11-03:
So I guess the easiest is to implement the suggestion from @rplevka and use /edit?edit=%2F/config.json or %2F/remotes.json as landing page and leave everything as-is? 
It's already highly unusual for anyone to use /edit and if you use it, it's for remotes.json only?

arjenhiemstra, 2021-11-03:
No, i'm not going to change the code in this case. The 404 is normally not visible and the file editor will just show all the files. The reason that this issue is still left open is because I'm not sure where the the not showing the files on the flash partition is originating from. 
I'm sure tho it has nothing to do `/edit?edit=%2Findex.htm`. 
Please also be aware that '%2F' in this case just is the HTML URL encoded representation for the '/' character.

jasperslits, 2021-11-15:
A format and reconfigure made the problems go away. So I can close this one as I guess finding the root cause for the corruption that caused the problems has to be a lucky shot.

closed 2021-11-15 by arjenhiemstra