arjenhiemstra/ithowifi #46: Countdown counter mix up with minutes and seconds

jasperslits, 2021-11-01:
I am too lazy to make this a proper PR. 

**Problem**
When making changes to the web interface page, some changes require a restart. The count down is less than a minute and so is using the 'restart device button'. 
The page shows "This page will reload to the start page in 00:20 seconds". 
This should be either " in 20 seconds" or "00:20 minutes".  

var message = 'This page will reload to the start page in ' + min.toString() + ':' + sec + ' seconds...';

Upon checking the code, it seems secondsRemaining is always set to 20 it seems cleaner to just show "20 seconds" by removing min.toString() all together.

arjenhiemstra, 2021-11-01:
thanks, this is probably part of the oldest code in there, didn't know anything about javascript back then (still don't ;-))
I'll change it the next version

arjenhiemstra, 2021-11-01:
fixed with https://github.com/arjenhiemstra/ithowifi/commit/a985ad0ddba9d1800f84376f21286daa95eb09e3

closed 2021-11-01 by arjenhiemstra