arjenhiemstra/ithowifi #48: MQTT lost last 2 ip address octets

evertide, 2021-11-05:
I am running the latest 2.3-beta4 beta but it happened in the previous beta3 as well.

After a reset or the MQTT last 2 octets disappear. After reentering the config MQTT connects again.

PS.
2 feature requests
1- Reset thru MQTT/API to keep the config fresh. 
2- If Domoticz is is used in MQTT keep the other MQTT topic as well. I know I can use node-red but an easy Domoticz topic is higher appreciated especially with the original MQTT topic for some other automation things

arjenhiemstra, 2021-11-16:
I believe I do not really understand the issue. When is this happening? After a reset of the config, the module or something else? Could you share a screenshot?

arjenhiemstra, 2021-11-30:
[STALE ISSUE]

closed 2021-11-30 by arjenhiemstra