arjenhiemstra/ithowifi #52: Module keeps crashing

a3426rage, 2021-11-17:
Module keeps disconnecting from Wi-Fi and gets Warning: I2C timeout's during the day. Have to hard-reset the module.

Current firmware version: 2.3-beta4
Hardware revision: 2

`2021-11-16 20:32:29 N: Setup: SHT30 sensor not present
2021-11-16 20:32:29 N: Setup: init of CC1101 RF module successful
2021-11-16 20:32:30 N: Warning: I2C timeout
2021-11-16 20:32:31 N: MQTT: connected, System config: 1
2021-11-16 20:32:31 N: Webserver: started
2021-11-16 20:32:31 N: mDNS: started
2021-11-16 20:32:31 N: Hostname: nrg-itho-0e58
2021-11-16 20:32:31 N: Setup: done
2021-11-16 20:33:09 N: Error: Task SysControl timed out!
2021-11-16 20:33:17 N: System boot, last reset reason: SDIO_RESET
2021-11-16 20:33:17 N: HW rev: 2, FW ver.: 2.3-beta4
2021-11-16 20:33:20 N: WiFi: connection successful
2021-11-16 20:33:20 N: WiFi info:
2021-11-16 20:33:21 N: Mode:STA
2021-11-16 20:33:21 N: Status:3
2021-11-16 20:33:21 N: IP:192.168.2.65
2021-11-16 20:33:21 N: Setup: Virtual remote ID: 155,14,88
2021-11-16 20:33:25 N: Setup: SHT30 sensor not present
2021-11-16 20:33:25 N: Setup: init of CC1101 RF module successful
2021-11-16 20:33:26 N: Warning: I2C timeout
2021-11-16 20:33:26 N: MQTT: connected, System config: 1
2021-11-16 20:33:26 N: Webserver: started
2021-11-16 20:33:27 N: mDNS: started
2021-11-16 20:33:27 N: Hostname: nrg-itho-0e58
2021-11-16 20:33:27 N: Setup: done`

Please let me know if there is more specific info needed.

arjenhiemstra, 2021-11-17:
Could you provide the remaining info?

Subject "[hw/sw] short description"
Issue description (short) :
Current Firmware version: 2.3-beta4
Current Hardware revision (backside addon) : 
Type/model number Itho (on sticker outside the box):
IP [static or DHCP]:
MQTT in use [yes/no]: yes
HTTP API in use [yes/no]: 
CC1101 RF module enabled [yes/no]: yes
Restart through webinterface resolves issue (temporary) [yes/no]: 
Power cycle Itho resolves issue (temporary) [yes/no]:
In case of wifi issues, wifi access point make and model:

a3426rage, 2021-11-18:
Subject "[hw/sw] short description" 
Issue description (short) : Module keeps getting timeouts on I2C bus
Current Firmware version: 2.3-beta4
Current Hardware revision (backside addon) : v2
Type/model number Itho (on sticker outside the box): Itho daalderop CVE-S ECO
IP [static or DHCP]: DHCP
MQTT in use [yes/no]: yes
HTTP API in use [yes/no]: No
CC1101 RF module enabled [yes/no]: yes
Restart through webinterface resolves issue (temporary) [yes/no]: No
Power cycle Itho resolves issue (temporary) [yes/no]: Yes
In case of wifi issues, wifi access point make and model: Ubiquiti UAP-AC-Lite

arjenhiemstra, 2021-11-18:
What firmware version do you run on your Ubiquiti?

a3426rage, 2021-11-18:
I'm running: UniFi OS UDM Pro 1.11.0-20 with network version 6.5.52
UAP-AP Lite running: Firmware Version 5.43.51

arjenhiemstra, 2021-11-18:
And futhermore, could you share the config.json file if possible? (you can edit the file and remove passwords or scramble any sensitive info if needed)
You can download this file through http://[IP or DNS]/edit (when connected to the wifi network, not in AP mode)

a3426rage, 2021-11-18:
[image]
Seems empty?

arjenhiemstra, 2021-11-18:
That's not ok. This indicates a corrupted flash partition.
A possible way to solve this is to format the flash drive, this can be done through the menu reset. The module will then restart and start as AP again after which you will have to do the setup of the module again.

a3426rage, 2021-11-18:
By format you mean: Format Filesystem?

[image]

arjenhiemstra, 2021-11-18:
Yes indeed

a3426rage, 2021-11-18:
After the format filesystem the /edit page is still empty. It's looking for /index.htm. 

[image]

I'll check if this had helped for the I2C connection issues with the fan.

a3426rage, 2021-11-19:
Issue with the i2c connection is gone. No more errors while controlling the fan.

closed 2021-11-19 by a3426rage

arjenhiemstra, 2021-11-19:
Good! But still strange that you cannot see the files.
If the issue pops up again, please let me know