arjenhiemstra/ithowifi #54: HRU 350 Settings field misalignment

jasperslits, 2021-11-24:
Latest firmware and HRU-350

Seems around position 34 the field labels and values no longer match and seem off by one.
itho status seems OK, the settings not. 

best illustrated here: 

  |   |   |   |   |  
-- | -- | -- | -- | -- | --
  | 77 | PoorCo2Quality (ppm) | 1 | 0 | 1
  | 78 | GoodCo2Quality (ppm) | 1600 | 1501 | 2000
  | 79 | Inhabitants (Inhab) | 1400 | 500 | 1500
  | 80 | Night min vent speed 1 person (%) | 2 | 1 | 2

Seems to start around index 35 but hoping you still have the HRU 350 yourself, is this something you can reproduce?

jasperslits, 2021-11-24:
In the MQTT payload I see this too:
```
{"temp":0,"hum":0,"ppmw":0,"Requested fanspeed (%)":6552.6,"Balance (%)":149.9,"Supply fan (rpm)":1142,"Supply fan actual (rpm)":1144,"Exhaust fan (rpm)":762,"Exhaust fan actual (rpm)":762,"Supply temp (°C)":13.79,"Exhaust temp (°C)":8.56,"Status":0,"Room temp (°C)":13.79,"Outdoor temp (°C)":8.56,"Valve position":0,"Bypass position":0,"Summercounter":0,"Summerday (K_min)":0,"Frost timer":0,"Boiler timer":177,"Frost block":120,"Current position":0,"VKKswitch":0,"GHEswitch":0,"Airfilter counter":2960,"Global fault code":0,"Actual Mode":24,"Pir fan speed level":-1,"Highest received CO2 value (ppm)":592,"Highest received RH value (% RH)":239,"Air Quality (%)":100,"Remaining override timer (sec)":0,"Fallback speed timer (sec)":0,"label out of bound error":0}
```

"label out of bound" is not something I recall seeing before and grepping for this in the code doesn't return anything. Not sure if it's related to the misalignment? It shows up under "status" and not "settings".

arjenhiemstra, 2021-11-24:
ohoh.... I'm afraid I know what is the issue. I did an export out of the DB but left some settings with high password levels out. The settings are handled on index in the correct order. If you leave settings out the index does not match the index of the settings anymore... stupid
Two solutions:
1- include all settings of supported devices
2- create a index, setting map per version instead of just a simple array

I guess the first is the easiest method

The out of bound error indicates that somehow the length returned by the itho query is longer than there are status mappings in the array. Something might be missing there also I'm afraid.

arjenhiemstra, 2021-11-24:
[NRG_itho_wifi.ino.esp32.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/7599024/NRG_itho_wifi.ino.esp32.bin.zip)
updates the mapping

jasperslits, 2021-11-25:
Not much of an improvement:

1) Status screen shows out of bounds at the bottom
2) Status & Settings show technical codes: e.g. exhaust-temp_c
3) Current,min,max for settings are no displayed. 
4) Index sequence for settings is wrong, are duplicated. 

If you plan to remove spaces and other formatting, please do this only for the MQTT payload and not for the values displayed on the screen. 
```JSON
"{\"ithosettings\":{\"Index\":21,\"Description\":\"low-frequency-var-pwm_khz\",\"update\":false,\"loop\":true,\"Current\":null,\"Minimum\":null,\"Maximum\":null}}"
```

arjenhiemstra, 2021-11-25:
could you provide a screenshot?

jasperslits, 2021-11-25:
See [https://github.com/jasperslits/ithowiki_content/blob/master/hru_settings.PNG](url) and [https://github.com/jasperslits/ithowiki_content/blob/master/hru_status.PNG](url)

arjenhiemstra, 2021-11-29:
[nrgitho-noncve-v2.3-beta7.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/7616908/nrgitho-noncve-v2.3-beta7.bin.zip)

I have tweaked the script to create the settings and status scripts further and found some DB issues which caused labels to drop out. 
I'm not able to check wether the status labels and settings labels are correct for version 3. It would be great if you could post the data from both the status page and the settings page in order for me to check whether the contents are as expected.

Not that this upgrade will format the flash drive and alle settings will be reset.

arjenhiemstra, 2021-11-30:
implemented with version 2.3.0
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.3.0

closed 2021-11-30 by arjenhiemstra