arjenhiemstra/ithowifi #55: HRU 350 firmware v4 support

janbredenbeek, 2021-11-24:
I have a HRU 350 with firmware v4 and a 'retrieve settings' command returns an error 'Message: Settings not available for this device or its firmware version'. Also status information doesn't return any useful results.

arjenhiemstra, 2021-11-24:
[NRG_itho_wifi.ino.esp32.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/7599018/NRG_itho_wifi.ino.esp32.bin.zip)
Could you test this firmware?

janbredenbeek, 2021-11-25:
I do get status information now (thanks!)
When retrieving settings I do get the descriptions but no values (only the 3 bullets), and then it seems to lose connection.
Haven't tested the remotes yet (I have touch key remotes, one Co2 and one RH).

arjenhiemstra, 2021-11-25:
I heard the same from Jasper in this issue:
https://github.com/arjenhiemstra/ithowifi/issues/54

I need to dive in a bit further to find out what is happing. Thanks for testing!

jasperslits, 2021-11-29:
Copy of setting current values didn't quite work but they look OK. 
For status 'label out of bound error' is still there. 

**Status:**

Label | Value
---- | ----
Requested fanspeed (%) | 6552.6
Balance (%) | 149.8
Supply fan (RPM) | 1141
Supply fan actual (RPM) | 1148
Exhaust fan (RPM) | 762
Exhaust fan actual (RPM) | 768
Supply temp (°C) | 11.65
Exhaust temp (°C) | 4.96
Status | 0
Room temp (°C) | 11.65
Outdoor temp (°C) | 4.96
Valve position | 0
Bypass position | 0
Summercounter | 0
Summerday (K_min) | 0
Frost timer | 0
Boiler timer | 177
Frost block | 120
Current position | 0
VKKswitch | 0
GHEswitch | 0
Airfilter counter | 3040
Global fault code | 0
Actual Mode | 24
Pir fan speed level | -1
Highest received CO2 value (Ppm) | 436
Highest received RH value (%RH) | 239
Air Quality (%) | 100
Remaining override timer (Sec) | 0
Fallback speed timer (Sec) | 0
label out of bound error | 0

**Settings**

index|label|current|min|max
-- | -- | -- | -- | --  
 0 | Percentage set 2 (%) |   | 25 | 60 
1 | Temp frost protection (°C) || 0 | 7 
2 | Frost temp offset (°C) || 5 | 25 
3 | Min frost speed (%) || 0 | 25 
4 | Max frost speed (%) || 10 | 30 
5 | Frost supply fan off time (Sec) || 60 | 7200 
6 | Frost supply fan on time (Sec) || 60 | 1500 
7 | Frost speed change (%) || 1 | 50 
8 | Frost speed change (%) || 0 | 0 
9 | Max ratio exhaust - supply during frost protection (%) || 100 | 200 
10 | Summer temp (°C) || 15 | 25 
11 | Summer temp GHE (°C) || 12 | 25 
12 | Summer day time (K*hr) || 5 | 15 
13 | Wanted temp bypass regulation (°C) || 15 | 30 
14 | Offset bypass regulation (K) || 1 | 50 
15 | Max bypass open time (hrs) || 24 | 240 
16 | Max rpm fast var. Pwm (RPM) || 1000 | 2000 
17 | Max rpm slow var. Pwm (RPM) || 1500 | 2500 
18 | Max rpm fast fixed pwm (RPM) || 2000 | 2800 
19 | Speed offset (RPM) || 15 | 75 
20 | High frequency var pwm (kHz) || 15 | 20 
21 | Low frequency var pwm (kHz) || 10 | 18 
22 | High frequency fixed pwm (kHz) || 5 | 10 
23 | Low frequentie fixed pwm (kHz) || 1 | 4 
24 | Start speed vkk exhaust fan (%) || 0 | 10 
25 | Vkk valve close time (Sec) || 10 | 180 
26 | Rpm exhaust fan with min position potmeter low (RPM) || 500 | 1500 
[... 57 lines cut ...]
84 | Night min vent optima 1 multi-floor 1 person (%) || 0 | 100 
85 | Night min vent optima 1 multi-floor 2 persons (%) || 0 | 100 
86 | Night min vent optima 1 multi-floor 3+ persons (%) || 0 | 100 
87 | Night min vent optima 1 one-floor 1 person (%) || 0 | 100 
88 | Night min vent optima 1 one-floor 2 persons (%) || 0 | 100 
89 | Night min vent optima 1 one-floor 3+ persons (%) || 0 | 100 
90 | Night min vent optima 2 multi-floor 1 person (%) || 0 | 100 
91 | Night min vent optima 2 multi-floor 2 persons (%) || 0 | 100 
92 | Night min vent optima 2 multi-floor 3+ persons (%) || 0 | 100 
93 | Night min vent optima 2 one-floor 1 person (%) || 0 | 100 
94 | Night min vent optima 2 one-floor 2 persons (%) || 0 | 100 
95 | Night min vent optima 2 one-floor 3+ persons (%) || 0 | 0 
96 | Summer night boost enable (-) || 0 | 1 
97 | Summer night degree (-) || 0 | 100 
98 | Summer night setpoint (°C) || 15 | 35 
99 | Summer night boost ventilation level (%) || 30 | 100 
100 | Summer night max area degree hours (-) || 0 | -0.1 
101 | FrostTimer (sec) || 0 | 9999 
102 | StartupCounter (sec) || 0 | 125 
103 | BoilerTestTimer (min) || 0 | 360 
104 | SummerCounter (K_min) || 0 | 900 
105 | BypassTimer (min) || 0 | 20 
106 | MaximumOpenCounter (min) || 0 | 14400 
107 | FilterUsagePerMin (-) || 0 | 6001 
108 | Balance (%) || 0 | 200

arjenhiemstra, 2021-11-29:
Labels and settings do match with the DB, so that is fine. The out of bound error should mean that the length read from the status format query is different from what mapping, which is strange cause it seems that the result does match.

Could you post the result of the status format query?

jasperslits, 2021-11-29:
Result of status format: 80 82 A4 00 01 21 91 11 10 90 10 90 92 92 00 92 92 00 00 91 00 10 10 00 90 00 00 10 00 00 90 90 00 0F 10 10 00 91 00 DD

Status itself: 80 82 A4 01 01 35 FF F6 05 DB 04 7C 04 76 02 FE 02 FB 04 7C 02 01 00 04 7C 02 01 00 00 00 00 00 00 00 00 B1 78 00 00 00 00 0B E1 00 18 FF FF 01 9E EF C8 00 00 00 00 02 00 05 00 C9

arjenhiemstra, 2021-11-29:
thanks, we might have found a bug in the itho firmware here... 
Length according to the status format: 21 = 33 decimal
Length according to the itho service tool, 30 decimal

jasperslits, 2021-11-29:
But I don't quite get on why it was working before? That itho firmware bug must have been there from the beginning, no?

arjenhiemstra, 2021-11-29:
Well, it is still working as before... I just included code to check for these kind of issues cause normally this would mean the mapping is missing some references. But in this case it shines light on itho firmware issues as well.

jasperslits, 2021-11-29:
Thanks. For now it's probably fine to leave it as-is. It doesn't look pretty all the relevant statuses are showing up. 

But maybe @janbredenbeek has more problems when he checks the results...

janbredenbeek, 2021-11-29:
Now able to retrieve the settings (similar to @jasperslits).  👍 
Also still get label out of bound error at end of status...

arjenhiemstra, 2021-11-30:
Implemented with version 2.3.0
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.3.0

closed 2021-11-30 by arjenhiemstra