arjenhiemstra/ithowifi #56: CC1101 failed to initialize

jasperslits, 2021-11-27:
**Describe the bug**
Non-CVE unit with latest firmware doesn't detect CC1101 properly. This is showing up when connected to either WPU or HRU-350. This used to work fine but I cannot recall since when it didn't work as I had format the add-on and switched it between HRU and WPU as I only had 1 device. 

**To Reproduce**
Steps to reproduce the behaviour:
1) Go to System settings
2) Enable Itho RF Remote Support
3) Reboot 
4) Device should be available immediately but it's then inaccessible
5) When the add-on web page is eventually is available CC1101 is disabled in the menu

**Expected behaviour**
RF Remotes shown in left menu, immediate availability. 

**Debug logging**
Logs:
2021-11-27 13:08:17 N: IP:192.168.178.37
2021-11-27 13:08:17 N: Setup: Virtual remote ID: 128,223,108
2021-11-27 13:08:17 N: Webserver: started
2021-11-27 13:08:18 N: Setup: init of CC1101 RF module successful
2021-11-27 13:08:18 N: mDNS: started
2021-11-27 13:08:18 N: Hostname: wpu
7377 N: Setup: done
2021-11-27 13:08:26 N: System boot, last reset reason: DEEPSLEEP_RESET
2021-11-27 13:08:26 N: HW rev: NON-CVE 1, FW ver.: 2.3-beta6
2021-11-27 13:08:31 N: WiFi: connection successful
2021-11-27 13:08:31 N: WiFi info:
2021-11-27 13:08:31 N: Mode:STA
2021-11-27 13:08:31 N: Status:3
2021-11-27 13:08:31 N: IP:192.168.178.37
2021-11-27 13:08:31 N: Setup: Virtual remote ID: 128,223,108
2021-11-27 13:08:31 N: Webserver: started
2021-11-27 13:08:31 N: mDNS: started
2021-11-27 13:08:31 N: Hostname: wpu
2021-11-27 13:08:31 N: Setup: done
2021-11-27 13:08:33 N: Setup: init of CC1101 RF module failed
2021-11-27 13:08:37 N: System boot, last reset reason: SDIO_RESET
2021-11-27 13:08:37 N: HW rev: NON-CVE 1, FW ver.: 2.3-beta6
2021-11-27 13:08:42 N: WiFi: connection successful
2021-11-27 13:08:42 N: WiFi info:
2021-11-27 13:08:42 N: Mode:STA
2021-11-27 13:08:42 N: Status:3
2021-11-27 13:08:42 N: IP:192.168.178.37
2021-11-27 13:08:42 N: Setup: Virtual remote ID: 128,223,108
2021-11-27 13:08:42 N: Webserver: started
2021-11-27 13:08:42 N: mDNS: started
2021-11-27 13:08:42 N: Hostname: wpu
2021-11-27 13:08:42 N: Setup: done

**Device information**
- Firmware version [e.g. 2.2.1] 2.3-b6
- Hardware revision (e.g. non-cve 1.1) non-cve 1
- Type/model number Itho (on sticker outside the box) HRU-350
- CC1101 RF module enabled [ yes, no ] yes
- In case of wifi issues, wifi access point make and model: N/A

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] MacOS Monterey 
 - Browser [e.g. chrome, safari] Safari
 - Version [e.g. 22] 15

arjenhiemstra, 2021-11-29:
I cannot reproduce this. It might indicate a hardware issue where either a solder joint of the cc1101 module isn't connected anymore or the module itself is broken.

jasperslits, 2021-11-29:
A visual check didn't yield anything. I noticed that it does say 'CC1101 task memory: 0 bytes free' but seems to show up also for devices without the CC1101? 

Maybe there's indeed something wrong as this was also giving issues when doing the level3 debugging for the WPU. 
Could it be damaged by connecting this unit to the WPU? Maybe the CC1101 doesn't really like that? 
The add-on itself works fine with the WPU...

arjenhiemstra, 2021-11-29:
>I noticed that it does say 'CC1101 task memory: 0 bytes free'

That's normal if the module is not active

>Could it be damaged by connecting this unit to the WPU? Maybe the CC1101 doesn't really like that?

No, the CC1101 module is connected only to the ESP32 and if somehow a power spike damaged the board then the ESP32 should be dead as well. I always test the board after soldering because sometimes the solder joint seems ok but does not make contact. It could be that in this case a join was barely making contact (which could also explain issues with debug indeed) and is now not making contact anymore at all.
You could test it with a conductivity / diode meter of a multimeter. If it doesn't sort it out I will repair / replace the module.

jasperslits, 2021-11-29:
And this one worked fine for several weeks so definitely something happened. But it maybe explains the lockups when doing the debugging for the WPU control commands...

Do you have a picture of the back of the non-CVE unit with CC1101 so I can do a comparison against my board? 
[https://github.com/jasperslits/ithowiki_content/blob/master/IMG_0500.jpeg](url)

It's a bit blurry, no idea why.

jasperslits, 2021-12-19:
I ordered a new one with CC1101, so we can resume our WPU temp control attempt. I don’t have a multimeter and the unit works fine so that is now connected to autotemp unit in the attic.

closed 2021-12-19 by jasperslits