arjenhiemstra/ithowifi #58: Feature request: support for Itho Spider remote

Pistoletjes, 2021-12-01:
Please add support for the remote/thermostat module [https://www.ithodaalderop.nl/nl-NL/professional/product/03-00476](https://www.ithodaalderop.nl/nl-NL/professional/product/03-00476). This remote contains an RF transceiver which seems to pair two-way with the fan. 

I needed I can help by supplying logs or dumps.

jasperslits, 2021-12-01:
I have a spider myself and this was discussed in #50 where you see some preliminary captures already. 

As my CC1101 transceiver is a bit broken, continuing the analysis (not merely dumping logs / dumps) for some key scenarios is needed:
1) use api.html?debug=level3 to capture traffic
2) Figure out what it is sent for each mode (low, medium,high,auto)
3) Do the same for WPU (if you have it)

Do you have the spider linked to a WPU and HRU? Or just CVE?

Pistoletjes, 2021-12-01:
Thanks for your response, I didn't check #50. I only have a CVE to play with but I'll check the other issue.

lvanjaarsveld, 2022-02-10:
I have a spider with wpu. Will have a look at it soon and update here.

HAuser2019, 2022-04-25:
And?

closed 2022-07-15 by stale[bot]