arjenhiemstra/ithowifi #59: Demandflow does not respond to non-CVE wifi module api calls

0x00-0xFF, 2021-12-01:
The Demandflow does not respond to any of the api calls given by the non-CVE wifi module

A request to any of the following gives OK:
/api.html?timer=1337
/api.html?command=timer1
/api.html?vremote=timer1
/api.html?vremote=timer2
/api.html?vremote=timer3

I would expect that on the status page 3_6_9-hours-demandflow-timer_sec would change and the Demandflow would turn on high. 

[NRG.Watch non cve wifi with cc1101](https://www.nrgwatch.nl/product/itho-non-cve-wifi-module/)
[Itho DemandFlow](https://www.ithodaalderop.nl/nl-NL/professional/product/536-0405)
[Itho WTW HRU 350](https://www.ithodaalderop.nl/nl-NL/professional/product/03-00389)
[Itho Remote 536-0146](https://www.ithodaalderop.nl/nl-NL/professional/product/536-0146)

arjenhiemstra, 2021-12-03:
What procedure did you follow to join the remote to the demandflow?

0x00-0xFF, 2021-12-03:
The procedure to join the physical remote is to press two buttons diagonally across from each other, eg bottom left and top right. 
The nrg.watch settings are: 
- Send join command every power on
- Map RF remotes to virtual remote

arjenhiemstra, 2021-12-03:
If I look at the manual RFT remotes can only be joined in IBS Fase 1.  That will also be the case for the virtual remote (it presents itself to the demandflow in the same way as the internal RF module of the demandflow)

0x00-0xFF, 2021-12-03:
So it seems... the installation guy told me that the manual should be joined within x 15 seconds after power-on. I will try to join the module using the IBS fase.

arjenhiemstra, 2021-12-03:
The IBS fase is 5 minutes according to my manual, I think this is also the time you have to join a remote.

0x00-0xFF, 2021-12-03:
Tried it three times... too bad. I can join/leave the normal remote and the demandflow blinks orange as it's supposed to. But joining the nrg.watch using api.html?vremote=high or the join button on the debug page, does nothing. No blinking leds and no response after IBS fase.

arjenhiemstra, 2021-12-03:
Ok, thanks, more investigation needed then

arjenhiemstra, 2022-03-04:
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha3
implements the needed different types of virtual remotes and API options

closed 2022-04-15 by arjenhiemstra