arjenhiemstra/ithowifi #60: Feature request - Support Itho RFT Remote 536-0146

0x00-0xFF, 2021-12-01:
Add support for the [Itho Remote 536-0146](https://www.ithodaalderop.nl/nl-NL/professional/product/536-0146) buttons and functions on the NRG.Watch Non CVE WiFi Module.

Top left / Normal
1-12-2021 10:27:01: W 62 --,--,-- --,--,-- 52,5C,BB 22F8 3:00,01,02 (cmd:unknown)

Top right / Eco Comfort
1-12-2021 10:27:01: W 62 --,--,-- --,--,-- 52,5C,BB 22F8 3:00,01,02 (cmd:unknown)

Lower left / Cook 
1-12-2021 10:27:42: W 65 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,1E,02,03 (cmd:unknown)

Lower right / Timer 3 6 9
1-12-2021 10:28:24: W 71 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,42,09,03,03 (cmd:unknown)
1-12-2021 10:28:24: W 70 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,42,06,03,03 (cmd:unknown)
1-12-2021 10:28:24: W 69 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,42,03,03,03 (cmd:unknown)

[NRG.Watch non cve wifi with cc1101](https://www.nrgwatch.nl/product/itho-non-cve-wifi-module/)
[Itho DemandFlow](https://www.ithodaalderop.nl/nl-NL/professional/product/536-0405)
[Itho WTW HRU 350](https://www.ithodaalderop.nl/nl-NL/professional/product/03-00389)
[Itho Remote 536-0146](https://www.ithodaalderop.nl/nl-NL/professional/product/536-0146)

JoostBaltissen, 2021-12-01:
Just to get some more information. Could you post the messages received with two presses on the cook button?
Two presses should be 60 minutes in instead of 30. I just wonder if the message will change from 1E to 3C.

0x00-0xFF, 2021-12-01:
It did: 
1-12-2021 20:45:13: W 102 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,3C,02,03 (cmd:unknown)
1-12-2021 20:45:13: W 102 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,3C,02,03 (cmd:unknown)
1-12-2021 20:45:13: W 102 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,3C,02,03 (cmd:unknown)
1-12-2021 20:45:13: W 101 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,1E,02,03 (cmd:unknown)
1-12-2021 20:45:13: W 101 --,--,-- --,--,-- 52,5C,BB 22F3 5:00,02,1E,02,03 (cmd:unknown)

arjenhiemstra, 2021-12-03:
Is the idea to receive those signals or send them as virtual remote commands?

0x00-0xFF, 2021-12-03:
The idea is to send them as virtual remote commands. Receiving them would be a nice to have...

#proud-owner-of-three-nrg-watch-modules-lol

arjenhiemstra, 2021-12-03:
> #proud-owner-of-three-nrg-watch-modules-lol

wow! 🥇 

> The idea is to send them as virtual remote commands. Receiving them would be a nice to have...

The reason I'm asking is that if I look at the manual only 1 of that type of remote can be joined to the demandflow (it becomes the base controller), other remotes referenced in my manual are (must be) of the 536-0124 type, which can be joined per room.

0x00-0xFF, 2021-12-03:
Wouldn't that be the answer also for issue  #59

arjenhiemstra, 2022-03-02:
It should be working as of this version:
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha2

closed 2022-04-15 by arjenhiemstra