arjenhiemstra/ithowifi #61: Temperature stuck on certain value

terual, 2021-12-01:
**Describe the bug**
After starting the add-on the temperature measurement works correctly, but after a while it remains stuck on a certain value. After reoot temperature works again. The same stuck value is visible in the web interface and in the MQTT messages.

**To Reproduce**
Start add-on and look at temperature readings.
Wait.

**Expected behaviour**
Correct temperature reading.

**Screenshots**
[image]

**Device information**
- Firmware version: 2.3-beta6
- Hardware revision: 2 
- Type/model number Itho: CVE-S ECO HE (03-00402)
- CC1101 RF module enabled: no
- In case of wifi issues, wifi access point make and model: not relevant

**Debug logging**
Please provide debug logging from the debug page if possible:
```
2021-11-24 17:18:59 N: System boot, last reset reason: SDIO_RESET
2021-11-24 17:18:59 N: HW rev: 2, FW ver.: 2.3-beta4
2021-11-24 17:19:00 N: wifi config saved
2021-11-24 17:19:00 N: Setup: Wifi config load failed
2021-11-24 17:19:01 N: wifi AP mode started
2021-11-24 17:19:01 N: Setup: AP mode active
2021-11-24 17:19:01 N: Setup: Virtual remote ID: 60,244,252
2021-11-24 17:19:01 N: Webserver: started
2021-11-24 17:19:01 N: mDNS: started
2021-11-24 17:19:01 N: Hostname: ***************
2021-11-24 17:19:01 N: Setup: done
      2159 N: System boot, last reset reason: SDIO_RESET
      2177 N: HW rev: 2, FW ver.: 2.3-beta4
      2215 N: Setup: Wifi config load failed
      3891 N: wifi AP mode started
      3911 N: Setup: AP mode active
      3929 N: Setup: Virtual remote ID: 60,244,252
      3957 N: Webserver: started
      3978 N: mDNS: started
      3999 N: Hostname: ***************
      4019 N: Setup: done
    137034 N: wifi config saved
      2159 N: System boot, last reset reason: SDIO_RESET
      2178 N: HW rev: 2, FW ver.: 2.3-beta4
      8642 N: WiFi: connection successful
      8661 N: WiFi info:
      8679 N: Mode:STA
      8697 N: Status:3
      8716 N: IP:***************
      8734 N: Setup: Virtual remote ID: 60,244,252
      8760 N: Webserver: started
      8781 N: mDNS: started
      8799 N: Hostname: ***************
      8819 N: Setup: done
    233456 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
  21600022 N: Mem free: 165252, Mem low: 133168, Mem block: 97392
  43200045 N: Mem free: 157128, Mem low: 133168, Mem block: 97392
  44273018 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
  64800077 N: Mem free: 165164, Mem low: 133156, Mem block: 91376
  86400114 N: Mem free: 164980, Mem low: 123456, Mem block: 97392
 108000156 N: Mem free: 164976, Mem low: 123456, Mem block: 97392
 128950070 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
 129600196 N: Mem free: 164864, Mem low: 123456, Mem block: 97392
 130791107 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
 151200237 N: Mem free: 164676, Mem low: 123456, Mem block: 97392
 157852763 N: Firmware update: nrgitho-hw2-v2.3-beta6.bin
 157870173 N: Reboot requested
      2159 N: System boot, last reset reason: SDIO_RESET
      2178 N: HW rev: 2, FW ver.: 2.3-beta6
      5648 N: WiFi: connection successful
      5667 N: WiFi info:
      5685 N: Mode:STA
      5703 N: Status:3
      5722 N: IP:***************
      5741 N: Setup: Virtual remote ID: 60,244,252
     11565 N: MQTT: connected, System config: 1
     11593 N: Webserver: started
     11614 N: mDNS: started
     11633 N: Hostname: ***************
     11654 N: Setup: done
     15313 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
  21600023 N: Mem free: 152928, Mem low: 122720, Mem block: 96352
  43200052 N: Mem free: 162192, Mem low: 122672, Mem block: 96352
  55786428 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
  64800090 N: Mem free: 162088, Mem low: 122672, Mem block: 96352
  86400125 N: Mem free: 160740, Mem low: 122672, Mem block: 96352
 108000170 N: Mem free: 160732, Mem low: 122672, Mem block: 96352
 129600204 N: Mem free: 161892, Mem low: 122672, Mem block: 96352
 144098236 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
 151200237 N: Mem free: 161684, Mem low: 122672, Mem block: 96352
 172800265 N: Mem free: 161684, Mem low: 122672, Mem block: 90336
```

**Desktop (please complete the following information):**
Not relevant. Issue on both web interface as MQTT

**Smartphone (please complete the following information):**
Not relevant. Issue on both web interface as MQTT

arjenhiemstra, 2021-12-02:
Hmm, interesting issue. 
2 things:
1. please upgrade to the latest version 2.3.1, let's see if readout stability improves
2. when no temp updates are posted anymore, what does the itho status page of the add-on show?

terual, 2021-12-02:
I have upgraded to 2.3.2 just a minute ago. I will post an update when the temperature becomes stuck again or when everything is fine.

Regarding you second question: the add-on itho status page displayed the same stuck temperature value as on the MQTT messages.

terual, 2021-12-03:
Unfortunately the upgrade did not fix the issue:
[image]

Also on the add-on itho status page the stuck temp value is visible:
[image]

arjenhiemstra, 2021-12-03:
Weird!

Could you post your config.json file? (you can download it on the page Debug -> file editor).
All personal info that you do not want to have shared can be removed.

Furthermore, if the temp is stuck, what is the result of for example "query status" on the debug page?

terual, 2021-12-03:
Here are the contents of config.json. Essentially a stock config with enabled MQTT.
```
{"sys_username":"admin","sys_password":"admin","syssec_web":0,"syssec_api":0,"syssec_edit":0,"api_normalize":0,"syssht30":0,"itho_rf_support":0,"rfInitOK":false,"itho_fallback":20,"itho_low":20,"itho_medium":120,"itho_high":220,"itho_timer1":10,"itho_timer2":20,"itho_timer3":30,"itho_updatefreq":5,"itho_sendjoin":0,"itho_forcemedium":0,"itho_vremoteapi":0,"nonQ_cmd_clearsQ":1,"mqtt_active":1,"mqtt_serverName":"************","mqtt_username":"************","mqtt_password":"*********","mqtt_port":1883,"mqtt_version":1,"mqtt_state_topic":"itho/state","mqtt_ithostatus_topic":"itho/ithostatus","mqtt_remotesinfo_topic":"itho/remotesinfo","mqtt_lastcmd_topic":"itho/lastcmd","mqtt_state_retain":"yes","mqtt_cmd_topic":"itho/cmd","mqtt_lwt_topic":"itho/lwt","mqtt_domoticz_active":0,"mqtt_ha_active":1,"mqtt_ha_topic":"homeassistant","mqtt_idx":1,"sensor_idx":1,"version_of_program":"005"}
```
The result of "Query status":
```
Result: 80 82 A4 01 01 17 02 02 16 02 15 00 00 07 00 24 00 00 76 BE 00 00 02 DB 34 14 99 06 5F 8E
```

arjenhiemstra, 2021-12-03:
Thanks!

At the moment the temp is still stuck?

Could you then also post the results of "query devicetype" en "query status format"?

terual, 2021-12-03:
Yes, the temp is still stuck, I haven't rebooted since this morning.

Devicetype:
```
Result: 80 82 90 E0 01 25 00 01 00 1B 2E 19 01 FE FF FF FF FF FF 1D 0B 07 E1 43 56 45 2D 52 46 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58
```

Status format:
```
Result: 80 82 A4 00 01 0C 80 10 10 10 00 10 20 10 10 00 92 92 29
```

terual, 2021-12-03:
And now the temp isn't stuck anymore! Since the "query devicetype" and "query status format" the temp is the correct reading! I'm not sure which one of the queries did the trick

arjenhiemstra, 2021-12-03:
according to this info the current temp is 16.31 C and Humidity is 52.73%, that is different from the values on the status page so something must be happening in the add on firmware

arjenhiemstra, 2021-12-03:
Next time it is stuck, could you go to the page 'itho settings' and look if there is still a version mentioned? (without clicking on the buttons of the debug page).
It could be that due to an i2c error the version gets lost (en then the status won't update anymore)

terual, 2021-12-03:
Thanks, will do

terual, 2021-12-04:
Unfortunately the temperature reading just got stuck again.

On the itho settings page I see the following:
```
Itho device type: CVE-Silent
Itho fw version: 25
```

Something I noticed: I saw that on the itho status page the temp variable is stuck, but Temperature says `not available`, whereas earlier that was the same value as in the temp field.

To get it to update again I had to press two times on Query status format and now the Temperature field is the same again as temp. The first time I clicked on Query status format nothing appeared.

arjenhiemstra, 2021-12-05:
>On the itho settings page I see the following:

I'd hoped the version would be 0 but unfortunately that is not the case. This means the issue is further down the line.

>Something I noticed: I saw that on the itho status page the temp variable is stuck, but Temperature says not available, whereas earlier that was the same value as in the temp field.

This is useful info! Could you post the complete output of the status page next time the temp sensor value is stuck?

>To get it to update again I had to press two times on Query status format and now the Temperature field is the same again as temp. 

When you query the status format internally a structure gets filled with the needed info to be able to accommodate the result of the query status command. This takes some time, at max 2x the update frequency setting. Next time, please, after you captured the status page, please click it once and wait 2x the update frequency setting and see if the values return.

terual, 2021-12-07:
Itho status page when temp is stuck:

Label | Value
-- | --
temp | 16.9
hum | 53.2
ppmw | 6395
Speed status | 2
Internal fault | 0
Frost cycle | 0
Filter dirty | 0
AirQuality (%) | 200
AirQbased on | 64
CO2level (ppm) | 568
Indoorhumidity (%) | 53
Outdoorhumidity (%) | not available
Exhausttemp (°C) | not available
SupplyTemp (°C) | not available
IndoorTemp (°C) | not available
OutdoorTemp (°C) | not available
SpeedCap | 63496
BypassPos (%) | not available
FanInfo | auto
ExhFanSpeed (%) | 2
InFanSpeed (%) | 0
RemainingTime (min) | 0
PostHeat (%) | not available
PreHeat (%) | not available
InFlow (l sec) | not available
ExhFlow (l sec) | not available
Ventilation setpoint (%) | 2
Fan setpoint (rpm) | 534
Fan speed (rpm) | 533
Error | 0
Selection | 7
Startup counter | 36
Total operation (hours) | 30488
Absence (min) | 0
Highest CO2 concentration (ppm) | 568
Highest RH concentration (%) | 53
RelativeHumidity | 53.23
Temperature | not available

Now, when pressing once on Query status format the temperature updates again.

terual, 2021-12-14:
Any further ideas how to debug this?

arjenhiemstra, 2021-12-14:
Not yet, sorry had some busy weeks with work. 

Interesting to see that the RelativeHumidity is still present but Temperature not. Both are being send in 1 message... Is that the case every time the issue comes up?

terual, 2021-12-14:
Yes, humidity has never been stuck.

terual, 2022-07-08:
Issue is still relevant

arjenhiemstra, 2022-07-08:
[nrgitho-hw2-v2.4.0-alpha9-tmpdebug.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/9072461/nrgitho-hw2-v2.4.0-alpha9-tmpdebug.bin.zip)

Sorry for not picking up this issue again sooner!

I have created a debug version, could you try it out? Is you upgrade from 2.3.5 you might have to configure some parts again. 

This version shows two debug values on the itho status page,  I'm interested if these values change before/after the temp value gets stuck. Also the error message of the temp value itself has been suppressed (not showing "not available") but shows the raw value. Hopefully both will point in the direction of a solution.

kvandt, 2022-08-29:
Just had the same issue with 2.4.0 alpha 8. Now I installed alpha9. Let's see if the extra debug values give a clue if it happens again.

terual, 2022-10-13:
@arjenhiemstra I missed your post about the special debug version ([nrgitho-hw2-v2.4.0-alpha9-tmpdebug.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/9072461/nrgitho-hw2-v2.4.0-alpha9-tmpdebug.bin.zip)). Should I try the new beta to see if the issue is resolved?

arjenhiemstra, 2022-10-13:
I'm afraid that's of no use anymore. I now have more information available and the beta1 version attached to this issue does also not help (but gave more insight in the issue). I'm now going to work on another solution and will link this issue once it is out.

arjenhiemstra, 2022-10-21:
Please try https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.3-beta2

terual, 2022-10-21:
Installed the beta firmware: first finding with the new version is that the I2C connection is not established yet: 
` Itho I2C connection status: init failed - please power cycle the itho unit -`

I'm going to power cycle now and report back.

terual, 2022-10-23:
I got a power outage this morning and thus a forced reset of the Itho box and wifi addon.... Up until the point before the power outage everything ssemed to be working, no stuck values. But I will continue monitor the coming days.

arjenhiemstra, 2022-10-23:
Thanks for the update!

terual, 2022-10-24:
This morning I saw that all the values on MQTT didn't get any update anymore.

Checking the Itho status page diplays this:
` Itho I2C connection status: init failed - please power cycle the itho unit -`

arjenhiemstra, 2022-10-24:
Anything in the log?

terual, 2022-10-24:
```
  25983138 I: Warning: I2C timeout, trying I2C bus reset...
  25985746 I: I2C bus could not clear: SCL clock line held low
  25987013 I: Info: I2C bus busy
  25997023 I: Warning: I2C timeout, trying I2C bus reset...
  25999694 I: I2C bus could not clear: SCL clock line held low
  26001146 I: Info: I2C bus busy
  26011163 I: Warning: I2C timeout, trying I2C bus reset...
  26013820 I: I2C bus could not clear: SCL clock line held low
  26015162 I: Info: I2C bus busy
  26025166 I: Warning: I2C timeout, trying I2C bus reset...
  26028097 I: I2C bus could not clear: SCL clock line held low
  26029441 I: Info: I2C bus busy
  26039432 I: Warning: I2C timeout, trying I2C bus reset...
  26042122 I: I2C bus could not clear: SCL clock line held low
  26043434 I: Info: I2C bus busy
  26053446 I: Warning: I2C timeout, trying I2C bus reset...
  26056178 I: I2C bus could not clear: SCL clock line held low
  26057579 I: Info: I2C bus busy
  26067579 I: Warning: I2C timeout, trying I2C bus reset...
  26070585 I: I2C bus could not clear: SCL clock line held low
  26072021 I: Info: I2C bus busy
  26082025 I: Warning: I2C timeout, trying I2C bus reset...
  26084715 I: I2C bus could not clear: SCL clock line held low
  26086091 I: Info: I2C bus busy
  26096081 I: Warning: I2C timeout, trying I2C bus reset...
  26098828 I: I2C bus could not clear: SCL clock line held low
  26100211 I: Info: I2C bus busy
  26110206 I: Warning: I2C timeout, trying I2C bus reset...
  26112985 I: I2C bus could not clear: SCL clock line held low
  26114519 I: Info: I2C bus busy
  26124532 I: Warning: I2C timeout, trying I2C bus reset...
  26127230 I: I2C bus could not clear: SCL clock line held low
  26128576 I: Info: I2C bus busy
  26138580 I: Warning: I2C timeout, trying I2C bus reset...
  26141291 I: I2C bus could not clear: SCL clock line held low
  26142677 I: Info: I2C bus busy
  26152675 I: Warning: I2C timeout, trying I2C bus reset...
  26155469 I: I2C bus could not clear: SCL clock line held low
  26157099 I: Info: I2C bus busy
  26166830 I: Warning: I2C timeout, trying I2C bus reset...
  26169522 I: I2C bus could not clear: SCL clock line held low
  26170855 I: Info: I2C bus busy
  26180860 I: Warning: I2C timeout, trying I2C bus reset...
  26183591 I: I2C bus could not clear: SCL clock line held low
  26184945 I: Info: I2C bus busy
  26194956 I: Warning: I2C timeout, trying I2C bus reset...
  26197718 I: I2C bus could not clear: SCL clock line held low
  26199152 I: Info: I2C bus busy
  26209095 I: Warning: I2C timeout, trying I2C bus reset...
  26211794 I: I2C bus could not clear: SCL clock line held low
  26213146 I: Info: I2C bus busy
  26223146 I: Warning: I2C timeout, trying I2C bus reset...
  26225881 I: I2C bus could not clear: SCL clock line held low
  26227256 I: Info: I2C bus busy
  26237255 I: Warning: I2C timeout, trying I2C bus reset...
  26240014 I: I2C bus could not clear: SCL clock line held low
  26241454 I: Info: I2C bus busy
  26251655 I: Warning: I2C timeout, trying I2C bus reset...
  26254479 I: I2C bus could not clear: SCL clock line held low
  26255782 I: Info: I2C bus busy
  26265795 I: Warning: I2C timeout, trying I2C bus reset...
  26268514 I: I2C bus could not clear: SCL clock line held low
  26269910 I: Info: I2C bus busy
  26279914 I: Warning: I2C timeout, trying I2C bus reset...
  26282682 I: I2C bus could not clear: SCL clock line held low
  26284091 I: Info: I2C bus busy
  26294090 I: Warning: I2C timeout, trying I2C bus reset...
  26297059 I: I2C bus could not clear: SCL clock line held low
  26298383 I: Info: I2C bus busy
[... 151 lines cut ...]
  27018578 I: I2C bus could not clear: SCL clock line held low
  27020249 I: Info: I2C bus busy
  27029959 I: Warning: I2C timeout, trying I2C bus reset...
  27032670 I: I2C bus could not clear: SCL clock line held low
  27033973 I: Info: I2C bus busy
  27043992 I: Warning: I2C timeout, trying I2C bus reset...
  27046737 I: I2C bus could not clear: SCL clock line held low
  27048135 I: Info: I2C bus busy
  27058119 I: Warning: I2C timeout, trying I2C bus reset...
  27060899 I: I2C bus could not clear: SCL clock line held low
  27062343 I: Info: I2C bus busy
  27072271 I: Warning: I2C timeout, trying I2C bus reset...
  27074947 I: I2C bus could not clear: SCL clock line held low
  27076290 I: Info: I2C bus busy
  27086295 I: Warning: I2C timeout, trying I2C bus reset...
  27089063 I: I2C bus could not clear: SCL clock line held low
  27090445 I: Info: I2C bus busy
  27100444 I: Warning: I2C timeout, trying I2C bus reset...
  27103211 I: I2C bus could not clear: SCL clock line held low
  27104587 I: Info: I2C bus busy
  27114814 I: Warning: I2C timeout, trying I2C bus reset...
  27117615 I: I2C bus could not clear: SCL clock line held low
  27118955 I: Info: I2C bus busy
```
I'm not sure how I can relate these messages to the moment that the IC2 connection failed. According to Home Assistant I got the last temperature updates around 02:24.

terual, 2022-10-24:
The behaviour I now got is not the behaviour I had. Now the whole I2C connection is lost, previously only the temperature stopped updating. Was that expected?

arjenhiemstra, 2022-10-24:
Could be 2 different things then. I assumed they were related but maybe not.

terual, 2022-10-24:
Did a power cycle of the Itho. Everything works now.

Maybe this can be of help?
```
2239 I: System boot, last reset reason: POWERON_RESET
2486 I: HW rev: 2, FW ver.: 2.4.3-beta2
8892 I: WiFi: connection successful
9120 I: WiFi info:
9351 I: SSID:**** | BSSID[****
9597 I: RSSI:-53dBm
9844 I: Mode:STA
10114 I: Status:WL_CONNECTED
10565 I: IP:***
10756 I: Setup: Virtual remotes, start ID: 3C,F4,FC - No.: 1
10970 I: Setup: remotes configfile loaded
14174 I: MQTT: connection failed, System config: 1
14413 I: Webserver: started
14678 I: mDNS: started
14919 I: Hostname: nrg-itho-f4fc
15164 I: Setup: done
15412 I: I2C init: QueryDevicetype - fw:25 hw:27
16050 I: I2C init: QueryStatusFormat - items:12
16502 I: I2C init: QueryStatus
16982 I: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
```

TD-er, 2022-10-24:
@arjenhiemstra Not sure how you implemented this, but have you also included to temporary swap the I2C pins, restart `Wire`, perform an I2C scan, revert the pin swap and restart `Wire` ?

arjenhiemstra, 2022-10-24:
@TD-er I followed your approach here (but the blocking version for now):
https://github.com/letscontrolit/ESPEasy/blob/14fb3a2e07627ded9b5a592d45e7a1e479a8229b/src/src/Helpers/I2C_access.cpp#L14-L17

I'm not using the `Wire` lib because it lacks a proper slave implementation. I use the ESP-SDK implementation instead. What I see from the I2C check code is that the pins are setup as `INPUT`. On my add-on there are pull-up resistors on the board itself. So I conclude that if after switching the pins to an `INPUT` state the SCL and/or SDA line is still `LOW` there must be an other device on the bus actively driving the line `LOW`. I'm not sure if the add-on can force that to `HIGH`.

A bit of background on the issue;
From my end I had implemented (and I made a stupid mistake there) a check to determine whether the I2C bus is busy. I did this by checking if the SCL line was in a `LOW` state. That should have been the SDA line because a start condition first pulls the SDA line `LOW`, thereafter the SCL line. By only checking the SCL line there was quite reasonable chance a master already initiated a start sequence after which the add-on still tried to send a start sequence itself.
I have changed that in the latest beta2 version of the firmware checking both the SDA and SCL line states. (But I think that there is still a possibility for a collision in the time between this check and the actual initiation of the master communication though, which I I'm not sure how to mitigate partly but not fully)
Nevertheless, apart from the latest crash mentioned here it seems that other users haven't experienced issues anymore since these changes.

arjenhiemstra, 2022-10-24:
Reading a bit further, checking SDA/SCL pin state alone is not enough it seems. Both pins `HIGH` could also occur mid transmission. Better probably to check pin state for at least 2 SCL cycles?

Furthermore, I2C multi master also describes the need for implementation of collision detection and arbitration between each byte / I2C interaction... This should then be done within ESP-IDF.

There are references in code to that ie.
https://github.com/espressif/esp-idf/blob/daf3da5e4bcd001fb339c2d5424077d819fc8051/components/driver/i2c.c#L1502-L1506

But I cannot find any info on the implementation of `i2c_ll_is_bus_busy`

TD-er, 2022-10-24:
The clock can also be pulled down by any device.
This is called clock stretching and is part of the I2C standard.
Any device can use this to effectively lower the transmission speed, or stall the communication to get some requested data ready.

But just wondering about your remark of using as an I2C slave....
You know you cannot act as a master when you're a slave device?
Only a master can start communication.

arjenhiemstra, 2022-10-24:
I know about clock stretching, but clock stretching for hours is a bit outside the scope of the standard I guess ;-)

>You know you cannot act as a master when you're a slave device?
Yes I know. What happens is that a command needs to be send from the add-on (master) to the itho firmware (slave) and then I switch the add-on to slave mode and wait for the itho firmware to answer the request (which then acts as a I2C master).

arjenhiemstra, 2022-10-25:
Interesting to see from the logging that collisions are quite common:

```
2022-10-24 23:01:00 I: Info: I2C bus busy, cleared after 3ms
2022-10-24 23:04:36 I: Info: I2C bus busy, cleared after 1ms
2022-10-24 23:19:24 I: Info: I2C bus busy, cleared after 2ms
2022-10-24 23:32:51 I: Info: I2C bus busy, cleared after 3ms
2022-10-24 23:47:59 I: Info: I2C bus busy, cleared after 1ms
2022-10-25 00:21:30 I: Info: I2C bus busy, cleared after 0ms
2022-10-25 00:26:26 I: Info: I2C bus busy, cleared after 4ms
2022-10-25 00:35:38 I: Info: I2C bus busy, cleared after 2ms
2022-10-25 00:50:26 I: Info: I2C bus busy, cleared after 1ms
2022-10-25 01:31:29 I: Info: I2C bus busy, cleared after 1ms
2022-10-25 01:46:17 I: Info: I2C bus busy, cleared after 4ms
2022-10-25 01:51:13 I: Info: I2C bus busy, cleared after 4ms
2022-10-25 02:44:08 I: Info: I2C bus busy, cleared after 3ms
2022-10-25 02:49:03 I: Info: I2C bus busy, cleared after 7ms
2022-10-25 02:58:15 I: Info: I2C bus busy, cleared after 3ms
2022-10-25 03:17:59 I: Info: I2C bus busy, cleared after 0ms
2022-10-25 03:21:55 I: Info: I2C bus busy, cleared after 0ms
2022-10-25 03:22:55 I: Info: I2C bus busy, cleared after 5ms
2022-10-25 03:56:06 I: Info: I2C bus busy, cleared after 0ms
2022-10-25 04:18:46 I: Info: I2C bus busy, cleared after 4ms
2022-10-25 05:01:49 I: Info: I2C bus busy, cleared after 1ms
2022-10-25 05:15:16 I: Info: I2C bus busy, cleared after 2ms
2022-10-25 05:33:40 I: Info: I2C bus busy, cleared after 7ms
2022-10-25 05:37:16 I: Info: I2C bus busy, cleared after 0ms
```

arjenhiemstra, 2022-10-27:
I just released version 2.4.3 which addresses this further. In some situations the add-on could still access the I2C bus while another master was active. The latest code changes in 2.4.3 should prevent this or at least further lower the change of this happening.

TD-er, 2022-10-27:
> or at least further lower the change of this happening.

Also known as "making it harder to reproduce" ;)

terual, 2022-10-27:
Right now the temperature still hasn't been stuck since the last reboot. Would it create useful information to keep the add-on running on the beta to check if it stays that way? Or should I just upgrade to version 2.4.3?

arjenhiemstra, 2022-10-27:
> > or at least further lower the change of this happening.
> 
> Also known as "making it harder to reproduce" ;)

haha, something like that.... in this "multi-master and the other master is out of our control" situation there is just so much you can do I'm afraid.

arjenhiemstra, 2022-10-27:
> Right now the temperature still hasn't been stuck since the last reboot. Would it create useful information to keep the add-on running on the beta to check if it stays that way? Or should I just upgrade to version 2.4.3?

I would upgrade

terual, 2022-10-27:
Done! If the coming week everything keeps working I will close this issue, thanks!

terual, 2022-11-01:
Forgot to check over the weekend, but friday the temperature got stuck again, but now also the humidity and CO2 levers are stuck as well.

The I2C connection status says connected by the way and the logfile doesn't show anything interesting.

arjenhiemstra, 2022-11-02:
Ok, thanks for the update. Will work on it further. In the mean time, you can always downgrade to another version if you should wish so.

terual, 2022-11-02:
No problem. Thanks for your work on this issue! Is there anything further I can do to track down the issue?

KoeKk, 2022-11-07:
If needed I can also test for this issue, I have it also (sometimes stuck temp, sometimes stuck humidity), also with the same hints (stuck temp, and Temperature not available)

I have a CVE-Silent with fw version 27

arjenhiemstra, 2023-01-06:
Could you try the latest 2.4.4-beta4 firmware and report how it works for you? If you have an itho with built-in humidity sensor please enable the i2c safeguard feature (if your add-on hw revision supports it)

KoeKk, 2023-01-06:
Sure, I've updated to the 2.4.4-beta4 and enabled the i2c safeguard. I'll let you know when something happens or in a few weeks if the situation improves.

Thanks!

terual, 2023-01-06:
I also updated to 2.4.4-beta4, enabled the i2c safeguard and did a reboot. Everything looks fine for now and I will report in a couple of days.

terual, 2023-01-12:
Everything is still working right now, so it is looking good thus far.

KoeKk, 2023-02-17:
After running 2.4.4-beta4 for 6 weeks it looks perfect. The issues has not returned. Thanks for your work!

arjenhiemstra, 2023-02-17:
2.5.0 has been released as new stable release, including fixes for the described issue.

https://github.com/arjenhiemstra/ithowifi/releases/tag/2.5.0

arjenhiemstra, 2023-02-21:
> 2.5.0 has been released as new stable release, including fixes for the described issue.
> 
> https://github.com/arjenhiemstra/ithowifi/releases/tag/2.5.0

I've found a serious issue with the fail save firmware update, please do not install 2.5.0 and if you have already, wait with further updates until 2.5.1 comes out.

terual, 2023-04-22:
Issue is fixed, thanks!

closed 2023-04-22 by terual