arjenhiemstra/ithowifi #62: Win10 WebSocket connection not established

arjenhiemstra, 2021-12-03:
**Describe the bug**
WebSocket connection is not established resulting in empty fields on the web page.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 10 (only?)
 - Browser [e.g. chrome, safari] Edge and Chrome
 - Version [e.g. 22] not known

arjenhiemstra, 2021-12-03:
Possible fix by moving to another WebSocket server library, please test:

[nrgitho-hw2-v2.3.3-wstest.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/7650817/nrgitho-hw2-v2.3.3-wstest.bin.zip)
[nrgitho-noncve-v2.3.3-wstest.bin.zip](https://github.com/arjenhiemstra/ithowifi/files/7650819/nrgitho-noncve-v2.3.3-wstest.bin.zip)

wimb0, 2021-12-03:
I'm having the same issue, and the version with another WebSocker lib solves the issue for me.

arjenhiemstra, 2021-12-05:
I hope this has been solved as of version 2.3.3
https://github.com/arjenhiemstra/ithowifi/commit/36bbefce3b88b730f65fe3d8e30f41fbf96d6ea2

If not, please feel free to reopen the issue

closed 2021-12-05 by arjenhiemstra