arjenhiemstra/ithowifi #65: Temp and Hum value rounding issue since v2.3.2

wimb0, 2021-12-03:
**Describe the bug**
Temperature and Humidity data via MQTT is not rounded anymore since switch from 2.3b6 to 2.3.2

**To Reproduce**
View payload via MQTTExplorer or check Home Assistant sensors.

**Expected behaviour**
Have the Temp and Hum data rounded at 1 or 2 decimals.

**Screenshots**
[image]

**Device information**
HW v2 with firmware 2.3.2

arjenhiemstra, 2021-12-03:
That's strange, the code didn't really change at this point between those versions and here it still reads two decimals...

How are you reading the temperature? Directly from the sensor or out of the itho firmware?

sanderkob, 2021-12-04:
Running 2.3.2, here the mqtt itho/ithostatus message shows 2 decimals in temperature and humidity. 
So something in Home Assistant?
Ppmw value (absolute humidity) now is correctly rounded, thanks.

wimb0, 2021-12-04:
I'm using the original built-in sensor.
Reading the values from MQTT or the auto created `sensor.nrg_itho_b3dc_temperature` sensor.

Also tried formatting the flash again and starting with a new config.

Raw MQTT topic `itho/ithostatus`:

`{"temp":20.40436,"hum":47.03441,"ppmw":7084,"Speed status":30,"Internal fault":0,"Frost cycle":0,"Filter dirty":0,"AirQuality (%)":"not available","AirQbased on":0,"CO2level (ppm)":"not available","Indoorhumidity (%)":46,"Outdoorhumidity (%)":"not available","Exhausttemp (°C)":"not available","SupplyTemp (°C)":"not available","IndoorTemp (°C)":"not available","OutdoorTemp (°C)":"not available","SpeedCap":63488,"BypassPos (%)":"not available","FanInfo":"auto","ExhFanSpeed (%)":30,"InFanSpeed (%)":0,"RemainingTime (min)":0,"PostHeat (%)":"not available","PreHeat (%)":"not available","InFlow (l sec)":"not available","ExhFlow (l sec)":"not available"}`

wimb0, 2021-12-04:
Don't know why and how, but after flashing `nrgitho-hw2-v2.3.3-wstest.bin`, format flash, reconnect to wifi it seems to be OK again. Values are rounded again.
[image]

arjenhiemstra, 2021-12-05:
Rounding forced to 1 decimal from version 2.3.3
https://github.com/arjenhiemstra/ithowifi/commit/36bbefce3b88b730f65fe3d8e30f41fbf96d6ea2

closed 2021-12-05 by arjenhiemstra