arjenhiemstra/ithowifi #66: Fan niet meer te bedienen via web/mqtt/api

teunlaan, 2021-12-09:
Subject "alles behalve debug commando's werken niet meer"
Issue description (short) : box staat op max en gaat niet meer terug
Current Firmware version: 2.3.4
Current Hardware revision (backside addon) : v2
Type/model number Itho (on sticker outside the box): cve-s eco se
IP [static or DHCP]: dhcp (fixed ip via dhcp server) 
MQTT in use [yes/no]: yes
HTTP API in use [yes/no]: yes 
CC1101 RF module enabled [yes/no]: yes
Restart through webinterface resolves issue (temporary) [yes/no]: no
Power cycle Itho resolves issue (temporary) [yes/no]: yes

Na x tijd (is variable),  reageert niks meer. 
Wanneer de fan op 100% gezet is (handmatig of via HA) kan de fan niet meer naar een lager nivo gezet worden. 
Via slider, api of mqtt kan de fan niet meer geregeld worden. Via debug wel. Wordt hij via debug op low gezet dan geageerd web, api, mqtt wel, maar de fan gaat altijd naar 100% en is daarna niet meer terug te zetten. 

De fysieke knoppen werken op dit moment ook anders. Stand 1 is nog steeds low, maar 2 en 3 zijn beide high geworden. 

Alleen een losse RF unit werkt nog 'normaal', maar dit geeft niet de controle terug voor de andere apparaten. 

Itho system, geeft netjes alle waardes aan. Speed staat dus op 100. Selector zie ik gewoon verspingen met de RF en fysieke knop

Dacht nog even aan de vochtsensor dat die de boel actief houd, maar ook na een uur als him onder de 50% is blijft hij op vol vermogen draaien (weet niet waar de grens ligt voor de sensor) 

Na een power cycle werk alles weer, tijdelijk, normaal. Kan dan hem via mqtt/web/api + fysiekeknop in willekeurig volgorde aansturen. 

2021-12-09 13:23:55 N: Firmware update: nrgitho-hw2-v2.3.4.bin
2021-12-09 13:24:13 N: Reboot requested
2021-12-09 13:10:50 N: System boot, last reset reason: SDIO_RESET
2021-12-09 13:10:50 N: HW rev: 2, FW ver.: 2.3.4
2021-12-09 13:10:54 N: WiFi: connection successful
2021-12-09 13:10:55 N: WiFi info:
2021-12-09 13:10:55 N: Mode:STA
2021-12-09 13:10:55 N: Status:3
2021-12-09 13:10:55 N: IP:192.168.1.120
2021-12-09 13:10:55 N: Setup: Virtual remote ID: 60,175,196
2021-12-09 13:10:55 N: Setup: init of CC1101 RF module successful
2021-12-09 13:10:56 N: MQTT: connected, System config: 1
2021-12-09 13:10:56 N: Webserver: started
2021-12-09 13:10:56 N: mDNS: started
2021-12-09 13:10:56 N: Hostname: itho-fan
2021-12-09 13:10:57 N: Setup: done
2021-12-09 13:24:31 N: I2C init: QueryDevicetype - fw:25 hw:27
2021-12-09 13:24:31 N: I2C init: QueryStatusFormat - items:12
2021-12-09 13:24:31 N: I2C init: QueryStatus
2021-12-09 13:24:31 N: I2C init: Virtual remote join command send
15916 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...

teunlaan, 2021-12-10:
update:  vanochtend uit het niets staat fan weer op 100%. En niet meer terug te zetten

2021-12-09 17:38:43 N: I2C init: Virtual remote join command send
2021-12-09 17:38:43 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
2021-12-09 23:38:27 N: Mem free: 117120, Mem low: 68752, Mem block: 59320
2021-12-10 05:38:28 N: Mem free: 119792, Mem low: 68752, Mem block: 59320

itho status

<html><body>

Label | Value
-- | --
temp | 21.5
hum | 46.6
ppmw | 7514
Speed status | 100
Internal fault | 0
Frost cycle | 0
Filter dirty | 0
AirQuality (%) | not available
AirQbased on | 0
CO2level (ppm) | not available
Indoorhumidity (%) | 46
Outdoorhumidity (%) | not available
Exhausttemp (°C) | not available
SupplyTemp (°C) | not available
IndoorTemp (°C) | not available
OutdoorTemp (°C) | not available
SpeedCap | 63488
BypassPos (%) | not available
FanInfo | auto
ExhFanSpeed (%) | 100
InFanSpeed (%) | 0
RemainingTime (min) | 0
PostHeat (%) | not available
PreHeat (%) | not available
InFlow (l sec) | not available
ExhFlow (l sec) | not available
Ventilation setpoint (%) | 100
Fan setpoint (rpm) | 2700
Fan speed (rpm) | 2700
Error | 0
Selection | 7
Startup counter | 128
Total operation (hours) | 25040
Absence (min) | 0
Highest CO2 concentration (ppm) | not available
Highest RH concentration (%) | 46
RelativeHumidity | 46.56
Temperature | 21.5

</body>
</html>

NA PowerCycle. 
En alles werkt weer zoals verwacht

2021-12-09 23:38:27 N: Mem free: 117120, Mem low: 68752, Mem block: 59320
2021-12-10 05:38:28 N: Mem free: 119792, Mem low: 68752, Mem block: 59320
2225 N: System boot, last reset reason: POWERON_RESET
2375 N: HW rev: 2, FW ver.: 2.3.4
6352 N: WiFi: connection successful
2021-12-10 06:33:10 N: WiFi info:
2021-12-10 06:33:10 N: Mode:STA
2021-12-10 06:33:11 N: Status:3
2021-12-10 06:33:11 N: IP:192.168.1.120
2021-12-10 06:33:11 N: Setup: Virtual remote ID: 60,175,196
2021-12-10 06:33:11 N: Setup: init of CC1101 RF module successful
2021-12-10 06:33:11 N: MQTT: connected, System config: 1
2021-12-10 06:33:11 N: Webserver: started
2021-12-10 06:33:12 N: mDNS: started
2021-12-10 06:33:12 N: Hostname: itho-fan
2021-12-10 06:33:12 N: Setup: done
2021-12-10 06:33:19 N: I2C init: QueryDevicetype - fw:25 hw:27
2021-12-10 06:33:19 N: I2C init: QueryStatusFormat - items:12
2021-12-10 06:33:20 N: I2C init: QueryStatus
2021-12-10 06:33:20 N: HA DISCOVERY: Start publishing MQTT Home Assistant Discovery...
2021-12-10 06:33:20 N: I2C init: Virtual remote join command send

<html><body>

Label | Value
-- | --
temp | 21.8
hum | 45.8
ppmw | 7517
Speed status | 20
Internal fault | 0
Frost cycle | 0
Filter dirty | 0
AirQuality (%) | not available
AirQbased on | 0
CO2level (ppm) | not available
Indoorhumidity (%) | 45
Outdoorhumidity (%) | not available
Exhausttemp (°C) | not available
SupplyTemp (°C) | not available
IndoorTemp (°C) | not available
OutdoorTemp (°C) | not available
SpeedCap | 63488
BypassPos (%) | not available
FanInfo | auto
ExhFanSpeed (%) | 20
InFanSpeed (%) | 0
RemainingTime (min) | 0
PostHeat (%) | not available
PreHeat (%) | not available
InFlow (l sec) | not available
ExhFlow (l sec) | not available
Ventilation setpoint (%) | 20
Fan setpoint (rpm) | 1058
Fan speed (rpm) | 1058
Error | 0
Selection | 7
Startup counter | 129
Total operation (hours) | 25040
Absence (min) | 0
Highest CO2 concentration (ppm) | not available
Highest RH concentration (%) | 45
RelativeHumidity | 45.77
Temperature | 21.78

<br>
</body>
</html>

arjenhiemstra, 2021-12-10:
Op het eerste gezicht geen echt rare dingen in de logs of status info. Zou je een update willen doen naar 2.3.5? Er zat nog een fout in de API wat (een deel) van dit issue zou kunnen verklaren.

teunlaan, 2021-12-10:
Ga ik nu doen. 
Mocht het weer mis gaan, dan heb ik nog een paar dingen die ik wil/kan gaan testen.
* vochtsensor verwijderen en hem aansturen met een externe sensor (is bestel maar nog niet binnen). Vanochten net voordat hij ophol sloeg  gaat de waarde van de vochtsensor van 46% in 1x naar 54%  als ik in de history kijk (HA slaat pas aan bij 55%)
* perilex stekker vervangen voor een normalen (heb ik nog liggen)  om er zeker van de zijn dat dat die geen gek signiaal afgeeft

arjenhiemstra, 2021-12-10:
Dat van die vochtsensor is interessant, die neemt inderdaad de besturing van de box over en dan reageert de itho niet op input. Dat zou het heel goed kunnen verklaren!
Om dat te voorkomen kun je twee dingen doen:
1. Additional sensor support aan zetten
2. De volgende settings op de "itho settings" pagina aanpassen: "Rise in absolute humidity (X) at which it goes to high (mg_kg)". Ik heb deze nu op 3000 staan (standaard is die iets van 400 oid)

teunlaan, 2021-12-10:
1) zal ik aanzetten. Had de text anders "gelezen" , dacht dat je de hele sensor werd uitzet.
2) standaard staat deze zo bij mij op 700. Zal dezen ook eens verhogen naar 3000.

Vind het wel vreemd. Zonder de add-on heeft de fan nooit dit gedrag vertoont. 
700mg_kg is ook wel een heel laag getal. Dat is bij 21 graden 45% luchtvochtigheid?
Als je dan bedankt dat hij aangaat omdat er warme waterdamp van het douchen voorbij komt, wat waarschijnlijk meer dan 21 graden is, dan ga je de fan idd niet meer uit krijgen. Zeker niet als waarde " Hysteresis X before falling back to normal operation (mg_kg)"  op 400 staat. Dat is een luichtvochtigheid van minder dan 28% bij 20 graden

arjenhiemstra, 2021-12-10:
> 1. zal ik aanzetten. Had de text anders "gelezen" , dacht dat je de hele sensor werd uitzet.

De sensor wordt in de itho firmware uit gezet, de add-on leest de sensor dan nog wel uit.
Ik zou als ik jou was voor optie 2 gaan, effectief heeft dit hetzelfde effect alleen de sensor handling laat je dan aan de itho firmware (ook het eventueel verwarmen van de sensor indien nodig bijvoorbeeld)

teunlaan, 2021-12-10:
We gaan optie 2 proberen. 
Als hij weer gaat " fippen"  meld ik mij weer. 
Gaat alles goed (en is heyt dus deze setting) , dan zal ik volgende week vrijdag dat ook even laten weten

bwired-nl, 2021-12-10:
ik test deze setting nu ook

sanderkob, 2021-12-10:
> Vind het wel vreemd. Zonder de add-on heeft de fan nooit dit gedrag vertoont. 700mg_kg is ook wel een heel laag getal. Dat is bij 21 graden 45% luchtvochtigheid? Als je dan bedankt dat hij aangaat [...]

Even ter verduidelijking (alhoewel off-topic voor dit issue): de waarden genoemd in settings 77 en 78 (CVE versie) zijn de _verhoging_ resp. _verlaging_ van de absolute vochtigheid (in mg/kg ofwel ppmw).   
Bijvoorbeeld, bij 21°C en 45% RH is de absolute vochtigheid 7000 ppmw. De Itho unit zal met de standaardinstelling (700 resp. 400) reageren wanneer de vochtigheid 700 ppmw hoger wordt, dus bij 7700 ppmw; dit komt overeen met 21°C en 49% RH. De unit zal weer terugschakelen bij 7300 ppmw (21°C en 47% RH).  
Zelf gebruik ik mijn eigen regelstrategie: ventilatie verhogen wanneer de vochtigheid +1000 ppmw is t.o.v. de laagste waarde in de afgelopen 8 minuten. En ventilatie terugschakelen wanneer de vochtigheid -700 ppmw is t.o.v. de hoogste waarde in de afgelopen 8 minuten.

rubenkuipers, 2021-12-13:
Ik test deze setting nu ook. Heb momenteel de setup met ingebouwde CO2 sensor die gekoppeld zit op de Itho wifi module. Ook na commands om de unit op high te laten draaien, gebeurt dit voor enkele seconden, waarna de unit vanzelf weer terugschakelt naar automatisch niveau.

arjenhiemstra, 2021-12-14:
De CO2 sensor heeft prio lijkt het, er zijn nog wat andere settings waarmee dat gedrag misschien te beinvloeden is maar het kan ook zijn dat je aangewezen bent op virtual remote commando's (die volgens mij de CO2 sensor kunnen overrulen).
Misschien is het veranderen van "CO2 period time (min)" naar 0 of 30 een optie?

teunlaan, 2021-12-14:
Voor mij is het probleem opgelost.
De setting aanpassen naar 3000 heeft geholpen.

Misschien handing om ergens hier iets ove rneer te zetten (zo te zien hebben meer mensen hier last van)

rubenkuipers, 2021-12-15:
@arjenhiemstra Lijkt er tot nu toe op dat die instelling (CO2 period time (min)) weinig effect heeft. Is het nodig om map RF remote to virtual remote aan te hebben staan voor het versturen van commands naar /vremote ?

arjenhiemstra, 2021-12-16:
Nee is niet nodig, map rf is vooral handig als je een device hebt welke geen traploze sturing via I2C accepteert (dus met name voor non-CVE devices en HRU200)

closed 2022-03-02 by arjenhiemstra