arjenhiemstra/ithowifi #68: Firmware 2.3.5  kills MQTT

teunlaan, 2021-12-10:
HW v2
NA het updaten van de firmware 2.3.4 > 2.3.5 werkt HA middels MQTT niet meer.
MQTT  server krijgt verbinding (staat ook als connected in de add-on)
1639148620: New client connected from 192.168.1.120 as itho-fan (p2, c1, k15) 

maar er komt niks binnen, temp/hum en fan setting worden niet meer geupdate. ook niet meer te besturen vanuit HA. Device info in HA blijft ook FW2.3.4 aangeven

alleen  terug flashen naar 2.3.4 lost het probleem op

arjenhiemstra, 2021-12-10:
Weet je zeker dat je het juiste firmware bestand geflashed hebt?
Hier precies dezelfde setup en HA en MQTT doen het goed met 2.3.5. Er zijn tussen 2.3.4 en 2.3.5 twee variabelen veranderd van speed naar timer die geen relatie hebben met HA en/of MQTT status updates.

teunlaan, 2021-12-10:
Heb de download knop vanuit web gui gebruikt, de hw2 versie

arjenhiemstra, 2021-12-10:
Zou goed moeten zijn dan, dat heb ik zelf ook gedaan.
Geen idee wat het zou kunnen verklaren. Opnieuw proberen en wellicht reboot, power cycle of opnieuw instellen proberen?

teunlaan, 2021-12-11:
gedaan: power cycle fan, config mqtt gedisabled/enables in fan. HA gereboot, complet MQTT config verwijderd, en weer toegevoegd. 
Op 2.3.5 komt er gewoon niks binnen. FAn geeft aan HA dautodisovery, HA geeft aan dat het ip adres is connected, maar (gat een reinstall van MQTT wordt er niks gediscoverd.
Zodfra ik terug ga naar 2.3.4 werkt alles weer (meerdere malen heen en weer geflashed) 

Ik blijf nog wel even op 2.3.4 :)

teunlaan, 2021-12-11:
Ik sluit hem maar. snap er geen snars van.
Ivm werkzaamheden moest de stroom af van de verdieping waar mijn HA unit staat.

Alles weer opgekomen, geen connectie met de fan?
IP/web interface ook down . power cycle gedaan. 
mqtt  is connected (op web fan + HA) maar geen info (zoals ik gisteren had met de update naar  2.3.5) 
Kirijg het met geen mogelijkheid meer aan de praat.
Toch maar een update naar 2.3.5 gedaan, en het werkt weer???

closed 2021-12-11 by teunlaan

arjenhiemstra, 2021-12-11:
Soms is het echt om je haren uit je hoofd te trekken.... fijn dat het werkt iig!