arjenhiemstra/ithowifi #69: Non-CVE - Device not responding to commands

jasperslits, 2022-01-04:
**Describe the bug**
HRU-350 with 2.3.5 non-CVE firmware. Using MQTT API or Web API doesn't lead to HRU performing actions. As I have multiple devices, I changed MQTT config to use ithohru as prefix instead of itho. I don't see it's relevant to this problem but worth mentioning it. 

lastcmd topic is correctly updated:
2022-01-04T11:06:08+0100 ithohru/lastcmd {"source":"HTML API","command":"medium","timestamp":1641290523}

ithohru/ithostatus returns all data just fine, I don't see errors in the log and devices responds to manual controls given via the physical remotes.  

Query 31DA/31D9 return failed, everything works but not sure if it's related? 

**To Reproduce**
Steps to reproduce the behaviour:
1. Execute /api.html?command=medium OR
This gives an 'OK' but nothing happens. 

2. Send 'medium' as command to ithohru/cmd topic
This doesn't return anything obviously but has the same outcome: nothing happens. 

**Expected behaviour**
HRU-350 goes to medium

**Screenshots**
N/A

**Device information**
- Firmware version [e.g. 2.2.1]
- Hardware revision (e.g. non-cve 1.1)
- Type/model number Itho (on sticker outside the box)
- CC1101 RF module enabled [ yes, no ]
- In case of wifi issues, wifi access point make and model:

**Debug logging**
Please provide debug logging from the debug page if possible:

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

ThijsMaas, 2022-01-07:
I have the same issue with my HRU 350

arjenhiemstra, 2022-01-07:
Out of the box, the add-on sends i2c_pwm command if you use the api.html?command option. I'm not sure but it looks like this does not work for your device.
Could you try the virtual remote commands? You can also map these virtual remote commands to the api.html?command by enabling this features on the system settings page. The virtual remote needs to be joined just as a regular physical remote.

jplab1983, 2022-01-07:
This sounds like the exact issue with my DemandFlow setup.  IaAlso tried pairing the virtual remote after power up of Itho module (using "join" command in "debug" menu option), but unit does still not respond to any of the virtual commands.

arjenhiemstra, 2022-01-07:
> This sounds like the exact issue with my DemandFlow setup. I also tried pairing the virtual remote after power up of Itho module (using "join" command in "debug" menu option), but unit does still not respond to any of the virtual commands.

With the difference that I can confirm that the virtual remote commands work on the HRU eco. It could be different on the HRU350 but I doubt it.

Joparmen, 2022-01-25:
I have the same problem with the 350, manually I can’t push another speed to the Itho. I neither am capable of discovering my remote.

the non-cve can see the values requested values I ask trough the remote (and it changes) to the itho but I am not capable of controlling the unit.

closed 2022-03-13 by jasperslits