arjenhiemstra/ithowifi #70: Web interface supposed to work?

jplab1983, 2022-01-07:
What an awesome add-on!  Hats off, I was about to start trying to hack it, then I stumbled upon this add-on. 

It arrived today and I have set it up as per the instructions. I am however stuck on the web interface, not being able to control the unit with the controls provided. As soon as I click either of the buttons (or move the slider), it moves back to the default view in a few seconds.

[image]

I have a DemandFlow, so not sure whether these controls are supposed to be working since it is classified as non-CVE?

I am able to see the _Itho Status_, so I doubt there is any communication issues.

Would be great if you can suggest how to move past this issue, as I can't wait to integrate into Home Assistant.

jasperslits, 2022-01-07:
It is not supposed to work and probably should be removed. 

You can check the wiki for more info but you would typically use this with Home assistant via mqtt. The web api should work, so /api.html?command=medium

jplab1983, 2022-01-07:
ah, thanks for the info.

checked the wiki, and did the setup as per the instruction.  sad to say, I hae the Itho status values in HA (and web interface) but unable to change the speed at all (not even low, medium and high).  Reverted to checking web api, does not work either.  Even thought "OK" message is displayed, it makes no change to the speed.

any suggestions?  I am not sure what else I can try

arjenhiemstra, 2022-01-07:
The slider is not functional for the DemandFlow because the pwm_i2c commands do not work on the DemandFlow. 

Probably each itho device will need its own interface on this page. As the firmware moved from the CVE version which does support the pwm_i2c commands this is still there.

Tomorrow I hope to make a test setup with the DemandFlow and HRU eco that I have laying around here at home. Goals are to include the proper remote commands for the base DemandFlow remote and to be able to join an extra virtual remote per room. If I manage to get it working control through the DemandFlow should be possible.

jplab1983, 2022-01-07:
Thanks Arjen, looking forward to it.  In the meantime I am unable to set even low/med/high for overall speed. Almost as if the Demandflow unit is only sending out the sensor info, and not accepting any commands.

jplab1983, 2022-01-07:
Another thing where I am confused, below is a representation of my setup: Demandflow with CVE-S ECO HE
[image]

I purchased the non-CVE add-on due to the fact that I have the DemandFlow, but at the same time it is coupled with "CVE-S ECO HE". Does that mean that my setup is indeed CVE and not non-CVE?

closed 2022-07-15 by stale[bot]