arjenhiemstra/ithowifi #72: HRU 350 is not respondig to / receiving commands

ThijsMaas, 2022-01-21:
I have a non-cve module connected to my HRU 350 unit.
I can connect to the web interface and view all itho settings, but I'm unable to change the fan speed.
What I have tried:
```
http://<ip>/api.html?vremote=timer3
http://<ip>/api.html?vremote=high
http://<ip>/api.html?command=high
```

The module seems to be connected as I get this model information:
```
Configuration of the Itho unit
Itho device type: HRU 350
Itho fw version: 3
```
and this is the debug log:
```
2022-01-21 10:06:56 N: HW rev: NON-CVE 1, FW ver.: 2.3.5
2022-01-21 10:07:00 N: WiFi: connection successful
2022-01-21 10:07:00 N: WiFi info:
2022-01-21 10:07:00 N: Mode:STA
2022-01-21 10:07:01 N: Status:3
2022-01-21 10:07:01 N: IP:192.168.1.4
2022-01-21 10:07:01 N: Setup: Virtual remote ID: 96,79,68
2022-01-21 10:07:01 N: Webserver: started
2022-01-21 10:07:01 N: Setup: init of CC1101 RF module successful
2022-01-21 10:07:01 N: mDNS: started
2022-01-21 10:07:03 N: Hostname: nrg-itho-4f44
2022-01-21 10:07:03 N: Setup: done
2022-01-21 10:07:11 N: I2C init: QueryDevicetype - fw:0 hw:0
2022-01-21 10:07:15 N: I2C init: QueryDevicetype - fw:3 hw:43
2022-01-21 10:07:15 N: I2C init: QueryStatusFormat - items:33
2022-01-21 10:07:16 N: I2C init: QueryStatus
```

What can I do here?

arjenhiemstra, 2022-01-21:
Have you joined the virtual remote to the HRU?

You can also do this manually by going the the debug page and click on the join button (within 2 minutes after power-up of the HRU)

ThijsMaas, 2022-01-21:
Ah, I totally missed that the virtual remote needs to join, similar to how you would join normal remotes. It works now, thanks!

closed 2022-01-21 by ThijsMaas