arjenhiemstra/ithowifi #73: 3d printed case?

jplab1983, 2022-01-31:
Not a bug, but wanted to find out if someone has already designed a 3D case for the hardware unit?  I was about to start and create one for my purchase, but if there one available already....

arjenhiemstra, 2022-01-31:
Not yet, so if you want to take the 'token', it's yours!
Do you need any measurements or are you fine with the actual product and the design files?

jplab1983, 2022-02-01:
Let me get going with the design files and let you know in case I need anything else.

BlueCase, 2022-03-07:
Hi,
because nothing seems to happen here I created a Case for my own.
Testprint in progress.

[image]

arjenhiemstra, 2022-03-07:
There is also one on thingiverse but not added here I see:
[https://www.thingiverse.com/thing:5236721](https://www.thingiverse.com/thing:5236721)
[https://www.thingiverse.com/thing:5236658](https://www.thingiverse.com/thing:5236658)

jplab1983, 2022-03-07:
Nice! And yeah, my bad.  Did not get around to designing one yet. Keen to see how it turned out for you?

BlueCase, 2022-03-07:
@arjenhiemstra strange... Searched last Sunday on thingiverse for this without result.

Nevertheless I finished my Prototyp today. Still some tweaks to do but overall the result ist good.
[image]

BlueCase, 2022-03-09:
Hi,
so my final version is now available on thingiverse.
https://www.thingiverse.com/thing:5277696

closed 2022-07-15 by stale[bot]