arjenhiemstra/ithowifi #75: reduce network traffic/load

jtonk, 2022-02-16:
**Describe the bug**
currently the module produces 580mb of network traffic per 24h. That's quite a lot, for a 'simple' sensor.

**To Reproduce**
default settings, with Mqtt (and home assistant), but I don't think that's relevant.

**Expected behaviour**
I don't expect a module to generate this much traffic. My p1 port esp module only produces 38mb, my ems-esp module only 16mb of network traffic in 24h. 

**Screenshots**
n/a

**Device information**
- Firmware version 2.3.5
- Hardware revision 2
- Type/model number Itho, don't know, it's an ecofan
- CC1101 RF module enabled: yes
- In case of wifi issues, wifi access point make and model: unifi in-wall HD

**Debug logging**
Please provide debug logging from the debug page if possible: n/a

**Desktop (please complete the following information):**
not relevant

**Smartphone (please complete the following information):**
not relevant

jasperslits, 2022-02-16:
I have changed the update frequency from 5 to 60 seconds which I think is a better default. This also reduces mqtt volume. Higher granularity of sensor date doesn’t add a lot of value.  

I guess not all fields are relevant to monitor but this was discussed earlier and seems to add much complexity. 

On the other hand, it’s all local traffic and typically everything that is not consumed of the MQTT is discarded anyway. I have 5 add-ons and can’t be bothered with the volume unless it’s starts to impact network performance which is very unlikely

arjenhiemstra, 2022-02-20:
You can reduce network traffic by lowering the update frequency. The way the firmware is currently created all information that is available in an itho device is send out on an update. This can of course be changed to a selectable set of parameters, I'll put it on the backlog but other items have higher priority first.

arjenhiemstra, 2022-03-04:
Since https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha3 the standard is now an update every 10 seconds instead of 5 seconds. This can be lowered further in the user interface.

closed 2022-03-04 by arjenhiemstra