arjenhiemstra/ithowifi #76: Partly working remote RTF 536-0020

DoubelJ, 2022-02-17:
I have a remote that is working for 75%

It is the RTF zender 536-0020 with the leave function. The buttons for low,medium and high just work as expected only the leave function doesn't work (jet).

arjenhiemstra, 2022-02-20:
Could you post a picture of the remote? 

It would be nice if you could capture this leave command via the debug function (see manual for more info on this function).
It might be wise to use the 2.4.0-alpha1 version for that because the 2.3.5 version contains a bug that can crash the websocket.

roblomq, 2022-02-20:
Maybe integrate the ESPEasy rules described in the Domoticz section of the document below?
[Google documents Itho ventilatie](https://docs.google.com/document/d/1dH6AReayBHbQreJHS6-qGyChBb5rmrGqW4OEB5V01c8/edit?usp=drivesdk)

TD-er, 2022-02-20:
@roblomq Document is not accessible

roblomq, 2022-02-20:
It's not my document, you can ask for access.

arjenhiemstra, 2022-02-20:
> Maybe integrate the ESPEasy rules described in the Domoticz section of the document below?

I'm not sure what you mean. Looking at the document, something like this is already possible with the current firmware but I fail to see how it is related to the issue?

DoubelJ, 2022-02-20:
Is not related. I already tried to capture the command but i did get
something like error. Will try again with the other version.

roblomq, 2022-02-20:
Sorry, I posted this in the wrong issue... I will add it again to the right issue..

DoubelJ, 2022-02-22:
This is the cmd showed in the RF log:
[image]

As you can see low, medium and high are working, but the 4th button is unknown.

Hereby the picture of the remote:
[image]

DoubelJ, 2022-02-23:
It looks like the 536-0024 is the newer version of the RF I have, but has the same functions:
[image]

arjenhiemstra, 2022-03-04:
With the latest test version I expect this button to work:
https://github.com/arjenhiemstra/ithowifi/releases/tag/Version-2.4.0-alpha3

arjenhiemstra, 2022-03-07:
"in the status of the device itself"

What do you mean by this, could you share a screenshot?

dennisvbussel, 2022-03-09:
Hi I have a similar issue using the Itho RFT Co2 remote (536-0150) using 2.4.0-alpha4 it is the exact same as in de picture in https://github.com/arjenhiemstra/ithowifi/tree/master/remotes.

It seems that for this particular remote only the join and leave commands are recognised correctly. From what I observe is that the first field is different for my remote (many other remotes uses 3:00,.... my remote uses 3:63,...) I'm not sure what this field means however I think this is safe to ignore.

```
10-3-2022 00:02:01: _W 165 --,--,-- --,--,-- 74,EE,11 22F1 3:63,02,04 (cmd:unknown) -> Mode 1 button (top left)
10-3-2022 00:02:06: _W 166 --,--,-- --,--,-- 74,EE,11 22F1 3:63,04,04 (cmd:unknown) -> Mode 3 button (top right)
10-3-2022 00:02:10: _W 231 --,--,-- --,--,-- 74,EE,11 22F1 3:63,03,04 (cmd:unknown) -> Mode Automatic pushed once (bottom left)
10-3-2022 00:02:15: _W 168 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,0A (cmd:unknown) -> 10-min timer button pushed once (bottom right)
10-3-2022 00:02:17: _W 169 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,14 (cmd:unknown) -> 20-min timer button pushed twice (bottom right)
10-3-2022 00:02:18: _W 170 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,1E (cmd:unknown) -> 30-min timer button pushed three times (bottom right)
```

Complete debug log:
```
10-3-2022 00:02:22: _W 171 --,--,-- --,--,-- 74,EE,11 62F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:02:22: _W 171 --,--,-- --,--,-- 74,EE,15 22F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:02:22: _W 171 --,--,-- --,--,-- 74,EE,11 22F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:02:21: _W --- 53,88,1C 53,EB,DD --,--,-- 4401 20:10,00,00,49,23,E9,00,00,EC,76,CA,76,00,00,00,00,00,00,00,63 (cmd:unknown)
10-3-2022 00:02:21: RQ --- 53,EB,DD 51,88,1C --,--,-- 4401 20:10,00,00,49,23,E9,00,00,00,00,00,00,00,00,00,00,00,00,00,63 (cmd:unknown)
10-3-2022 00:02:18: _W 170 --,--,-- --,--,-- 74,EE,13 22F3 3:61,00,1E (cmd:unknown)
10-3-2022 00:02:18: _W 170 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,1E (cmd:unknown)
10-3-2022 00:02:18: _W 170 --,--,-- --,--,-- 74,EE,11 22F2 2:63,00 (cmd:unknown)
10-3-2022 00:02:17: _W 169 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,14 (cmd:unknown)
10-3-2022 00:02:17: _W 169 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,14 (cmd:unknown)
10-3-2022 00:02:16: _W 169 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,1C (cmd:unknown)
10-3-2022 00:02:15: _W 168 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,0A (cmd:unknown)
10-3-2022 00:02:15: _W 168 --,--,-- --,--,-- 74,EE,11 22F3 3:63,10,0A (cmd:unknown)
10-3-2022 00:02:15: _W 168 --,--,-- --,--,-- 74,EE,11 22F3 3:63,00,0A (cmd:unknown)
10-3-2022 00:02:13: _W --- 54,96,EA --,--,-- 44,96,EA 3110 4:00,00,00,10 (cmd:unknown)
10-3-2022 00:02:12: _W --- 53,84,6B 53,86,2B --,--,-- 4401 20:30,00,01,4A,7E,07,00,01,6B,57,EC,B1,00,00,00,00,00,00,00,63 (cmd:unknown)
10-3-2022 00:02:10: _W 167 --,--,-- --,--,-- 74,E6,11 22F1 3:61,0B,04 (cmd:unknown)
10-3-2022 00:02:10: _W 167 --,--,-- --,--,-- 74,EE,11 22F1 3:61,03,04 (cmd:unknown)
10-3-2022 00:02:10: _W 231 --,--,-- --,--,-- 74,EE,11 22F1 3:63,03,04 (cmd:unknown)
10-3-2022 00:02:06: _W 166 --,--,-- --,--,-- 74,EE,11 22F1 3:63,04,04 (cmd:unknown)
10-3-2022 00:02:06: _W 166 --,--,-- --,--,-- 74,EE,11 22F1 3:63,04,04 (cmd:unknown)
10-3-2022 00:02:06: _W 166 --,--,-- --,--,-- 74,EE,11 22F1 3:63,04,04 (cmd:unknown)
10-3-2022 00:02:02: _W --- --,--,-- --,--,-- 53,88,1C 10E2 3:00,13,75 (cmd:unknown)
10-3-2022 00:02:01: _W 165 --,--,-- --,--,-- 74,EE,11 22F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:02:01: _W 165 --,--,-- --,--,-- 74,EE,11 22F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:01:55: _W 164 --,--,-- --,--,-- 74,EE,11 1FC9 12:62,22,F8,74,EE,11,01,10,E0,74,EE,11 (cmd:unknown)
10-3-2022 00:01:55: _W 164 --,--,-- --,--,-- 74,EE,11 1FC9 12:63,22,F8,74,EE,11,01,10,E0,74,EE,11 (cmd:join)
10-3-2022 00:01:55: _W 160 --,--,-- --,--,-- 74,EE,11 1FC9 12:63,22,F8,74,EE,11,01,10,E0,74,6E,11 (cmd:join)
10-3-2022 00:01:55: _W 163 --,--,-- --,--,-- 74,EE,11 22F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:01:55: _W 163 --,--,-- --,--,-- 74,EE,11 20F1 3:63,02,04 (cmd:unknown)
10-3-2022 00:01:53: _W --- 53,84,6B 53,84,39 --,--,-- 4401 20:10,00,01,41,45,85,00,01,6B,57,DE,3A,00,00,00,00,00,00,00,63 (cmd:unknown)
10-3-2022 00:01:49: _W --- 54,98,1C --,--,-- 54,98,1C 3EF0 6:02,00,00,0C,02,40 (cmd:unknown)
10-3-2022 00:01:48: _W 162 --,--,-- --,--,-- 74,EE,11 1FC9 6:63,1F,C9,74,EE,11 (cmd:leave)
10-3-2022 00:01:48: _W 162 --,--,-- --,--,-- 74,EE,11 1FC9 6:63,1F,C9,74,EE,11 (cmd:leave)
10-3-2022 00:01:47: _W 162 --,--,-- --,--,-- 74,EE,11 1FC9 6:63,1F,C9,74,EE,11 (cmd:leave)
10-3-2022 00:01:47: _W 161 --,--,-- --,--,-- 74,EE,11 22F1 3:63,03,04 (cmd:unknown)
```

arjenhiemstra, 2022-03-09:
Thanks for the log!
The 536-0150 is a different kind of remote with different commands compared to the ones on the image above.

The strange thing is, the commands you need for this remote are in code (I have the same remote and that one is working), really strange!

https://github.com/arjenhiemstra/ithowifi/blob/d681d92deee01434ca609867d140c3a5eae14c2f/software/NRG_itho_wifi/IthoPacket.h#L90-L98

dennisvbussel, 2022-03-09:
No problem :)

Yes I forgot to ask if it is better to open a new issue.

Can it be that there is no device ID in the log. The orgin and direction of the communication is not visible in the log. "--,--,-- --,--,--".

In the Virtual Remotes setup UI the remote is detected as id: "74,EE,11"

arjenhiemstra, 2022-03-09:
It's probably not the command that is the issue here, there is a bug somewhere else I'm afraid. Now find out where.

>Can it be that there is no device ID in the log. The orgin and direction of the communication is not visible in the log. "--,--,-- --,--,--".

I recently learned that the first address is the from, 2nd is to and 3rd is log/broadcast kind of a thing

arjenhiemstra, 2022-03-12:
@dennisvbussel  could you send me the config files?
You can find them on the debug page -> file editor -> right click file and click download
All the JSON files except the wifi config would be very nice. If you prefer not to post it here you can send them by e-mail as well.

dennisvbussel, 2022-03-12:
Yes no problem:
[logfile1.log](https://github.com/arjenhiemstra/ithowifi/files/8237838/logfile1.log)
[logfile0.current.log](https://github.com/arjenhiemstra/ithowifi/files/8237840/logfile0.current.log)
[config.json.log](https://github.com/arjenhiemstra/ithowifi/files/8237843/config.json.log)
[remotes.json.log](https://github.com/arjenhiemstra/ithowifi/files/8237844/remotes.json.log)
[vremotes.json.log](https://github.com/arjenhiemstra/ithowifi/files/8237845/vremotes.json.log)

Changed json to log for uploading to GitHub :)

arjenhiemstra, 2022-03-12:
> Yes no problem: [logfile1.log](https://github.com/arjenhiemstra/ithowifi/files/8237838/logfile1.log) [...]

Thanks!

Looking at your config, you haven't joined the remote with the add-on (there is no remote configured in the remotes.json file).
Apparently, if you haven't joined the remote, the command will alway show 'cmd:unknown'. Which is obviously not good and a bug.

This line of code is the culprit:
https://github.com/arjenhiemstra/ithowifi/blob/21b9c83c9a023324b1aa42641493bc4ebaa55b8e/software/NRG_itho_wifi/10_Gen_func.ino#L459

I wil change it in the next version, thanks again!

dennisvbussel, 2022-03-13:
Happy to help :)

So if I understand correctly this is a bug in the debug printing, the remote should functionally work?

arjenhiemstra, 2022-03-13:
Yes, you can test that yourself easily.
Join the remote on the 'RF remotes' page. go back to the debug function and activate debug level3. Then you should see the commands.

A little word of care; looking at the config file you supplied you have configured this remote as virtual remote, maybe also joined this remote to your itho. If that is the case, that means that once you join the remote to the add-on and haven't removed the remote binding to the itho itself the resulting speed changes might conflict each other because the itho is reacting to the remote and the add-on at (almost) the same time.

dennisvbussel, 2022-03-14:
Ah I see, I misunderstood the concept of a virtual remote. I assumed that it is a virtual (non physical) remote which is emulated in software and sends commands via RF. After reading https://github.com/arjenhiemstra/ithowifi/wiki/Remotes I see that it is more of a virtual RF receiver (RF -> I2C). 

What I want is for the ithowifi to act as a physical remote, or make a copy of my existing physical remote in software. 

So in my case the remote is paired to the itho HRU unit. The remote seems to be communicating uni directional (Remote -> HRU). How does ithowifi react when I physically push the button on the remote, does it update it's status?

arjenhiemstra, 2022-03-14:
Partly correct. I'll try to make a flow diagram to explain it. There are al lot of possible combinations that desperate need a better explanation.

arjenhiemstra, 2022-03-14:
First attempt:

[image]

closed 2022-04-15 by arjenhiemstra