arjenhiemstra/ithowifi #78: Doesn't receive signal from remote

roblomq, 2022-02-19:
I think I have a remote with number 536-0106 (is this something I can find somewhere?), see picture below. But the module doesn't seem to receive signals from it.
[image]

What I did was turning debug level 3 on, then I get a message window in the debug menu. There is some data received but when I press on the buttons on the remote there is no info added in the window.
Is it possible that this remote doesn't work?
[image]

Below a link to an overview of the units and remotes I found.
Maybe it's helpful to create a list of supported devices and remotes.
https://myshop.s3-external-3.amazonaws.com/shop4072000.images.ITHO-CVE-Eco-fan-historisch_overzicht_rf_bedieningen.pdf

arjenhiemstra, 2022-02-20:
So it seems signals are received ;-)

These remote commands are totally different from the other remotes but probably quite easy to add.

It would be nice if you could capture all the commands via the debug function and label them (ie. low, medium etc). 
It might be wise to use the 2.4.0-alpha1 version for that because the 2.3.5 version contains a bug that can crash the websocket.

roblomq, 2022-02-20:
It does receive signals but that signal shown in the screenshot is not from the remote. When I press the remote there is nothing shown in the log.
But have I done the procedure correctly to enable the logging for the remote? Because the log window sometimes disappears. The I have to send the log level 3 again.
Of course I can try to capture the signals but for now I can't get it working. Maybe out has to do with the bug you are talking about?
It's there a help file on how to update?

Some feedback from my side, maybe it's helpful for you to improve your fantastic product!
- Maybe it's more user friendly to add a log option in the menu instead of manually configure a http command?
- make an overview of the models that are compatible and which are not. For the units and also for the remotes and sensors. (I have a CVE ECO 2E but the addon won't fit this unit. So I had to buy a refurbished motorplate with print for €70 to be able to use your addon.
- Also on the web UI there are a lot of functions that are not explained in the help file. Probably there is some information in the tweakers topic but this topic is almost impossible to follow because there are so many users, questions and answers.

arjenhiemstra, 2022-02-20:
>But have I done the procedure correctly to enable the logging for the remote? Because the log window sometimes disappears. The I have to send the log level 3 again.

That has most probably something to do with the bug. 

>It's there a help file on how to update?

If you go to the web interface and click on the menu item 'Update', download this file: https://github.com/arjenhiemstra/ithowifi/blob/master/compiled_firmware_files/hardware_rev_2/nrgitho-hw2-v2.4.0-alpha1.bin

Select the file and click the Update button. That's all.

>Some feedback from my side, maybe it's helpful for you to improve your fantastic product!

Thanks!!

>Maybe it's more user friendly to add a log option in the menu instead of manually configure a http command?

Good point, will do that!

>make an overview of the models that are compatible and which are not.

I believe that the most comprehensive list of supported models is listed here:
https://github.com/arjenhiemstra/ithowifi/wiki
apart from the list on the webshop (which I try to keep in sync)

If any info is missing; the wiki is public and can be edited by anybody with an GitHub account, so feel welcome to change what you like. 

>Also on the web UI there are a lot of functions that are not explained in the help file. 

The manual covers the basics to get you up and running, the wiki provides the remainder of the available information. Not all information presented in the interface is really known. 
Part of the information and settings are presented in the interface but its goal or function is probably only known by itho engineers. The more we find out together the richer the documentation will be :)

roblomq, 2022-02-20:
Thank you. I will see if I can add some info to the wiki.

Just to be sure, as my current unit is still up and running and I use the new separate motor and print for testing. The remotes are still connected to the current unit. Is that a problem? Or do I need to exclude the remotes first to be able to receive signals on my test setup?

arjenhiemstra, 2022-02-20:
You can test with the remotes still connected. Your itho will react but that doesn't invalidate the received signals.

roblomq, 2022-02-20:
What's not clear for me where the wiki is exactly related to. Is it related to your addon you can buy or is it related to the software and the possibility to use this on the Itho fans? because the wiki says in the table that the CVE ECO 2 is supported, but I have a CVE ECO 2E (the "E" is for a 2 pins connector with earth) but this one doesn't have the 8-pins connector so not possible to add the module.

arjenhiemstra, 2022-02-20:
>Is it related to your addon you can buy or is it related to the software and the possibility to use this on the Itho fans?

Both

>because the wiki says in the table that the CVE ECO 2 is supported, but I have a CVE ECO 2E (the "E" is for a 2 pins connector with earth) but this one doesn't have the 8-pins connector so not possible to add the module.

So this means there are apparently two versions of the CVE ECO 2, one with the older style connector and one the the newer 2x4 pins I2C interface! That is new info thanks!

roblomq, 2022-02-20:
[image]

roblomq, 2022-02-20:
I've updated the firmware but it doesn't seem to change something. I get the RF log window but when I press on a button (or all 4 buttons together) on the remote I don't see a new log entry in the log window. I have an Evohome system and I think that's also on the 868mHz frequency so what I see in the log is signals form that. This is what I get after a while (the log was clear when I pressed some buttons, then it didn't show any entries, then after a while it shows this.

`20-2-2022 19:28:51: _W --- 58,25,86 --,--,-- 58,25,86 0008 2:00,C2 (cmd:unknown)
20-2-2022 19:28:48: _W --- 58,25,86 --,--,-- 58,25,86 0008 2:00,C2 (cmd:unknown)
20-2-2022 19:28:40: _W --- 12,4C,EB --,--,-- 07,6C,7C 12B0 3:01,00,00 (cmd:unknown)
20-2-2022 19:28:38: _W --- 12,4C,EB --,--,-- 07,6C,7C 12B0 3:01,00,00 (cmd:unknown)
20-2-2022 19:28:30: _W --- 07,6C,7C --,--,-- 07,6C,7C 1F09 3:FF,05,41 (cmd:unknown)
20-2-2022 19:28:14: _W --- 07,6C,7C --,--,-- 07,6C,7C 3B00 2:FC,C8 (cmd:unknown)
20-2-2022 19:28:00: RP --- 28,91,17 07,6C,7C --,--,-- 3EF0 9:00,00,10,0A,00,FF,03,1F,64 (cmd:unknown)
20-2-2022 19:27:00: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:26:45: _W --- 12,4C,E9 --,--,-- 07,6C,7C 2309 3:03,07,D0 (cmd:unknown)
20-2-2022 19:26:18: _W --- 28,91,17 --,--,-- 28,91,17 1FD4 3:00,34,E6 (cmd:unknown)
20-2-2022 19:26:16: _W --- 07,6C,7C --,--,-- 07,6C,7C 1F09 3:FF,05,41 (cmd:unknown)
20-2-2022 19:25:59: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:25:00: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:24:52: _W --- 12,4C,E9 --,--,-- 12,4C,E9 30C9 3:00,07,D2 (cmd:unknown)
20-2-2022 19:24:40: _W --- 12,4C,EB --,--,-- 12,4C,EB 30C9 3:00,08,02 (cmd:unknown)
20-2-2022 19:24:03: _W --- 58,25,86 --,--,-- 58,25,86 30C9 3:00,07,89 (cmd:unknown)
20-2-2022 19:24:02: _W --- 07,6C,7C --,--,-- 07,6C,7C 1F09 3:FF,05,41 (cmd:unknown)
20-2-2022 19:24:00: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:23:59: _W --- 58,25,86 --,--,-- 58,25,86 30C9 3:00,07,89 (cmd:unknown)
20-2-2022 19:23:36: _W --- 28,91,17 --,--,-- 28,91,17 1FD4 3:00,34,E5 (cmd:unknown)
20-2-2022 19:22:59: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:21:59: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:21:47: _W --- 07,6C,7C --,--,-- 07,6C,7C 1F09 3:FF,05,41 (cmd:unknown)
20-2-2022 19:20:59: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:20:53: _W --- 28,91,17 --,--,-- 28,91,17 1FD4 3:00,34,E4 (cmd:unknown)
20-2-2022 19:20:21: _W --- 5B,67,97 --,--,-- 5B,67,97 0008 2:00,A6 (cmd:unknown)
20-2-2022 19:19:59: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:19:57: RQ --- 07,6C,7C 28,91,17 --,--,-- 3EF0 1:00 (cmd:unknown)
20-2-2022 19:19:33: _W --- 07,6C,7C --,--,-- 07,6C,7C 1F09 3:FF,05,41 (cmd:unknown)`

arjenhiemstra, 2022-02-20:
Thanks, evohome uses the same RF protocol (HoneyWell RAMSES-II ) as itho. But it seems that this older type of remote is using a RF protocol that does not adhere to the honeywell  standard, maybe it uses an older itho own developed protocol.
To be able to decipher this you need to go a bit deeper and probably listen to the radio communication itself. 

No quick and easy solution it seems unfortunately...

roblomq, 2022-02-20:
I will see if i can buy some new remotes. When they are working, I can send one of the old ones to you if you like, then you can see if you can add it to your firmware.

roblomq, 2022-02-22:
I now have a 536-0150 remote. This one is recognized flawless by the add-on. But when I press the remote, the fan does not react. It drives me a bit crazy...

roblomq, 2022-02-22:
For some reason after several reboots it does work...

arjenhiemstra, 2022-02-23:
Sorry for the late answer, I’m not in the NL currently. Dad to read that it took so much effort, that shouldn’t be the case!

closed 2022-03-12 by arjenhiemstra

ShadowJonathan, 2022-07-13:
Sorry for the bump, but it does seem that that fan (545-5020) uses an older standard than the RFT one, and online everywhere they're kinda treated as the same.

See [this comment](https://community.home-assistant.io/t/itho-daalderop-fan-rf-remote-with-c1101-and-esp8266/38710/114?u=j0j0) i made on a likewise thread about this issue. TLDR; there are two standards, and the oldest one hasnt been reverse engineered anywhere.