arjenhiemstra/ithowifi #80: Domoticz MQTT settings dissapears

roblomq, 2022-02-20:
When you go to MQTT and turn o the Domoticz MQTT option you get a device IDX and a sensor IDX. But when you press save, the IDX fields disappears. So when you want to change it you first have to turn them off and then on again.
Small issue but if they are shown constant it saves a lot of searching for newbies like me.

arjenhiemstra, 2022-02-20:
Hmm, stupid bug! I'll fix it in the next version.
Thanks for spotting this!

roblomq, 2022-02-20:
A bit related to this, it is possible to add a slider to Domoticz with this, but is it possible to add a tile to Domoticz with a selector switch (as shown below)? Because in this case I find buttons easier to use than a slider.
Of course I can use MQTT or API commands and fill them in manually in domoticz but then it's only one directional, so when someone uses the RF remote then the status in Domoticz isn't updated.

roblomq, 2022-02-20:
Maybe integrate the ESPEasy rules described in the Domoticz section of the document below?
[Google documents Itho ventilatie](https://docs.google.com/document/d/1dH6AReayBHbQreJHS6-qGyChBb5rmrGqW4OEB5V01c8/edit?usp=drivesdk)

arjenhiemstra, 2022-02-20:
I'm not the greatest Domoticz expert but looking at the description from the document it should be possible to built the same behaviour with the current firmware. All information is available, the question is how to get in represented into Domoticz.... Maybe post the question on the Tweakers forum? Might be that someone just knows how to configure this.

roblomq, 2022-02-21:
When I know how to add the rules to the ESP I think I can do it. Do you know that?

closed 2022-07-15 by stale[bot]

roblomq, 2022-11-20:
Saw that this issue is till there. When you set the "Domoticz MQTT" to On, the settings "Device IDX" and "Sensor IDX" disappear.

roblomq, 2023-04-10:
just for information I updated to version 2.5.5 but the issue is still there.