arjenhiemstra/ithowifi #81: Unit only responds on button low of a 536-0150 in debug mode 3

roblomq, 2022-02-26:
**Describe the bug**
The unit only responds on the low button when the add-on is in debug 3 mode.

**To Reproduce**
Steps to reproduce the behaviour:
1. Turn on debug level 3 (by using the "..../api.html?debug=level3")
2. Press the high button on the remote.
3. Unit responds and goes to high rpm
4. Press the low button on the remote
5. Unit responds and goes to low rpm
6. Turn off debug (by using the ".../api.html?debug=level0")
7. Press the high button on the remote.
8. Unit responds and goes to high rpm
9. Press the low button on the remote
10. Unit does not responds to low command

**Expected behaviour**
Set low rpm also responds when debug is off.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information**
- Current firmware version: 2.4.0-alpha1
- Hardware revision: 2
- Latest firmware version: 2.3.5
- CVE ECO RFT
- CC1101 RF module enabled

**Debug logging**
Debug of the rf shows "low" command
`26-2-2022 17:08:51: _W 152 --,--,-- --,--,-- 74,F5,46 22F1 3:63,02,04 (cmd:low)
26-2-2022 17:08:51: _W 152 --,--,-- --,--,-- 74,F5,46 22F1 3:63,02,04 (cmd:low)
26-2-2022 17:08:50: _W 152 --,--,-- --,--,-- 74,F5,46 22F1 3:63,02,04 (cmd:low)`

arjenhiemstra, 2022-02-26:
Thanks for this! I noticed something weird with this remote just before my holiday, cannot get my head around it (yet)..... will get to the bottom on this!

arjenhiemstra, 2022-02-28:
The problem is in these lines:

https://github.com/arjenhiemstra/ithowifi/blob/8b6164feeef5bc2cb12fff727b7e0676d1e84131/software/NRG_itho_wifi/IthoCC1101.cpp#L1314-L1320

The first 'else if' should be 'if' and the second the other way around. Thanks for the detailed description of the issue, it really helped to spot this!

roblomq, 2022-03-02:
Great to see you have fixed it already! Is it possible to download the updated firmware?

arjenhiemstra, 2022-03-02:
Should work again with the above mentioned version

closed 2022-03-02 by arjenhiemstra