arjenhiemstra/ithowifi #83: Fan speed 0 gives 1% speed in Domoticz

roblomq, 2022-03-05:
**Describe the bug**
When you turn off the fan with the slider in the web API then it stays on 1% in Domoticz.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect the wifi addon to Domoticz using the Domoticz MQTT functionality.
2. Go to webAPI and set slider to 0rpm
3. Go to the connected device in Domoticz and notice the device is on and on 1%

**Expected behaviour**
When the fan is off (0rpm) then also the switch in domoticz should be on status off and 0%.

**Screenshots**
[image]
[image]

**Device information**
- Current firmware version: 2.4.0-alpha4
- Hardware revision: 2
- CC1101 RF module enabled

sanderkob, 2022-03-05:
In my setup the slider shows 'Uit' (off) when speed=0. The slider is configured as Dimmer in Domoticz.
What does the MQTT message to the domoticz/in topic say?

arjenhiemstra, 2022-03-06:
This behaviour can be changed with the only downside that Domoticz does not update the slider if you set it to 0. It shows off in the interface but the slider stays at the last on position.

like this:
[image]

If that is preferred I'll update the Domoticz code in the next release.

roblomq, 2022-03-13:
I understand. As it's actually not a dimmer switch but a percentage slider but that doesn't exists. It's more convenient to use a selector switch with the same buttons als the RF remote. I've nw created this with a DZvents script in Domoticz. To get also this switch display the correct status when the RF remote is pressed, I update the status when the slider switch is changed.

closed 2022-07-15 by stale[bot]

roblomq, 2023-02-13:
I use the Dzvents scripts below.

"switch_to_itho.lua"

'
----------------------------------------------------------------------------------------------------------
-- Domoticz IDX of the needed devices
----------------------------------------------------------------------------------------------------------
local FanIDX 		= 2886	-- IDX of the itho selector switch device
local IthoIP		= '192.168.178.12'
local ScriptName 	= 'Switch to Itho'

return {

	active = true,
	
	on = 
	{
		devices = 	
		{
			-- Devices on which this switch script should trigger.
			FanIDX,
		},
		httpResponses = 
		{
			ScriptName,
		},
	},
	
	logging =
	{
		level	=	domoticz.LOG_INFO,
--		level	=	domoticz.LOG_DEBUG,
		marker	=	ScriptName
	},
	
	data = 
	{
		previousLevel = { initial = 10 }
	},
	
	execute = function(domoticz, devices)
		if domoticz.data.previousLevel ~= devices.level then
			domoticz.log('Fan state change', domoticz.LOG_INFO)
			if devices.level == 0 then
				domoticz.log('Fan now Off', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?speed=0')
				domoticz.data.previousLevel = 0
			elseif devices.level == 10 then
				domoticz.log('Fan now Low', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=low')
				domoticz.data.previousLevel = 10
			elseif devices.level == 20 then
				domoticz.log('Fan now Medium', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=medium')
				domoticz.data.previousLevel = 20
			elseif devices.level == 30 then
				domoticz.log('Fan now High', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=high')
				domoticz.data.previousLevel = 30
			elseif devices.level == 40 then
				domoticz.log('Fan now Timer 10', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=timer1')
				domoticz.data.previousLevel = 40
			elseif devices.level == 50 then
				domoticz.log('Fan now Timer 20', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=timer2')
				domoticz.data.previousLevel = 50
			elseif devices.level == 60 then
				domoticz.log('Fan now Timer 30', domoticz.LOG_INFO)
				domoticz.openURL('http://' .. IthoIP .. '/api.html?command=timer3')
				domoticz.data.previousLevel = 60
			end
		end
		
	end
}'

and "itho_to_switch.lua"

`
----------------------------------------------------------------------------------------------------------
-- Domoticz IDX of the needed devices
----------------------------------------------------------------------------------------------------------
local FanIDX 		= 2886	-- IDX of the itho selector switch device
local FanrpmIDX 	= 2884	-- IDX of the itho slider device
local IthoIP		= '192.168.178.12'
local ScriptName 	= 'Itho to switch'

return {

	active = true,
	
	on = 
	{
		devices = 	
		{
			FanrpmIDX,
		},
		httpResponses = 
		{
			ScriptName,
		},
	},
	
	logging =
	{
		level	=	domoticz.LOG_INFO,
--		level	=	domoticz.LOG_DEBUG,
		marker	=	ScriptName
	},
	
	execute = function(dz, item)
		local IthoStatus	= '-'

		if item.isHTTPResponse then
			dz.log('The speed (slider) of the Itho fan is changed', dz.LOG_DEBUG)
			local CurrentSpeed = tonumber(item.data)
			dz.log('The current speed setting of the Itho fan is ' .. CurrentSpeed .. '.', dz.LOG_DEBUG)
			if CurrentSpeed<=1 then
				IthoStatus = 'Off'
				dz.devices(FanIDX).switchSelector(0).silent()
			elseif CurrentSpeed>1 and CurrentSpeed<=20 then
				IthoStatus = 'Low'
				dz.devices(FanIDX).switchSelector(10).silent()
			elseif CurrentSpeed>20 and CurrentSpeed<=120 then
				IthoStatus = 'Medium'
				dz.devices(FanIDX).switchSelector(20).silent()
			elseif CurrentSpeed>120 then
				IthoStatus = 'High'
				dz.devices(FanIDX).switchSelector(30).silent()
			end
			dz.log('The status of the Itho fan selector switch is set to ' .. IthoStatus , dz.LOG_INFO)
		else
			dz.log('HTTP request is send', dz.LOG_DEBUG)			
			dz.openURL({
				url = 'http://' .. IthoIP .. '/api.html?get=currentspeed',
				method = 'GET',
				callback = ScriptName,
			})
			return
		end
	end
}`

They use the (dummy) devices below.
[image]