arjenhiemstra/ithowifi #85: DHCP renew in Wifi config improvements

jasperslits, 2022-03-13:
The wifi config has a 'DHCP renew interval' of 60. 

There are a few problems with that:
- Unit type is not provided. Is it in hours, minutes or days?
- The issuing DHCP-server controls the validity of the issued IP, so that should be respected to avoid excessive renewal requests
- Default is a validity of 24h / 86400 seconds

It's not a big problem but there's some room for improvement I think to either make it more consistent or remove the user-configurable flag all together. 
So low priority and created to follow 'no issue = no problem' approach.

arjenhiemstra, 2022-03-13:
Thanks for pointing this out, it's even worse. This setting is a remnant of a period when I tried to develop a Wiznet W5500 based ethernet interface. I just looked through the code, this setting isn't even used.....
I'll remove it in the next version.

jasperslits, 2022-03-13:
Thanks. I couldn’t figure out how it was used in the code to derive if the value of 60 was in hours or seconds, but now i understand why.

jasperslits, 2022-04-01:
Confirmed it's implemented in 2.4.0 beta

closed 2022-04-01 by jasperslits