arjenhiemstra/ithowifi #88: Compatibility with CVE ECO 2E RF?

ejohb, 2022-04-28:
Hi,

I was comparing the pins on my unit before buying but am not at all sure. Can anyone confirm if it should work?

Thank you!

[image]

[image]

roblomq, 2022-04-28:
I can confirm the addon will NOT fit on the control board of this CVE ECO2E RF.
What you can do is buy a second hand control board of a newer version, I was told that they are compatible for all the DC motors. I had the same ECO fan and also changed the control board to be able to fit the add-on.

ejohb, 2022-04-28:
> 

Many thanks for the info. Does something like this look suitable? https://www.elektrobode.nl/products/itho-daalderop-print-cve-cvd-eco-rft-s-basic-pcb-s-545-5102

roblomq, 2022-04-28:
Yes on this control board the add-on will fit but I'm not 100% sure this print will work with your engine because the information about this is based on someone I do not know.
But it's cheaper if you buy a second hand from marktplaats or like I did is buy a revised engine including the control board for €75,-

roblomq, 2022-04-28:
Also keep in mind that according my information the remotes of your unit also doesn't work with the add-on and also not with the new control board.

ShadowJonathan, 2022-07-13:
We also have this unit, and using the ESPEasy route doesnt work, as the old RF protocol hasn't been reverse engineered.

Fietspomp86, 2022-07-28:
> We also have this unit, and using the ESPEasy route doesnt work, as the old RF protocol hasn't been reverse engineered.

Sorry to go off-topic, but I'm sure this is beneficial for people. (I also have the ECO 2E fan)

The ESPEasy code contains commented parts which make the old devices not work anymore.
I've forked the libraries and un-commented the affected lines.
Take a look:
https://github.com/Fietspomp86/esphome_itho

Using the ESPhome method I'm able to control my fan!

Anyway, I'm also looking at secondhand fans to replace the PCB to make life more simple :-)

arjenhiemstra, 2022-07-28:
Interesting! Could you give an idea which lines you uncommented? Your commit history doesn't show those details unfortunately.

Fietspomp86, 2022-07-28:
> Interesting! Could you give an idea which lines you uncommented? Your commit history doesn't show those details unfortunately.

Yeah I downloaded this lib a long time ago from somewhere I can't remember, so I freshly uploaded it instead of forking/editing.

Here's the changes:
IthoCC1101.cpp - lines 441, 453, 477, 481, 899, 925. (basically only removing "// or /*", since the code is already there)

There are some more small changes, however the library I've downloaded as I mentioned is already a few commits behind the one from Compatech. If I can find the time, I'll fork his properly and edit them.

arjenhiemstra, 2022-07-28:
In your repo or the one you forked?
In your repo there seems to be nothing at 441 and 453 for example
https://github.com/Fietspomp86/esphome_itho/blob/3dcda35274a5c92ffccd87bd20e7695ac6f80c85/IthoCC1101.cpp#L441
https://github.com/Fietspomp86/esphome_itho/blob/3dcda35274a5c92ffccd87bd20e7695ac6f80c85/IthoCC1101.cpp#L453

Fietspomp86, 2022-07-28:
> In your repo or the one you forked? In your repo there seems to be nothing at 441 and 453 for example https://github.com/Fietspomp86/esphome_itho/blob/3dcda35274a5c92ffccd87bd20e7695ac6f80c85/IthoCC1101.cpp#L441 https://github.com/Fietspomp86/esphome_itho/blob/3dcda35274a5c92ffccd87bd20e7695ac6f80c85/IthoCC1101.cpp#L453
> 
> edit: probably the lines shifted somewhat. You mean the message1 parts of the code?

Yeah I mixed some things up, it's been a while since I used it.
Apparently I did fork a while back, from Jodur and adjusted it, here it is
https://github.com/Fietspomp86/ITHO-Lib

I also now forked the lib from Compatech and adjusted the IthoCC1101.cpp file:
https://github.com/Fietspomp86/esphome_itho-old

Hope this makes some sense :-D

arjenhiemstra, 2022-07-28:
Yeah it does, thanks! Now I finally understand where the message1 code was for. Could not relate it back to the remote commands I captured. 
The good news; looking at the code this remote uses also manchester encoded rf data.
The bad news; The way it's currently implemented makes it very hard to change the deviceID, its most probably a totally different protocol and to make it to a workable version (ie. get to a version where you can control the deviceID and properly receive and recreate messages) I probably need an actual remote to make some captures.

ShadowJonathan, 2022-07-29:
FWIW ive reverse-engineered (some part) of the old RF protocol on my own; https://github.com/letscontrolit/ESPEasy/issues/4131#issuecomment-1194088921

Fietspomp86, 2022-08-05:
> Yeah it does, thanks! Now I finally understand where the message1 code was for. Could not relate it back to the remote commands I captured. The good news; looking at the code this remote uses also [...]

I just bought another Itho box secondhand (so I can use your device :-) ), so I'm ditching the old box. 
I could of course donate it to science, if that suits you?
It's not getting any bids on Marktplaats anyway.

Would still be an issue to connect your device to the itho box without the 4x2 pin header.

aprosvetova, 2022-08-21:
I was trying to figure out the way the RF PCB communicates with the main board.
Here are my findings.

The RF PCB has 3 ICs: voltage regulator, op amp (why?), and a weird general purpose "rf receiver" TDA5210.

9-pin header:
[image]

How does TDA5210 work? It doesn't feel like it can be programmed, it's just a receiver/demodulator.
Does it mean there is no I2C/UART on this 9-pin header, and the protocol handling is done by the main board?
If so, how is the PWM PCB supposed to work (536-0201), and why is it called I2C-PWM?

So many questions, almost no answers...
I contacted Itho and asked:
1. If they can still sell the PWM PCB with 9-pin header (I was unable to find it in stores, not even pictures)
2. If they can share some details about the transport/protocol exposed on this 9-pin header (I2C?) and it's pinout

Waiting 🤷🏻‍♀️ 
I'm close to giving up and just hooking up a zigbee relay to 220v contacts that are meant for the 3-positioned switch, but I really want precise control :D

arjenhiemstra, 2023-10-31:
Here are two images from the itho PWM module (from tweakers topic: https://gathering.tweakers.net/forum/list_messages/1690945)
[image]
[image]

looking at the print it seems that the signal that comes in is first rectified (S1NB) and thereafter fed through an optocoupler (SNY17) and the output of that is fed in a double comparator (LM358) to probably just to make it a nice square wave and match the signal level expected by the itho board.

ejohb, 2024-12-09:
Hi again all,

Hopefully this won't be seen too much like self-promotion/marketing, but I've since moved house, so have a compatible controller plus WiFi add-on on eBay, if anyone needs one:

https://www.ebay.co.uk/itm/235864114737