arjenhiemstra/ithowifi #89: Get this  working with ESPHome

hencou, 2022-05-01:
Great [roject! I want to use this module with ESPHome and HomeAssistant, are there already working examples from users which are using this hardware (and software pieces) and ESPHome?

jasperslits, 2022-05-01:
I have 5 modules connected to Itho autotemp, WPU and HRU/WTW and all integrated via MQTT to home assistant. What added value would ESPHome have in this case? Arjen’s code and add-on seems a great fit.

arjenhiemstra, 2022-05-01:
As Jasper mentioned, you can use this firmware with HomeAssistant without any changes, no need to port things to ESPHome.
If you feel certain ESPHome functionality is missing and should be in the firmware, please feel free to give feedback on that.

hencou, 2022-05-02:
Hi Arjen and Japer,

Of course I will get it work with MQTT, no doubt. But I have 30+ ESP devices around my home. They are all centrally managed by ESPHome. On that way I am able to centrally manage all the modules, and I was wondering if there was somebody who have already created somthing like that for this project. But no problem, I will have a look if I can port the code into a custom "esphome_component", a kind of wrapper around the existing code.

jasperslits, 2022-05-02:
Ok, but what kind of management would you then need? All updates from / to device are MQTT messages, it's really - for me at least - configure it once and then forget about it unless a firmware updates comes along that fixes something I need or adds a feature I am interested in.

hencou, 2022-05-02:
Hi Jasper, firmware updates and fixes can be a reason. I have many of the same kind of devices which I can update now with a single click of a mouse with ESPHome. ESPHome will download the latest libraries from the internet, compile the firmware with PlatformIO and does an automatich Over The Air update to the ESP device. I want to have ithowifi in the same ecosystem, and no MQTT server for only this device. To prevent reinventing the wheel I was opening this ticket, hoping that somebody did already something like this.

arjenhiemstra, 2022-05-10:
> Hi Arjen and Japer,
>
> Of course I will get it work with MQTT, no doubt. But I have 30+ ESP devices around my home. They are all centrally managed by ESPHome. On that way I am able to centrally manage [...]

Interesting use case, if you succeed in porting this I would appreciate it very much if you could share your progress!

hencou, 2022-05-21:
> Interesting use case, if you succeed in porting this I would appreciate it very much if you could share your progress!

Hi Arjen,

See here for the first working version: [https://github.com/hencou/esphome_components](https://github.com/hencou/esphome_components). The next step is to get the Itho CVE builtin sensors native in ESPHome. When that succeeds, its will be possible to create automations in ESPHome, On that way its possible the let the module do its own work, stanalone, and bypassing the Itho logic.

hencou, 2022-06-08:
The project is now fully operational and completed. It now also works standalone with an integrated PID control. This ticket can therefore be closed. 

It took more time than expected, I also had to rewrite the existing code to make it fit in ESPhome. For the future it would be nice if you split the code into "Core" and "Frontend" blocks. Then ESPhome and other frontends could include that "Core" code from your repository, and only have to do calls like "itho->setSpeed(50)" or "speed = itho->getSpeed()". 

Thanks for your input for now, I am happy with your module!

arjenhiemstra, 2022-07-09:
Thanks for all the work and thanks for the hints. This is basically my first public project and I have no background or education in coding. I really appreciate your work, will look at it further and try to learn from it. Thanks!

closed 2022-09-14 by stale[bot]

hencou, 2023-02-02:
Hi Clandmeter,

Should be possible with a FTDI but the most easy way is to choose "Manual download" at install in ESPHome. Then upload that downloaded file as new firmware on the IthoWifi webpage. 

Ensure that your wifi settings in the yaml config are correct, otherwise you have to flash with FTDI for sure after flashing with wrong wifi credentials.

clandmeter, 2023-02-04:
@hencou thanks for your explanation on how to get the firmware on the board.
Could you explain to me how to do it via FTDI?
I am not sure what happens, could also be that my ftdi does not provide enough power. 
Do i need any extra steps as to connect VCC/GND/RX/TX and upload via esphome?

arjenhiemstra, 2023-02-04:
VCC is particular critical during device flashing, that is when the power draw is greatest.
Ensure that your FTDI solution can deliver at least 300mA at 3.3V (most don't unfortunately)

clandmeter, 2023-02-04:
I guess due to no confirmation of any additional step i don't have to pull down anything?

arjenhiemstra, 2023-02-04:
FTDI and esptool based flashing tool should suffice.
Auto reset works to put the module in flash mode. After flashing you have to manually reset the module.

zipkid, 2023-05-12:
@hencou Does your EspHome solution also support the RF-MODULE?