arjenhiemstra/ithowifi #90: Virtual remote - RFT CO2 230V support (feature req)

jasperslits, 2022-05-05:
I have multiple RTF CO2 230V (04-00045) units which support low, medium, high, timer1,timer2,timer3, auto and autonight, join and leave. It's linked to the HRU-350 and its primary use to feed CO2 measurements but also allows for manual changes. 

Under virtual remotes the capabilities shown contain all of the above except for "medium", so none of the existing virtual remotes seems a good fit. 

Is it possible to either this remote a a supported virtual remote or add "Medium" via other means for MQTT control?

arjenhiemstra, 2022-05-10:
You mean that you want to be able to select the RFT CO2 als virtual remote type (next to the current 3 times)?

That can be implemented quite easy with just a few extra lines of code

jasperslits, 2022-05-10:
> You mean that you want to be able to select the RFT CO2 als virtual remote type (next to the current 3 types)?
> 
> That can be implemented quite easy with just a few extra lines of code

Great, that would be sufficient I think.

arjenhiemstra, 2022-05-20:
Could you try with the latest alpha7 version?

jasperslits, 2022-05-20:
Awesome, works fine!

closed 2022-05-20 by jasperslits